MoshiMoshi0/TTController #59: Configuration Help

xxtrilla, 2020-07-14:
**Describe your desired configuration**
A clear and concise description of what your desired effect and/or speed controller configuration is. Make sure you include how many controllers you have and what devices are connected to each controller.
I have no idea how to do this , I bought a pc online, and i got ThermalTake 120mm Pure Plus 12 LED RGB Radiator Fan TT Premium Edition PWM 500~1500rpm (3 Fan Pack) w/ Software Controller inside them, right now they are jus running at however i got them from the pc company, i would like the aurora effect please, just any effect that isnt this ugly rgb at all times. My processor is a ryzen 9 3900x and 2070 super gpu on a 570 motherboard, with the original software it showed could not find controller. 

if aurora is the default that comes with it , please give me a profile with flow, but if aurora is not default ill take that..
**Hardware info**
```
Replace this text (between ``` characters) with a debug report.
You can generate it by going to "Main Menu -> Debug -> Report" in the command line menu.

Make sure no official software is not running at the same time.
```

**Current configuration**
```json
If applicable, replace this text (between ``` characters) with contents of your current configuration here. 
```

MoshiMoshi0, 2020-07-14:
I would need that debug report first.
Run `TTController.Service.exe` and go to `Debug -> Report` and post the output.

xxtrilla, 2020-07-14:
-----------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 33.0 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 28.0 °C
»                Temperature #3 (/lpc/nct6798d/temperature/3): 26.0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 45.0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 2.0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 25.0 °C
»
»  AMD Ryzen 9 3900X:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 51.9 °C
»                CCD1 (Tdie) (/amdcpu/0/temperature/3): 41.0 °C
»                CCD2 (Tdie) (/amdcpu/0/temperature/4): 37.3 °C
»                CCDs Max (Tdie) (/amdcpu/0/temperature/5): 41.0 °C
»                CCDs Average (Tdie) (/amdcpu/0/temperature/6): 39.1 °C
»
»  NVIDIA GeForce RTX 2070 SUPER:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 38.0 °C
»
»  WDC WD30PURZ-85GU6Y0:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 27.0 °C
»
»  WDC WDS480G2G0A-00JH30:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 33.0 °C
»
»  WDC WDS100T2B0C-00PXH0:
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 34.0 °C

doesnt seem to show anything for controllers?

MoshiMoshi0, 2020-07-14:
Yes, if TTController and the official software do not detect the controller then I cant really do much.
Either the controller is bad, or it is not connected correctly to the motherboard. It should be connected to one of the usb headers on the motherboard.

MoshiMoshi0, 2020-07-15:
I will close this for now since I cant really help.
If you manage to make the official software detect the controller, reply here if you still want to use TTController and I will reopen.

closed 2020-07-15 by MoshiMoshi0