MoshiMoshi0/TTController #60: AuroraEffect Not Working

Twisted6, 2020-07-19:
**Describe the issue**
The colour on the fans isn't changing. I'm not sure if it's a config issue but the console isn't showing any errors and the PWM control works.

**Debug report**
```
»        OS: Windows 10 Enterprise 1903 [18362.239.x86fre.19h1_release_svc_prod1.190628-1641]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8501
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 31, Rpm: 512, Unknown: 255]
»                        Identifier: [9802, 8501, 1]
»                2:
»                        Data: [PortId: 2, Speed: 31, Rpm: 563, Unknown: 255]
»                        Identifier: [9802, 8501, 2]
»                3:
»                        Data: [PortId: 3, Speed: 31, Rpm: 544, Unknown: 255]
»                        Identifier: [9802, 8501, 3]
»                4:
»                        Data: [PortId: 4, Speed: 30, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8501, 4]
»                5:
»                        Data: [PortId: 5, Speed: 30, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8501, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6797D:
»        Temperature:
»                Temperature #1 (/lpc/nct6797d/temperature/1): 43.0 °C
»                Temperature #2 (/lpc/nct6797d/temperature/2): 38.0 °C
»                Temperature #3 (/lpc/nct6797d/temperature/3): 47.5 °C
»                Temperature #5 (/lpc/nct6797d/temperature/5): 47.0 °C
»                Temperature #6 (/lpc/nct6797d/temperature/6): -2.0 °C
»
»  AMD Ryzen 5 3600:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 55.3 °C
»                CCD1 (Tdie) (/amdcpu/0/temperature/3): 52.8 °C
»
»  NVIDIA GeForce GTX 1070 Ti:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 53.0 °C
»
»  INTEL SSDPEKNW020T8:
»        Temperature:
»                Temperature (/nvme/3/temperature/0): 38.0 °C
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [50, 30],
              [65, 35],
              [75, 45],
              [85, 55],
              [100, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
              "Step": 0.003,
              "Length": 64,
              "Mirror": false,
              "Brightness": 1.0,
              "Saturation": 1.0
          }
        }
      ]
    }
  ]
}
```

MoshiMoshi0, 2020-07-19:
You didnt add a trigger to the effect:
```json
      "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Step": 0.003,
            "Length": 64,
            "Mirror": false,
            "Brightness": 1.0,
            "Saturation": 1.0,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]

Twisted6, 2020-07-20:
Derp, thanks. The effect is working now but the lights are very dim.

MoshiMoshi0, 2020-07-20:
Sure

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [50, 30],
              [65, 35],
              [75, 45],
              [85, 55],
              [100, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Step": 0.003,
            "Length": 64,
            "Mirror": true,
            "Brightness": 1.0,
            "Saturation": 1.0,
            "ColorGenerationMethod": "SpanPorts",
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "Config": {
        "DeviceType": "RiingTrio"
      }
    }
  ]
}
```

Twisted6, 2020-07-21:
Thanks again. If anyone else finds this I found "Step": -0.003, "Length": 256 to be close to the official Aurora effect.

closed 2020-07-21 by Twisted6