MoshiMoshi0/TTController #61: Configuration Help

Fredzone123, 2020-07-21:
**Describe your desired configuration**
Hi there I would like some help in configuring to the settings I would like.
I have 2 controllers connected to a total of 8 TT RGB RIING fans and a TT RGB AIO pump
I was hoping to configure all the fans to alternating blue and orange colours.
My main problem is that I want to adjust the PWM fan curve which the app doesn't allow me to.
I want to decrease the fan curve so fan speeds are all at lowest possible for 30-65 degrees then from 65 degrees to be at 50% and then at 75 degrees to be at maximum
I want to decrease the fluctuation changes as well.

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Home 2004 [19041.1.x86fre.vb_release.191206-1406]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8102
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 60, Rpm: 1096, Unknown: 255]
»                        Identifier: [9802, 8102, 1]
»                2:
»                        Data: [PortId: 2, Speed: 60, Rpm: 1105, Unknown: 255]
»                        Identifier: [9802, 8102, 2]
»                3:
»                        Data: [PortId: 3, Speed: 60, Rpm: 1085, Unknown: 255]
»                        Identifier: [9802, 8102, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8103
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 60, Rpm: 1080, Unknown: 255]
»                        Identifier: [9802, 8103, 1]
»                2:
»                        Data: [PortId: 2, Speed: 60, Rpm: 1056, Unknown: 255]
»                        Identifier: [9802, 8103, 2]
»                3:
»                        Data: [PortId: 3, Speed: 60, Rpm: 1061, Unknown: 255]
»                        Identifier: [9802, 8103, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 1075, Unknown: 255]
»                        Identifier: [9802, 8103, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 1095, Unknown: 255]
»                        Identifier: [9802, 8103, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
```

**Current configuration**
Currently on default pregenerated one. I've been trying to edit and correct it. I'm very new to this and am sometimes unable to startup, or it starts but the effects do not take place.
```

Fredzone123, 2020-07-21:
Looking at my sensors i cannot see cpu temperature being recorded, could that be the issue? Im trying to edit PWM curve 

Sensors:
Intel Core i7-10700K:
»        Load:
»                CPU Core #1 (/intelcpu/0/load/1): 10.8 %
»                CPU Core #2 (/intelcpu/0/load/2): 4.9 %
»                CPU Core #3 (/intelcpu/0/load/3): 3.0 %
»                CPU Core #4 (/intelcpu/0/load/4): 1.1 %
»                CPU Core #5 (/intelcpu/0/load/5): 2.4 %
»                CPU Core #6 (/intelcpu/0/load/6): 1.1 %
»                CPU Core #7 (/intelcpu/0/load/7): 1.0 %
»                CPU Core #8 (/intelcpu/0/load/8): 6.7 %
»                CPU Total (/intelcpu/0/load/0): 3.9 %
»
»  Generic Memory:
»        Data:
»                Memory Used (/ram/data/0): 5 GB
»                Memory Available (/ram/data/1): 11 GB
»                Virtual Memory Used (/ram/data/2): 8 GB
»                Virtual Memory Available (/ram/data/3): 11 GB
»        Load:
»                Memory (/ram/load/0): 31.6 %
»                Virtual Memory (/ram/load/1): 41.7 %
»
»  NVIDIA GeForce GTX 1080:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 59.0 °C
»        Clock:
»                GPU Core (/gpu/0/clock/0): 1683 MHz
»                GPU Memory (/gpu/0/clock/1): 5006 MHz
»                GPU Shader (/gpu/0/clock/2): 3366 MHz
»        Fan:
»                GPU (/gpu/0/fan/0): 0 RPM
»        Load:
»                GPU Core (/gpu/0/load/0): 11.0 %
»                GPU Memory Controller (/gpu/0/load/1): 8.0 %
»                GPU Video Engine (/gpu/0/load/2): 0.0 %
»                GPU Bus (/gpu/0/load/4): 0.0 %
»                GPU Memory (/gpu/0/load/3): 16.9 %
»        Control:
»                GPU Fan (/gpu/0/control/0): 10.0 %
»        SmallData:
»                GPU Memory Total (/gpu/0/smalldata/3): 8192.0 MB
»                GPU Memory Used (/gpu/0/smalldata/2): 1388.5 MB
»                GPU Memory Free (/gpu/0/smalldata/1): 6803.5 MB
»
»  KINGMAX SSD 240GB:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 36.0 °C
»        Load:
»                Used Space (/hdd/1/load/0): 22.5 %
»                Write Activity (/hdd/1/load/32): 0.0 %
»                Total Activity (/hdd/1/load/33): 0.0 %
»        Throughput:
»                Read Rate (/hdd/1/throughput/34): 863373.1 B/s
»                Write Rate (/hdd/1/throughput/35): 265389.9 B/s
»
»  WDC WD10EZEX-60WN4A0:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 37.0 °C
»        Load:
»                Used Space (/hdd/2/load/0): 7.9 %
»                Write Activity (/hdd/2/load/32): 0.0 %
»                Total Activity (/hdd/2/load/33): 1.8 %
»        Throughput:
»                Read Rate (/hdd/2/throughput/34): 28201.6 B/s
»                Write Rate (/hdd/2/throughput/35): 900.6 B/s
»
»  Samsung SSD 860 EVO M.2 1TB:
»        Temperature:
»                Temperature (/ssd/0/temperature/0): 40.0 °C
»        Data:
»                Total Bytes Written (/ssd/0/data/0): 2620 GB
»        Load:
»                Used Space (/ssd/0/load/0): 80.9 %
»                Write Activity (/ssd/0/load/32): 0.0 %
»                Total Activity (/ssd/0/load/33): 0.5 %
»        Throughput:
»                Read Rate (/ssd/0/throughput/34): 23794.9 B/s
»                Write Rate (/ssd/0/throughput/35): 107.7 B/s
»
»  Ethernet:
»        Data:
»                Data Uploaded (/nic/{60A3A4B0-A460-4B43-B539-423E5C15CC47}/data/2): 0 GB
»                Data Downloaded (/nic/{60A3A4B0-A460-4B43-B539-423E5C15CC47}/data/3): 0 GB
»        Throughput:
»                Upload Speed (/nic/{60A3A4B0-A460-4B43-B539-423E5C15CC47}/throughput/7): 0.0 B/s
»                Download Speed (/nic/{60A3A4B0-A460-4B43-B539-423E5C15CC47}/throughput/8): 0.0 B/s
»        Load:
»                Network Utilization (/nic/{60A3A4B0-A460-4B43-B539-423E5C15CC47}/load/1): 0.0 %
»
»  Ethernet 2:
»        Data:
»                Data Uploaded (/nic/{4FA9930A-96D1-495C-8521-1DA7B1E2A2CC}/data/2): 0 GB
»                Data Downloaded (/nic/{4FA9930A-96D1-495C-8521-1DA7B1E2A2CC}/data/3): 0 GB
»        Throughput:
»                Upload Speed (/nic/{4FA9930A-96D1-495C-8521-1DA7B1E2A2CC}/throughput/7): 0.0 B/s
»                Download Speed (/nic/{4FA9930A-96D1-495C-8521-1DA7B1E2A2CC}/throughput/8): 0.0 B/s
»        Load:
»                Network Utilization (/nic/{4FA9930A-96D1-495C-8521-1DA7B1E2A2CC}/load/1): 0.0 %

MoshiMoshi0, 2020-07-21:
> Looking at my sensors i cannot see cpu temperature being recorded, could that be the issue? Im trying to edit PWM curve.

I have to update LibreHardwareMonitor for your cpu to report temperatures, ill post a build for you here.

> I was hoping to configure all the fans to alternating blue and orange colours.

You mean like smooth pulsing or blinking?

Fredzone123, 2020-07-21:
Okay I shall patiently await, thanks for your help mate really appreciate it.

In terms of RGB lighting to be exact it would be nice to set the fans with dark orange, then the next fan beside it to teal blue, and then dark orange etcetc throughout the system.
It'd be awesome if the colours then synchronously slowly all fade out and switch colours, so orange becomes blue and blue becomes orange, every few seconds.

MoshiMoshi0, 2020-07-21:
I had to write a new effect to achieve what you want, so there is no documentation but it is pretty self explanatory from the config. Also it might change a little on the next release, there should be documentation for it then tho.

You might need to change the sensor identifier to match your system, and shuffle the port identifiers between profiles because i dont know how they are physically in you computer case.

Build:
[TTController_0.10.0.0-alpha.9.4d467df620db91834d4e9528346df1949ef16a15.zip](https://github.com/MoshiMoshi0/TTController/files/4955265/TTController_0.10.0.0-alpha.9.4d467df620db91834d4e9528346df1949ef16a15.zip)

Config:
```json
{
  "Profiles": [
    {
      "Name": "Odd",
      "Ports": [
        [9802, 8102, 1],
        [9802, 8102, 3],
        [9802, 8103, 2],
        [9802, 8103, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8"],
            "MinimumChange": 4,
            "MaximumChange": 8
          }
        }
      ],
      "Effects": [
        {
          "Type": "ColorSequenceEffect",
          "Config": {
            "Sequence": [
              { 
                "TransitionTime": 1000,
                "HoldTime": 1000,
                "Color": { "Full": [0, 128, 128] }
              },
              {
                "TransitionTime": 1000,
                "HoldTime": 1000,
                "Color": { "Full": [255, 140, 0] }
              }
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Even",
      "Ports": [
        [9802, 8102, 2],
        [9802, 8103, 1],
        [9802, 8103, 3],
        [9802, 8103, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "CopySpeedController",
          "Config": {
            "Target": [9802, 8102, 1],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "ColorSequenceEffect",
          "Config": {
            "Sequence": [
              { 
                "TransitionTime": 1000,
                "HoldTime": 1000,
                "Color": { "Full": [255, 140, 0] }
              },
              {
                "TransitionTime": 1000,
                "HoldTime": 1000,
                "Color": { "Full": [0, 128, 128] }
              }
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ]
}
```

Fredzone123, 2020-07-22:
Hi Moshi,
The colour effects work perfectly!
The PWMSpeed controller still doesn't work though, I've installed the new build but fan speed are still static and don't change with temp changes, are you able to fix this? Thanks mate.

Fredzone123, 2020-07-22:
Nevermind I just added the Trigger line after the first speed controllers bit and it worked perfectly.
Thank you!!!

MoshiMoshi0, 2020-07-22:
Oh right I forgot the trigger, glad you sorted it out.

closed 2020-07-22 by MoshiMoshi0