MoshiMoshi0/TTController #62: Does not detect controller

TurtleHunter, 2020-07-26:
**Describe the issue**
Controller is not detected. I am able to use the Thermaltake software just fine.
**Debug report**
```
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6791D:
»        Temperature:
»                CPU Core (/lpc/nct6791d/temperature/0): 51.0 °C
»                Temperature #1 (/lpc/nct6791d/temperature/1): 42.5 °C
»                Temperature #2 (/lpc/nct6791d/temperature/2): 113.0 °C
»                Temperature #3 (/lpc/nct6791d/temperature/3): 38.0 °C
»                Temperature #4 (/lpc/nct6791d/temperature/4): 107.0 °C
»                Temperature #5 (/lpc/nct6791d/temperature/5): 104.0 °C
»                Temperature #6 (/lpc/nct6791d/temperature/6): 104.0 °C
»
»  Intel Core i5-4690K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 47.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 52.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 47.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 51.0 °C
»                CPU Package (/intelcpu/0/temperature/4): 52.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/5): 53.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/6): 48.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/7): 53.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/8): 49.0 °C
»                Core Max (/intelcpu/0/temperature/9): 52.0 °C
»                Core Average (/intelcpu/0/temperature/10): 49.3 °C
»
»  NVIDIA GeForce GTX 970:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 36.0 °C
»
»  SAMSUNG MZHPV128HDGM-00000:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 37.0 °C
»
»  WDC WD3003FZEX-00Z4SA0:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 44.0 °C
»
»  WDC WD5001FZWX-00ZHUA0:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 41.0 °C
»
-------------------------------```

**Current configuration**
```json
{
  "Profiles": [],
  "ComputerStateProfiles": [],
  "PortConfigs": [],
  "SensorConfigs": []
}```

MoshiMoshi0, 2020-07-26:
What fans do you have? Also make sure you dont use TTControler and thermaltake software at the same time.

TurtleHunter, 2020-07-26:
My fans are Thermaltake Riing Plus 12, and I have a Pacific Luma Plus that I just installed yesterday

MoshiMoshi0, 2020-07-26:
Please update to this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/34310433/artifacts and post the debug report again

TurtleHunter, 2020-07-26:
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha.11+Branch.develop.Sha.ab1f7b17439a6001467e12872b0f0f240bba632e
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»  [0x264a, 0x2260]: \\?\hid#vid_264a&pid_2260#7&30f86f06&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6791D:
»        Temperature:
»                CPU Core (/lpc/nct6791d/temperature/0): 51.5 °C
»                Temperature #1 (/lpc/nct6791d/temperature/1): 42.5 °C
»                Temperature #2 (/lpc/nct6791d/temperature/2): 113.0 °C
»                Temperature #3 (/lpc/nct6791d/temperature/3): 38.0 °C
»                Temperature #4 (/lpc/nct6791d/temperature/4): 107.0 °C
»                Temperature #5 (/lpc/nct6791d/temperature/5): 104.0 °C
»                Temperature #6 (/lpc/nct6791d/temperature/6): 104.0 °C
»
»  Intel Core i5-4690K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 48.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 52.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 49.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 51.0 °C
»                CPU Package (/intelcpu/0/temperature/4): 52.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/5): 52.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/6): 48.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/7): 51.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/8): 49.0 °C
»                Core Max (/intelcpu/0/temperature/9): 52.0 °C
»                Core Average (/intelcpu/0/temperature/10): 50.0 °C
»
»  NVIDIA GeForce GTX 970:
»        Temperature:
»                GPU Core (/gpu-nvidia/0/temperature/0): 36.0 °C
»
»  SAMSUNG MZHPV128HDGM-00000:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 35.0 °C
»
»  WDC WD3003FZEX-00Z4SA0:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 49.0 °C
»
»  WDC WD5001FZWX-00ZHUA0:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 46.0 °C
»
-------------------------------
Press any key to continue...

MoshiMoshi0, 2020-07-26:
It seems like you have a Riing Quad controller which is currently not supported. That is weird that you have Riing Plus fans with Riing Quad controller.

If you want to help me implement that you would need to provide me with a wireshark usb capture:

1. Install [Wireshark](https://www.wireshark.org/) with usbpcap (you will be asked if you want to install usbpcap in the installer)
1. Stop any thermaltake software
1. Start Wireshark
1. Double click on USBPCap1 capture on the main screen
1. Start the official software (without the TTController service running)
1. After 10s stop the capture in wireshark using the red square button
1. Save and upload the dump using File -> Save As

MoshiMoshi0, 2020-07-26:
Also preferably you would switch the colors of the fans a little bit while recording, just so they are not constantly the same color.

TurtleHunter, 2020-07-26:
Will do! It will take me a few days because of work.
The Quad controller is probably because I had to RMA a defective one. I also received one with the Lumi Plus, do you want me to take a capture using that as well?

MoshiMoshi0, 2020-07-27:
No problem.
You could set the lumi plus to a static color like red, but change the color of the fans a few times.

TurtleHunter, 2020-07-28:
Uploaded. I think its #3 that has the data you're looking for. Let me know if you need me to run it again. Dumps 1-3 are with the Quad controller, ill do a Lumi one here soon, will be 4-6. https://drive.google.com/drive/folders/15gY8IRPmxtkyr7OP_trNFfRIE5dnOCYx?usp=sharing

TurtleHunter, 2020-07-28:
Here's a report with the controller that came with the Lumi Plus

`Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha.11+Branch.develop.Sha.ab1f7b17439a6001467e12872b0f0f240bba632e
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»  [0x264a, 0x1fa5]: \\?\hid#vid_264a&pid_1fa5#7&a0ee4bd&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 70, Rpm: 1245, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 70, Rpm: 1232, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6791D:
»        Temperature:
»                CPU Core (/lpc/nct6791d/temperature/0): 56.5 °C
»                Temperature #1 (/lpc/nct6791d/temperature/1): 45.0 °C
»                Temperature #2 (/lpc/nct6791d/temperature/2): 112.0 °C
»                Temperature #3 (/lpc/nct6791d/temperature/3): 39.5 °C
»                Temperature #4 (/lpc/nct6791d/temperature/4): 107.0 °C
»                Temperature #5 (/lpc/nct6791d/temperature/5): 104.0 °C
»                Temperature #6 (/lpc/nct6791d/temperature/6): 104.0 °C
»
»  Intel Core i5-4690K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 51.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 54.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 51.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 51.0 °C
»                CPU Package (/intelcpu/0/temperature/4): 54.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/5): 49.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/6): 46.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/7): 49.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/8): 49.0 °C
»                Core Max (/intelcpu/0/temperature/9): 54.0 °C
»                Core Average (/intelcpu/0/temperature/10): 51.8 °C
»
»  NVIDIA GeForce GTX 970:
»        Temperature:
»                GPU Core (/gpu-nvidia/0/temperature/0): 42.0 °C
»
»  SAMSUNG MZHPV128HDGM-00000:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 39.0 °C
»
»  WDC WD3003FZEX-00Z4SA0:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 44.0 °C
»
»  WDC WD5001FZWX-00ZHUA0:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 48.0 °C
»
-------------------------------
Press any key to continue...`

MoshiMoshi0, 2020-07-29:
Can you tell me what device was connected to what port? Like what you had selected in thermaltake software for each port, because something is weird in that dump3.

And if you could do one more dump with this test setup in thermaltake software (only on the Riing Quad controller):
* Port 1: Riing Plus, full red color
* Port 2: Riing Duo, first zone full red color, second zone full green color
* Port 3: Riing Trio, first zone full red color, second zone full green color, third zone full blue color
* Port 4: Riing Quad, first zone full red color, second zone full green color, third zone full blue color, fourth zone full white color
* Port 5: Lumi Plus, full red color

I know this is what you have but it will force the software to send specific commands for all fan types.
You can disconnect the cables from the controller ports if you really want, but it should not be an issue.

Thank you for your help.

MoshiMoshi0, 2020-08-11:
Will you be able to provide the dump with above setup?
I would like to properly implement the quad controller but the previous dump had unexpected behavior.

TurtleHunter, 2020-08-11:
Sorry about that, I haven't forgotten!! I'll get to it within the next
couple days. Had the inlaws over last week and I've been working 12 hour
shifts ever since.

TurtleHunter, 2020-08-13:
Second round is uploaded to the same folder at https://drive.google.com/drive/folders/15gY8IRPmxtkyr7OP_trNFfRIE5dnOCYx?usp=sharing files are 27, 28, and 29 for USBcap1 USBcap2 and USBcap3 respectively. Let me know if this works or not, I really want to help!

MoshiMoshi0, 2020-08-13:
Nice! Apparently there wasnt anything special with this controller and I was just overthinking.
Try this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/34643080/artifacts
It should detect the controller on the debug menu.

Test config:
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 4],
        [9802, 8800, 5]
      ],
      "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Step": 0.003,
            "Length": 64,
            "Mirror": true
          }
        }
      ]
    }
  ]
}
```

MoshiMoshi0, 2020-08-28:
Sorry to nag you like this but I would like to mark riing quad support as solved.
Please post the debug report and check if the above config works.

benjaminsavoy, 2020-09-10:
Hi, I'm another Riing Quad user and I've tried the above build.

I've installed it as per the instructions, and as I try to get the controller list, I have the following error (sorry that it's a screenshot, but as the program closes right away when it encounters an exception, I could only be fast enough for this).

[image]

I'm also reasonably capable with C# so I'd be happy to help although don't have much time to look into it this week.

MoshiMoshi0, 2020-09-10:
@benjaminsavoy Well, it seems I forgot to implement the version check.
Fixed in this: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/35135865/artifacts

benjaminsavoy, 2020-09-10:
Everything looks good. For what it's worth, that's the config I used to test it.

```
{
  "Profiles": [
  {
      "Name": "Default",
      "Ports": [
        [9802, 8801, 1],
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 3]
      ],
	  "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Step": 0.003,
            "Length": 64,
            "Mirror": true
          }
        }
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 30],
              [40, 50],
              [50, 75],
              [65, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }]
	}],
  "PortConfigs": [{
      "Ports": [
        [9802, 8801, 1],
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 3]
      ],
      "Config": {
        "DeviceType": "RiingQuad",
      }
    }]
}
```

MoshiMoshi0, 2020-09-10:
Nice, could you also post the output of the controllers report from debug menu?

benjaminsavoy, 2020-09-10:
Here it is:
```
HID
-------------------------------
»
»  [0x264a, 0x2260]: \\?\hid#vid_264a&pid_2260#8&39a8a845&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»  [0x264a, 0x2261]: \\?\hid#vid_264a&pid_2261#8&31eba1a0&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Quad
»        VendorId: 9802
»        ProductId: 8800
»        Version: 3.0.3
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 39, Rpm: 666, Unknown: 255]
»                        Identifier: [9802, 8800, 1]
»                2:
»                        Data: [PortId: 2, Speed: 39, Rpm: 649, Unknown: 255]
»                        Identifier: [9802, 8800, 2]
»                3:
»                        Data: [PortId: 3, Speed: 39, Rpm: 636, Unknown: 255]
»                        Identifier: [9802, 8800, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8800, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8800, 5]
»        Available effect types: PerLed
»  Name: Riing Quad
»        VendorId: 9802
»        ProductId: 8801
»        Version: 3.0.3
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 39, Rpm: 630, Unknown: 255]
»                        Identifier: [9802, 8801, 1]
»                2:
»                        Data: [PortId: 2, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8801, 2]
»                3:
»                        Data: [PortId: 3, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8801, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8801, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8801, 5]
»        Available effect types: PerLed
»
-------------------------------
Press any key to continue...
```

MoshiMoshi0, 2020-09-10:
Nice, seems that everything should work.
I'll close this, if you find any problems with quad fans please make a new issue.

Thank you both for help!

closed 2020-09-10 by MoshiMoshi0