MoshiMoshi0/TTController #63: Ports & PortConfig shortcuts or aliases

Christophoclese, 2020-07-28:
**Describe the feature you'd like**
Hi, first I'd like to say thank you for your work on this project. I'm doing my first configuration, so hopefully I don't end up with too many questions-- I see lots of examples out here in the existing issues.

On to the feature request-- would you consider adding the ability to use the Names for defined PortConfigs as an acceptable input to the Ports property? This would make it easy to declare groups of fans using PortConfigs, then manipulate those fans in groups elsewhere in the configuration.

To that same end, it would be great if there were a shortcut to specify all fans-- although I suppose I would just create my own PortConfig for that if names were supported.

Or maybe I'm just missing something? Cheers!

Christophoclese, 2020-07-28:
For example, this Ports declaration which references all fans:
```
{
  "ComputerStateProfiles": [
    {
      "StateType": "Boot",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ]
    }
  ],
}
```

Could be rewritten as something like:
```
{
  "ComputerStateProfiles": [
    {
      "StateType": "Boot",
      "Ports": "All"
    }
  ],
}
```

Or:
```
{
  "ComputerStateProfiles": [
    {
      "StateType": "Boot",
      "Ports": [
                     "Front",
                     "Top"
    }
  ],
}
```

MoshiMoshi0, 2020-07-28:
I can see that this would be easier to configure but you only really have to input the ports once for each profile and once for each PortConfig and this would require a pretty big change in code.

I could maybe add a special case for specifying all ports. Maybe like this:
```[9802, 8101, -1]```
But the issue is that the order of ports matters for effects. So maybe just allow specifying multiple ports for one PortIdentifier:
```[9802, 8101, 2, 3, 1, 4, 5]```

But I dont really know if this change is worth it considering you should not be reconfiguring the service frequently, its more of a configure once and forget thing.

Christophoclese, 2020-07-28:
If you were using a PortConfig-defined group, the order could be derived from the order the ports were defined there, but now I clearly see the issue with having a built-in 'All'. Depending on the effect chosen there's no way to provide any order and the user would have to define their own 'All' group anyways.

In my case I was considering using StaticColorEffect during computer state changes and wanted it applied to all lights, so the order wouldn't have mattered, but I didn't consider the other effects.

I agree there isn't a ton of value in the sense that you cannot accomplish what you need in other ways, but certainly could be a nice-to-have for organization purposes. I won't be offended if you can't see an easy way to make this work though!

Cheers

MoshiMoshi0, 2020-08-14:
I thought about it some more and there really isnt a nice way to implement this without changing a bit of code.

This would also have to work for plugins, but plugins dont have access to the config so they wont be able to know what "All" means, so i think the best way would be for the groups to be unpacked when reading the config by a custom json converter for `List<PortIdentifier>`. The problem is that the converter needs to know the groups when being instantiated, but that can happen only after reading the config, and reading the config requires the converter. So you would have to read the config twice?
And then it would also have to be explained in the docs. 

So right now I think that there are not enough benefits for this feature.

closed 2020-08-14 by MoshiMoshi0