MoshiMoshi0/TTController #64: Configuration Help - RazerConnect isn't working

BackJiletic, 2020-07-28:
I've managed to get everything working except the Razer Sync, when loading the config it sets all of the fans to the correct colour, but only uses 4/5 LEDs per fan? I've set the LED count and to Lerp/Nearest/Copy/Wrap the colour but none seem to resolve the issue.

Also it seems 4 fans on my second controller automatically change back to the default rainbow wave after a few seconds of loading the config? How can I stop this, I do not have the default TT RGB software installed on my PC.
```
These are the fans playing up;
[9802, 8502, 1]
[9802, 8502, 2]
[9802, 8502, 3]
[9802, 8502, 4]
```

Also, if possible would you be able to add support to sync the RGB on the TForce RAM like the crappy TT RGB app does?

Thanks in advance,
Jack

--------------------------------------------------------------------------------------------

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8501
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 49, Rpm: 770, Unknown: 255]
»                        Identifier: [9802, 8501, 1]
»                2:
»                        Data: [PortId: 2, Speed: 49, Rpm: 799, Unknown: 255]
»                        Identifier: [9802, 8501, 2]
»                3:
»                        Data: [PortId: 3, Speed: 49, Rpm: 760, Unknown: 255]
»                        Identifier: [9802, 8501, 3]
»                4:
»                        Data: [PortId: 4, Speed: 25, Rpm: 479, Unknown: 255]
»                        Identifier: [9802, 8501, 4]
»                5:
»                        Data: [PortId: 5, Speed: 25, Rpm: 470, Unknown: 255]
»                        Identifier: [9802, 8501, 5]
»        Available effect types: PerLed
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8502
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 25, Rpm: 496, Unknown: 255]
»                        Identifier: [9802, 8502, 1]
»                2:
»                        Data: [PortId: 2, Speed: 25, Rpm: 509, Unknown: 255]
»                        Identifier: [9802, 8502, 2]
»                3:
»                        Data: [PortId: 3, Speed: 25, Rpm: 526, Unknown: 255]
»                        Identifier: [9802, 8502, 3]
»                4:
»                        Data: [PortId: 4, Speed: 25, Rpm: 520, Unknown: 255]
»                        Identifier: [9802, 8502, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8502, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6793D:
»        Temperature:
»                CPU Core (/lpc/nct6793d/temperature/0): 48.0 °C
»                Temperature #1 (/lpc/nct6793d/temperature/1): 36.0 °C
»                Temperature #2 (/lpc/nct6793d/temperature/2): 28.0 °C
»                Temperature #4 (/lpc/nct6793d/temperature/4): 44.0 °C
»                Temperature #5 (/lpc/nct6793d/temperature/5): 14.0 °C
»                Temperature #6 (/lpc/nct6793d/temperature/6): 19.0 °C
»
»  Intel Core i7-6700K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 43.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 43.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 45.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 44.0 °C
»                CPU Package (/intelcpu/0/temperature/4): 45.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/5): 57.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/6): 57.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/7): 55.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/8): 56.0 °C
»                Core Max (/intelcpu/0/temperature/9): 45.0 °C
»                Core Average (/intelcpu/0/temperature/10): 43.8 °C
»
»  NVIDIA GeForce GTX 1080:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 45.0 °C
»
»  ST4000DM004-2CV104:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 36.0 °C
»
»  ADATA SX8200PNP:
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 33.0 °C
```

**Current configuration**
```
{
  "Profiles": [
  
	{
	
		"Name": "Case Fans",
		"Ports": [
		
			[9802, 8501, 4],
			[9802, 8501, 5],
			[9802, 8502, 1],
			[9802, 8502, 2],
			[9802, 8502, 3],
			[9802, 8502, 4],
			
		],
		
		"SpeedControllers": [
		
			{
			
				"Type": "StaticSpeedController",
				"Config": {
				
					"Speed": 25,
					"Trigger": { 
					
						"Type": "AlwaysTrigger",
					
					},
				
				},
			
			},
		
		],
		
		"Effects": [
		
			{
			
				"Type": "RazerConnectEffect",
				"Config": {
				
					"Trigger": { 
					
						"Type": "AlwaysTrigger",
					
					},
				
				},
			
			},
		
		],
		
		"PortConfigs": [
		
			{

				"Ports": [
				
					[9802, 8501, 4],
					[9802, 8501, 5],
					[9802, 8502, 1],
					
				],
				"Config": {
				
					"LedCount": 30,
					"LedCountHandling": "Lerp",
				
				},
			
			},
			
			{

				"Ports": [
				
					[9802, 8502, 2],
					[9802, 8502, 3],
					[9802, 8502, 4],
					
				],
				"Config": {
				
					"LedCount": 18,
					"LedCountHandling": "Lerp",
				
				},
			
			},
		 
		],
		
		"SensorConfigs": [
		
			{
			
			  "Sensors": ["/intelcpu/0/temperature/9"],
			  "Config": {
			  
				"CriticalValue": 90,
				
			  },
			  
			},
		],	
			
	
	},
  
	{
	
		"Name": "Radiators - CPU Temp",
		"Ports": [
		
			[9802, 8501, 1],
			[9802, 8501, 2],
			[9802, 8501, 3],
			
		],
		
		"SpeedControllers": [
		
			{
			
				"Type": "PwmSpeedController",
				"Config": {
				
					"CurvePoints": [
	
						[20, 20],
						[25, 25],
						[30, 25],
						[35, 30],
						[40, 45],
						[45, 50],
						[50, 75],
						[55, 85],
						[60, 90],
						[70, 100],
						[80, 100],
						[90, 100],
						[100, 100],
						[110, 100],
					
					],
					"Sensors": ["/intelcpu/0/temperature/9"],
					"Trigger": { 
					
						"Type": "AlwaysTrigger",
					
					},
				
				},
			
			},
		
		],
		
		"Effects": [
		
			{
			
				"Type": "RazerConnectEffect",
				"Config": {
				
					"Trigger": { 
					
						"Type": "AlwaysTrigger",
					
					},
				
				},
			
			},
		
		],
		
		"PortConfigs": [
		
			{

				"Ports": [
				
					[9802, 8501, 1],
					[9802, 8501, 2],
					[9802, 8501, 3],
					
				],
				"Config": {
				
					"LedCount": 30,
					"LedCountHandling": "Lerp",
				
				},
			
			},
		 
		],		
		
		"SensorConfigs": [
		
			{
			
			  "Sensors": ["/intelcpu/0/temperature/9"],
			  "Config": {
			  
				"CriticalValue": 90,
				
			  },
			  
			},
		],	
	
	},
  
  ],
  
  "CpuSensorsEnabled": true,
  
}
```

MoshiMoshi0, 2020-07-28:
Sensor and port configs are separate to the profiles, so they have to be moved outside of the profiles.
Also for a long time there is no `LedCount` variable, dont know how did you have it. I assumed that your 18 led fans are RiingDuo.

> Also, if possible would you be able to add support to sync the RGB on the TForce RAM like the crappy TT RGB app does?

Not for the foreseeable future, there isnt really a supported way to send colors to other applications, the service was meant to only receive them. It could be hacked in a custom effect plugin, but that is for you/someone else to write. I'll add that as a feature request.

> Also it seems 4 fans on my second controller automatically change back to the default rainbow wave

Does this happen only when the fan colors are static? Try forcing the razer synapse to send constantly changing colors and see if that helps.

```json
{
	"Profiles": [
		{
			"Name": "Case Fans",
			"Ports": [
				[9802, 8501, 4],
				[9802, 8501, 5],
				[9802, 8502, 1],
				[9802, 8502, 2],
				[9802, 8502, 3],
				[9802, 8502, 4]
			],
			"SpeedControllers": [
				{
					"Type": "StaticSpeedController",
					"Config": {
						"Speed": 25,
						"Trigger": { 
							"Type": "AlwaysTrigger"
						}
					}
				}
			],
			"Effects": [
				{
					"Type": "RazerConnectEffect",
					"Config": {
						"Trigger": { 
							"Type": "AlwaysTrigger"
						}
					}
				}
			]
		},
		{
			"Name": "Radiators - CPU Temp",
			"Ports": [
				[9802, 8501, 1],
				[9802, 8501, 2],
				[9802, 8501, 3]
			],
			"SpeedControllers": [
				{
					"Type": "PwmSpeedController",
					"Config": {
						"CurvePoints": [
							[20, 20],
							[25, 25],
							[30, 25],
							[35, 30],
							[40, 45],
							[45, 50],
							[50, 75],
							[55, 85],
							[60, 90],
							[70, 100]
						],
						"Sensors": ["/intelcpu/0/temperature/9"],
						"Trigger": { 
							"Type": "AlwaysTrigger"
						}
					}
				}
			],
			"Effects": [
				{
					"Type": "RazerConnectEffect",
					"Config": {
						"Trigger": { 
							"Type": "AlwaysTrigger"
						}
					}
				}
			]
		}
	],
	"PortConfigs": [
		{
			"Ports": [
				[9802, 8501, 1],
				[9802, 8501, 2],
				[9802, 8501, 3],
				[9802, 8501, 4],
				[9802, 8501, 5],
				[9802, 8502, 1]
			],
			"Config": {
				"DeviceType": "RiingTrio",
				"LedCountHandling": "Lerp"
			}
		},
		{
			"Ports": [
				[9802, 8502, 2],
				[9802, 8502, 3],
				[9802, 8502, 4]
			],
			"Config": {
				"LedCountHandling": "Lerp"
			}
		}
	],
	"SensorConfigs": [
		{
		  "Sensors": ["/intelcpu/0/temperature/9"],
		  "Config": {
			  "CriticalValue": 90
		  }
		}
	]
}
```

BackJiletic, 2020-07-29:
Hey, Ive input your fixed code and all the fans have full colouring, thankyou!
Also, it seems that yes, when its on static, the second controller will revert back to the rainbow wave. How can i go about fixing this? :)

MoshiMoshi0, 2020-07-29:
Update to latest development build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/build/artifacts
And add `"IgnoreColorCache": true` to PortConfigs:
```json
	"PortConfigs": [
		{
			"Ports": [
				[9802, 8501, 1],
				[9802, 8501, 2],
				[9802, 8501, 3],
				[9802, 8501, 4],
				[9802, 8501, 5]
			],
			"Config": {
				"DeviceType": "RiingTrio",
				"LedCountHandling": "Lerp"
			}
		},
		{
			"Ports": [
				[9802, 8502, 1]
			],
			"Config": {
				"DeviceType": "RiingTrio",
				"LedCountHandling": "Lerp",
				"IgnoreColorCache": true
			}
		},
		{
			"Ports": [
				[9802, 8502, 2],
				[9802, 8502, 3],
				[9802, 8502, 4]
			],
			"Config": {
				"DeviceType": "RiingDuo",
				"LedCountHandling": "Lerp",
				"IgnoreColorCache": true
			}
		}
	],
```

BackJiletic, 2020-07-30:
Hi there, this build seems to not work sadly. The controller software won't appear in Razer Synapse and when stopping the service it errors and crashes;

[image]

MoshiMoshi0, 2020-07-30:
What synapse version are you using? I think there was an update that broke razer connect because on the official forum they have a problem with it too.

BackJiletic, 2020-08-02:
Newest version - 3.5.630.61222

BackJiletic, 2020-08-03:
I've fixed the crashing, seemed to be a config issue sorry. However the service still isn't appearing in Synapse

MoshiMoshi0, 2020-08-03:
What was the issue in the config?

Like I said I think they broke razer connect with latest update because there are many reports of games and applications not working with razer connect.
I'll take a look but I think you just have to wait for the next update.

MoshiMoshi0, 2020-08-28:
Sorry for late reply.

There was apparently an update to razer synapse that should possibly fix this issue.
Please try Synapse versions 3.5.0730.072314 or above.

MoshiMoshi0, 2020-09-01:
Confirmed the effect working fine on version "3.5.0806.080617".
Closing.

closed 2020-09-01 by MoshiMoshi0