MoshiMoshi0/TTController #65: Configuration Help

Monratis, 2020-08-04:
**Describe your desired configuration**
Hey i have pure plus 12  3 of them using the stock controller 
I would like to have them in wave mode on fast speed with a fan curve
50%->60 C cpu
60%->70 C cpu 
100%->80C cpu 
**Hardware info**
`Info
-------------------------------
»
»        OS: Windows 10 Enterprise 2004 [19041.1.x86fre.vb_release.191206-1406]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 60, Rpm: 1038, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 60, Rpm: 1019, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 60, Rpm: 1045, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  ITE IT8686E:
»        Temperature:
»                System #1 (/lpc/it8686e/temperature/0): 41,0 °C
»                Chipset (/lpc/it8686e/temperature/1): 49,0 °C
»                CPU (/lpc/it8686e/temperature/2): 48,0 °C
»                PCIe x16 (/lpc/it8686e/temperature/3): 45,0 °C
»                VRM (/lpc/it8686e/temperature/4): 53,0 °C
»
»  ITE IT8792E/IT8795E:
»        Temperature:
»                PCIe x8 (/lpc/it879xe/temperature/0): 41,0 °C
»                System #2 (/lpc/it879xe/temperature/2): 40,0 °C
»
»  AMD Ryzen 7 3700X:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 58,0 °C
»                CCD1 (Tdie) (/amdcpu/0/temperature/3): 48,8 °C
»
»  NVIDIA GeForce RTX 2080:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 59,0 °C
»
»  Micron_1300_MTFDDAK1T0TDL:
»        Temperature:
»                Temperature (/ssd/0/temperature/0): 35,0 °C
»
»  SSDPR-CL100-480-G2:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 45,0 °C
»
»  GOODRAM:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 33,0 °C
»
»  ADATA SX6000NP:
»        Temperature:
»                Temperature (/nvme/4/temperature/0): 58,0 °C
»
»  XPG GAMMIX S11:
»        Temperature:
»                Temperature (/nvme/3/temperature/0): 34,0 °C
»`

MoshiMoshi0, 2020-08-05:
Dont know what colors you want for the wave effect, but you should be able to tweak them.

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [60, 50],
              [70, 60],
              [80, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "WaveEffect",
          "Config": {
            "TickInterval": 3,
            "Color": {
              "Gradient": [
                  [0.00, [255, 0, 0]],
                  [0.25, [128, 255, 0]],
                  [0.50, [0, 255, 0]],
                  [0.75, [0, 255, 128]],
                  [1.00, [0, 0, 255]]
              ]
            },
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 35,
      "Color": {
        "Full": [255, 0, 0]
      }
    }
  ]
}
```

Monratis, 2020-08-07:
Awesome worked 👍  i've tweaked it a bit  but its perfect !

closed 2020-08-07 by MoshiMoshi0