MoshiMoshi0/TTController #66: Trigger when display turns on or off

Echaleon, 2020-08-11:
First off, incredible piece of software. I absolutely love it.

If I can make one suggestion though, I think a trigger based on the display turning on or off -- or more likely, being on or off could be very useful. I don't let my desktop go to sleep, as I use it for server tasks and/or need to remote into it frequently from afar, so currently I just lock it and let the screens turn off at night. It would be incredibly useful to me at least to have it, as I could make it so the RGB in my desktop is only on when I'm sitting at it using the computer for example.

It may be possible, albeit a little hacky, to watch for a process? I'm not sure if there is a process that dies when the screens turn off/on.

MoshiMoshi0, 2020-08-11:
This is a good idea. Unfortunately I tried to implement this but I cant find a way to detect user idle time/screensaver from a service.
Unless someone knows a way to do this I think the only way is to write a small app that runs in user space and sends that info to the service via IPC. The IPC trigger would be implemented with #46

Echaleon, 2020-08-11:
Yeah. From my quick googling it seems services are barred form interacting with login sessions, and detecting idle state of the user falls under that umbrella. I found a couple things having the same issue, perhaps there's some insight there?

https://social.msdn.microsoft.com/Forums/vstudio/en-US/4cdb1251-da4b-4693-a58f-841c64c8f8bc/screen-saver-notifications?forum=netfxbcl

https://stackoverflow.com/questions/248186/service-needs-to-detect-if-workstation-is-locked-and-screen-saver-is-active

https://www.codeproject.com/Articles/13384/Getting-the-user-idle-time-with-C

https://stackoverflow.com/a/59400162

Don't know if any of these are helpful or not, cause, at best, it seems it'd have to be done as a workaround basically. 

A small program over IPC would work easy enough maybe. Could double as a small GUI for turning off the service, restarting, etc. perhaps. The other suggestion I saw googling was to change the program to run as a background task in task scheduler, but I'm not sure it's as easy to translate a service like that, or what sorts of issues you run into doing it that way. My experience writing C# or windows services is nil.

MoshiMoshi0, 2020-08-13:
Unfortunately I tried many things but I think there is no way to get user idle time, or even if the screensaver is running.
Like I said this would be doable with the IPC trigger.

PR is welcome if anyone manages to do this.

closed 2020-08-13 by MoshiMoshi0

MoshiMoshi0, 2020-09-20:
Just to update, I merged IPC feature in [this](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/35303393/artifacts) build. It should allow you to write a small app that checks for screensaver/display off events in user space and send it to the service using websockets. Should be documented on next release, for now you can check the code to figure out how to enable it and what data to send.

Echaleon, 2021-08-03:
So I know it has been while, but I ran across this project, and am actually using it to control my LG TV. From a Windows service, it;s able to detect idle state, and when the displays are turned off, though I am not sure how. I could dig into it further and figure out exactly how, though I am not much of a Windows programmer. 

https://github.com/JPersson77/LGTVCompanion

Food for thought perhaps?