MoshiMoshi0/TTController #67: Configuration Help

bofus10, 2020-08-13:
**Describe your desired configuration**
First of all I would like to thank you for this work you are doing, I think is simply amazing. So far I being able to set up the set curve based on GPU for 2 Fans and on CPU for 1 fan. But I cannot make any color work... I would like some wave or aurora effect, but right now all Im trying is to change the static color, but I simple can... do you mind letting me know what Im missing?  
 Also, I would like the fans off until X temperature, but they simply wont turn off.

Thanks!

**Hardware info**
```
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing
»        VendorId: 9802
»        ProductId: 8001
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 0, Rpm: 360, Unknown: 3]
»                        Identifier: [9802, 8001, 1]
»                2:
»                        Data: [PortId: 2, Speed: 0, Rpm: 361, Unknown: 3]
»                        Identifier: [9802, 8001, 2]
»                3:
»                        Data: [PortId: 3, Speed: 33, Rpm: 311, Unknown: 3]
»                        Identifier: [9802, 8001, 3]
»                4:
»                        Data:
»                        Identifier: [9802, 8001, 4]
»                5:
»                        Data:
»                        Identifier: [9802, 8001, 5]
»        Available effect types: Flow, Full
»
-------------------------------
Sensors
-------------------------------
»
»  ITE IT8686E:
»        Temperature:
»                Temperature #1 (/lpc/it8686e/temperature/0): 28,0 °C
»                Temperature #2 (/lpc/it8686e/temperature/1): 37,0 °C
»                Temperature #3 (/lpc/it8686e/temperature/2): 29,0 °C
»                Temperature #4 (/lpc/it8686e/temperature/3): 30,0 °C
»                Temperature #5 (/lpc/it8686e/temperature/4): 36,0 °C
»                Temperature #6 (/lpc/it8686e/temperature/5): 30,0 °C
»
»  Intel Core i5-8400:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 40,0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 38,0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 38,0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 43,0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 38,0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 38,0 °C
»                CPU Package (/intelcpu/0/temperature/6): 43,0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): 60,0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): 62,0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): 62,0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): 57,0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): 62,0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): 62,0 °C
»                Core Max (/intelcpu/0/temperature/13): 43,0 °C
»                Core Average (/intelcpu/0/temperature/14): 39,2 °C
»
»  NVIDIA GeForce RTX 2080 SUPER:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 33,0 °C
»
-------------------------------
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "GPU",
      "Ports": [
        [9802, 8001, 1],
        [9802, 8001, 2]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [40, 0],
              [50, 30],
              [65, 35],
              [75, 45],
              [85, 55],
              [100, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "StaticColorEffect",
          "Config": {
             "Color":{
               "Full": [0, 0, 255]
             }
          }
       }
      ]
    },
    {
      "Name": "CPU",
      "Ports": [
        [9802, 8001, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [29, 0],
              [30, 30],
              [40, 35],
              [75, 45],
              [85, 55],
              [100, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/14"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
    "Type": "FlowEffect",
    "Config": {
        "FillStep": 0.1,
        "HueStep": 60,
        "Saturation": 1.0,
        "Brightness": 1.0
    }
}
      ]
    }
  ],
  "PortConfigs": [
     {
      "Ports": [
      	[9802, 8001, 1],
        [9802, 8001, 2],
        [9802, 8001, 3]
      ],
      "Config": {
        "Name": "Top Fans",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [
        "/gpu/0/temperature/1",
        "/intelcpu/0/temperature/14"
      ],
      "Config": {
        "CriticalValue": 85
      }
    }
  ]
}

```

MoshiMoshi0, 2020-08-13:
You are missing the triggers in each effect config.
Like in the speed controllers:
```json
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
```

So:
```json
        {
          "Type": "StaticColorEffect",
          "Config": {
             "Color":{
               "Full": [0, 0, 255]
             },
             "Trigger": {
               "Type": "AlwaysTrigger"
             }
          }
       }
```

bofus10, 2020-08-13:
> You are missing the triggers in each effect config.
> Like in the speed controllers:
>
> ```json
>             "Trigger": {
>               "Type": "AlwaysTrigger"
>             }
> ```
>
> So:
>
> ```json [...]

Hey! Thanks for the reply! Yes I was missing, that and I updated the original post without realiing your reply... but still does not change any color :S (Code is updated on first message)

MoshiMoshi0, 2020-08-13:
Oh, sorry I did not notice that you have the old Riing fans.
That means you can only use the two color modes that are built in the controller.

Flow:
```json
{
    "Type": "RawEffect",
    "Config": {
        "EffectType": "Flow"
    }
}
```

Static:
```json
{
    "Type": "RawEffect",
    "Config": {
        "EffectType": "Full",
        "Color": {
          "Full": [255, 0, 0]
        }
    }
}
```

bofus10, 2020-08-13:
> Oh, sorry I did not notice that you have the old Riing fans.
> That means you can only use the two color modes that are built in the controller.
>
> Flow:
>
> ```json
> {
>     "Type": "RawEffect", [...]

Okay cool! Thanks a lot! On one side its working now, but on the other side I was fearing that, that the effects are not compatible... so there is no other effect that can be applied to this fans? something like fading? nothing?

MoshiMoshi0, 2020-08-13:
Yes the only thing you could do is fading, unfortunately I changed all effects to generate colors for each led since every fan controller after yours supports it. I'll check how much work has to be done to make your controller work.

bofus10, 2020-08-13:
> Yes the only thing you could do is fading, unfortunately I changed all effects to generate colors for each led since every fan controller after yours supports it. I'll check how much work has to be done to make your controller work.

Oh thats unfortunate :(  I actually thought I had bought the other ones with full supports :S  If you happen to be able to add that feature I would really appriciate it (Im trying to fix some paypal issues to support your work, I think is very impresive)

 One last question if you dont mind, dont really know how the hwlib works, is it possible to "input" data on the fan controllers? Im currently using Aida64 Sensor panel, and Im taking the RPM from the mother board fan pins, but with this fans/Controller I would lose total visibility of them... is there a way to re direct the rpm measure by the controler to any of the /lpc/it8686e/fan/X ?

MoshiMoshi0, 2020-08-13:
Ok, I tweaked it so the riing controller accepts PerLed color effects, but only uses the first color. This does not matter for the effects that generate the same color for each led, like PulseEffect. So is will work with all currently added effects but obviously it might not look like it should since it only uses one color.

These effects should work with decent results:
* AuroraEffect
* BlinkEffect
* ColorSequenceEffect
* CopyColorEffect
* FlowEffect
* PulseEffect
* RazerConnectEffect
* SensorEffect
* SoundEffect
* StaticColorEffect

Try this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/34645986/artifacts
With this config:
```json
{
  "Profiles": [
    {
      "Name": "Test",
      "Ports": [
        [9802, 8001, 1],
        [9802, 8001, 2],
        [9802, 8001, 3]
      ],
      "Effects": [
        {
          "Type": "PulseEffect",
          "Config": {
            "BrightnessStep": 0.025,
            "Color":{
              "Full": [0, 0, 255]
            },
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
     {
      "Ports": [
      	[9802, 8001, 1],
        [9802, 8001, 2],
        [9802, 8001, 3]
      ],
      "Config": {
        "DeviceType": "Riing"
      }
    }
  ]
}
```

> Oh thats unfortunate :( I actually thought I had bought the other ones with full supports :S

There is a possibility that I have not fully implemented all of the features of riing controller, but if you cant do that in thermaltake's official software then I unfortunately you would have to buy a new fans with newer controller. On the page about riing fans they also specify that it only has two modes.

> One last question if you dont mind, dont really know how the hwlib works, is it possible to "input" data on the fan controllers? Im currently using Aida64 Sensor panel, and Im taking the RPM from [...]

There is actually support for reading thermaltake fans in aida64, i think its somewhere in the options, the issue is you cant have more than one application connected to the controller as it will mess with the returned data.

I bet there is a way to expose custom sensors to aida, but this is currently out of scope for this project and would require to be hacked in to a custom plugin.
I think the proper way would be to implement #46 and allow two-way communication to external programs. That could be then passed to aida64 by a small program, or maybe directly read by aida64.

bofus10, 2020-08-13:
Simply amazing, pulse effect is working like a charm! Is there any effect like "radar" out of curiosity?
Mmm, aida also does not pick up any of the fans... Im wondering if it is because they are "old" model...

MoshiMoshi0, 2020-08-13:
> Is there any effect like "radar" out of curiosity?

Dont know what exactly do you mean by radar, like this? https://www.thermaltake.com/pub/media/wysiwyg/key3/video/RiingPlus/video6.webm

Like I said you only have one color available, so you cant do any moving effects, it has to be one color on all leds. 

> Mmm, aida also does not pick up any of the fans... Im wondering if it is because they are "old" model...

True, its possible they only added Riing Plus fans.

bofus10, 2020-08-13:
Exactly like that! Okay, yes I thought maybe doing that effect by turning on and off the LED.
 Thanks for all the effort!

closed 2020-08-13 by MoshiMoshi0