MoshiMoshi0/TTController #68: Service does not start after waking up from sleep

kounoupis, 2020-08-14:
**Describe the issue**
The service does not start automatically after the computer wakes up from sleep. It starts automatically upon initial boot and restart.

**Debug report**
```
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 32, Rpm: 571, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 32, Rpm: 580, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 32, Rpm: 575, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  ITE IT8688E:
»        Temperature:
»                Temperature #1 (/lpc/it8688e/temperature/0): 35.0 °C
»                Temperature #2 (/lpc/it8688e/temperature/1): 45.0 °C
»                Temperature #3 (/lpc/it8688e/temperature/2): 31.0 °C
»                Temperature #4 (/lpc/it8688e/temperature/3): 37.0 °C
»                Temperature #5 (/lpc/it8688e/temperature/4): 36.0 °C
»
»  ITE IT8792E/IT8795E:
»        Temperature:
»                Temperature #1 (/lpc/it879xe/temperature/0): 37.0 °C
»                Temperature #3 (/lpc/it879xe/temperature/2): 40.0 °C
»
»  Intel Core i9-9900KF:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 33.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 32.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 33.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 31.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 32.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 31.0 °C
»                CPU Core #7 (/intelcpu/0/temperature/6): 31.0 °C
»                CPU Core #8 (/intelcpu/0/temperature/7): 34.0 °C
»                CPU Package (/intelcpu/0/temperature/8): 34.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/9): 67.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/10): 68.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/11): 67.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/12): 69.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/13): 68.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/14): 69.0 °C
»                CPU Core #7 Distance to TjMax (/intelcpu/0/temperature/15): 69.0 °C
»                CPU Core #8 Distance to TjMax (/intelcpu/0/temperature/16): 66.0 °C
»                Core Max (/intelcpu/0/temperature/17): 34.0 °C
»                Core Average (/intelcpu/0/temperature/18): 32.1 °C
»
»  NVIDIA GeForce RTX 2080 SUPER:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 28.0 °C
»
»  INTEL SSDPEKKW010T7:
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 36.0 °C
»
»  Samsung SSD 970 EVO 500GB:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 51.0 °C
»                Temperature 1 (/nvme/1/temperature/6): 51.0 °C
»                Temperature 2 (/nvme/1/temperature/7): 57.0 °C
»
-------------------------------
```

**Current configuration**
```json
{
  "Profiles":
  [
    {
      "Name": "Default",
      "Ports":
      [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers":
      [
        {
          "Type": "PwmSpeedController", 
          "Config":
	  {
            "CurvePoints":
	    [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8"],
            "Trigger": { "Type": "AlwaysTrigger" }
          }
        }
      ],
      "Effects":
      [
	{
          "Type": "RippleEffect",
          "Config": {
            "Length": 10,
	    "TickInterval": 2,
            "RippleColor": { "Gradient": [[0, [16, 16, 128]], [0.8, [30, 16, 100]], [1.0, [128,16,128]]] },
	    "BackgroundColor": { "Full": [128, 16, 118] },
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        },
        {
          "Type": "SensorEffect",
          "Config":
	  {
            "Sensors": ["/intelcpu/0/temperature/8"],
            "ColorGradient":
	    [
              [40, [16, 16, 128]],
              [60, [128, 16, 128]],
              [86, [128, 16, 16]]
            ],
            "Trigger": { "Type": "AlwaysTrigger" }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles":
  [
    {
      "StateType":  "boot",
      "Ports":
      [
	[9802, 8101, 1],
	[9802, 8101, 2],
	[9802, 8101, 3]
      ],
      "EffectType": "Wave_Extreme",
      "Color": { "Full": [150, 150, 150] }
    }
  ],
  "PortConfigs":
  [
    {
      "Ports":
      [
  	[9802, 8101, 1],
  	[9802, 8101, 2],
  	[9802, 8101, 3]
      ],
      "Config":
      {
        "Name": "Radiator Fans",
	"LedCountHandling": "Lerp",
	"DeviceType": "RiingPlus"
      }
    }  
  ],
  "SensorConfigs": []
}
```

kounoupis, 2020-08-14:
Here is the error I see in the event viewer:

  | Failed in handling the  PowerEvent. The error that occurred was: System.NullReferenceException: Object  reference not set to an instance of an object. at  TTController.Service.Hardware.HidDeviceProxy..ctor(HidDevice device) at  TTController.Service.Managers.DeviceManager..ctor() at  TTController.Service.TTService.Initialize() at  TTController.Service.TTService.OnStart(String[] args) at  TTController.Service.TTService.OnPowerEvent(PowerBroadcastStatus powerStatus) at  System.ServiceProcess.ServiceBase.DeferredPowerEvent(Int32 eventType, IntPtr  eventData).
-- | --

MoshiMoshi0, 2020-08-14:
Could you provide the log file created in `Logs` folder the service crashes?

As a workaround for now you can go into windows services, find "Thermaltake Controller" service and edit the recovery options to restart the service after 1 minuet after a crash.

kounoupis, 2020-08-18:
Thank you for the prompt reply!

Below is a log.

The service does not crash after waking up. It simply does not start. So setting the recovery options (which is the first thing I tried) does not help. For now, I used task scheduler to start the service using net start, after a wake up event, so all is good!

```
2020-08-15 17:43:47.7539|INFO|TTController.Service.TTService|================================================================
2020-08-15 17:43:47.7838|INFO|TTController.Service.TTService|Initializing...
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgController [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingController [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingPlusController [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingTrioController [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AuroraEffect [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PingPongEffect [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.10.0.0]
2020-08-15 17:43:47.7940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.10.0.0]
2020-08-15 17:43:47.8138|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.10.0.0]
2020-08-15 17:43:47.8138|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RippleEffect [0.10.0.0]
2020-08-15 17:43:47.8138|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorEffect [0.10.0.0]
2020-08-15 17:43:47.8239|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SoundEffect [0.10.0.0]
2020-08-15 17:43:47.8239|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.10.0.0]
2020-08-15 17:43:47.8239|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.10.0.0]
2020-08-15 17:43:47.8239|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.WaveEffect [0.10.0.0]
2020-08-15 17:43:47.8239|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopySpeedController [0.10.0.0]
2020-08-15 17:43:47.8239|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.10.0.0]
2020-08-15 17:43:47.8239|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.10.0.0]
2020-08-15 17:43:47.8239|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.10.0.0]
2020-08-15 17:43:47.8437|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.10.0.0]
2020-08-15 17:43:47.8437|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.LogicTrigger [0.10.0.0]
2020-08-15 17:43:47.8437|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.10.0.0]
2020-08-15 17:43:47.8437|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.10.0.0]
2020-08-15 17:43:47.8437|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseTrigger [0.10.0.0]
2020-08-15 17:43:47.8437|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.10.0.0]
2020-08-15 17:43:47.8437|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorTrigger [0.10.0.0]
2020-08-15 17:43:47.8539|INFO|TTController.Service.Managers.ConfigManager|Creating Config Manager...
2020-08-15 17:43:48.0256|INFO|TTController.Service.Managers.ConfigManager|Loading config...
2020-08-15 17:43:48.1872|INFO|TTController.Service.Utils.DataCache|Creating DataCache...
2020-08-15 17:43:48.1872|INFO|TTController.Service.Utils.PluginStore|Creating Plugin Store...
2020-08-15 17:43:48.1872|INFO|TTController.Service.Managers.SensorManager|Creating Sensor Manager...
2020-08-15 17:43:48.1872|INFO|TTController.Service.Hardware.LibreHardwareMonitorFacade|Initializing Libre Hardware Monitor...
2020-08-15 17:43:48.5850|INFO|TTController.Service.Managers.DeviceManager|Creating Device Manager...
2020-08-15 17:43:48.6630|INFO|TTController.Service.Managers.DeviceManager|Initialized "Riing Plus" controller [9802, 8101]
[... cut ...]
2020-08-16 13:54:58.6858|INFO|TTController.Service.Hardware.LibreHardwareMonitorFacade|Initializing Libre Hardware Monitor...
2020-08-16 13:54:59.1803|INFO|TTController.Service.Managers.DeviceManager|Creating Device Manager...
2020-08-16 13:54:59.1803|FATAL|TTController.Service.TTService|System.NullReferenceException: Object reference not set to an instance of an object.
   at TTController.Service.Hardware.HidDeviceProxy..ctor(HidDevice device)
   at TTController.Service.Managers.DeviceManager..ctor()
   at TTController.Service.TTService.Initialize()
   at TTController.Service.TTService.OnStart(String[] args)
```

MoshiMoshi0, 2020-08-19:
It crashes as you can see at the bottom of the log. You probably didn't set the option to also restart the service if it crashes.
Please uninstall the current service and install this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/34729759/artifacts

The service now should retry after 10s/20s/60s after consecutive failures.
Please also post the log again even if it works, the service should now report a better error message if it crashes there.

I think what is happening is that the service starts too fast, while the controller is not yet running.

kounoupis, 2020-09-02:
Thank you. I uninstalled the old service and installed this build. I will let you know how it works.

Looking at the recovery options for the service, I had:
[image]
But that was still not enough.

kounoupis, 2020-09-02:
So far this is working great! The service restarted after a wake up.

MoshiMoshi0, 2020-09-02:
There is a checkbox just below those options from you screenshot that needs to be checked for the service to restart after a crash.
I made it so its automatically checked when installing.

Please post the latest log again after you wake from sleep. I would like to know the exception that is thrown.

kounoupis, 2020-09-02:
I see these errors in the event viewer, associated with this wake up:
```
Failed in handling the PowerEvent. The error that occurred was: HidSharp.Exceptions.DeviceIOException: Unable to open HID class device (\\?\hid#vid_264a&pid_1fa5#8&1441c7e3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}).
   at HidSharp.Platform.Windows.WinHidStream.Init(String path)
   at HidSharp.Platform.Windows.WinHidDevice.OpenDeviceDirectly(OpenConfiguration openConfig)
   at HidSharp.Device.OpenDeviceAndRestrictAccess(OpenConfiguration openConfig)
   at HidSharp.HidDevice.Open()
   at TTController.Service.Hardware.HidDeviceProxy..ctor(HidDevice device) in C:\projects\ttcontroller\Source\TTController.Service\Hardware\HidDeviceProxy.cs:line 26
   at TTController.Service.Managers.DeviceManager.SearchForControllers() in C:\projects\ttcontroller\Source\TTController.Service\Managers\DeviceManager.cs:line 75
   at TTController.Service.Managers.DeviceManager..ctor() in C:\projects\ttcontroller\Source\TTController.Service\Managers\DeviceManager.cs:line 33
   at TTController.Service.TTService.Initialize() in C:\projects\ttcontroller\Source\TTController.Service\TTSe....
```
```
The Thermaltake Controller service terminated with the following error: 
Incorrect function.
```
```
The Thermaltake Controller service terminated unexpectedly.  It has done this 1 time(s).  The following corrective action will be taken in 10000 milliseconds: Restart the service.
```

And here is the log with the exception:
```
2020-09-02 09:46:37.0726|INFO|TTController.Service.TTService|================================================================
2020-09-02 09:46:37.0956|INFO|TTController.Service.TTService|Initializing service, version "0.10.0-alpha.22+Branch.develop.Sha.75095cf22b196ab56ed16010ceaa2f992dbc791f"
2020-09-02 09:46:37.3579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgController [0.10.0.0]
2020-09-02 09:46:37.3579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.Level20DeskController [0.10.0.0]
2020-09-02 09:46:37.3579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingController [0.10.0.0]
2020-09-02 09:46:37.3579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingPlusController [0.10.0.0]
2020-09-02 09:46:37.3579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingQuadController [0.10.0.0]
2020-09-02 09:46:37.3579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingTrioController [0.10.0.0]
2020-09-02 09:46:37.3579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AuroraEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ColorSequenceEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PingPongEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RippleEffect [0.10.0.0]
2020-09-02 09:46:37.3698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorEffect [0.10.0.0]
2020-09-02 09:46:37.3928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SoundEffect [0.10.0.0]
2020-09-02 09:46:37.3928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.10.0.0]
2020-09-02 09:46:37.3928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.10.0.0]
2020-09-02 09:46:37.3928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.WaveEffect [0.10.0.0]
2020-09-02 09:46:37.3928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopySpeedController [0.10.0.0]
2020-09-02 09:46:37.3928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.10.0.0]
2020-09-02 09:46:37.4027|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.10.0.0]
2020-09-02 09:46:37.4027|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.10.0.0]
2020-09-02 09:46:37.4027|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.10.0.0]
[... cut ...]
   at TTController.Service.Managers.DeviceManager.SearchForControllers() in C:\projects\ttcontroller\Source\TTController.Service\Managers\DeviceManager.cs:line 75
   at TTController.Service.Managers.DeviceManager..ctor() in C:\projects\ttcontroller\Source\TTController.Service\Managers\DeviceManager.cs:line 33
   at TTController.Service.TTService.Initialize() in C:\projects\ttcontroller\Source\TTController.Service\TTService.cs:line 57
   at TTController.Service.TTService.OnStart(String[] args) in C:\projects\ttcontroller\Source\TTController.Service\TTService.cs:line 125
```

kounoupis, 2020-09-02:
Full screen shot of the service recovery tag:
[image]

MoshiMoshi0, 2020-09-02:
So the controller is detected but cannot be opened by HidSharp, I guess somehow windows loads faster than the controller. I tested and it does not happen on my machine.

I'll mark this issue as solved, this could be prevented in many other ways but just restating the service is simple and works.
Thanks for reporting!

closed 2020-09-02 by MoshiMoshi0

kounoupis, 2020-09-03:
Thanks for fixing!