MoshiMoshi0/TTController #69: Configuration Help

powellplayz, 2020-08-18:
i cannot figure out how to use this software it has no options just start stop and report?

Info
-------------------------------
»
»        OS: Windows 10 Home 1903 [18362.239.x86fre.19h1_release_svc_prod1.190628-1641]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»
-------------------------------
Sensors
-------------------------------
»
»  ITE IT8688E:
»        Temperature:
»                Temperature #1 (/lpc/it8688e/temperature/0): 38.0 °C
»                Temperature #2 (/lpc/it8688e/temperature/1): 34.0 °C
»                Temperature #3 (/lpc/it8688e/temperature/2): 32.0 °C
»                Temperature #4 (/lpc/it8688e/temperature/3): 36.0 °C
»                Temperature #5 (/lpc/it8688e/temperature/4): 32.0 °C
»
»  ITE IT8792E/IT8795E:
»        Temperature:
»                Temperature #1 (/lpc/it879xe/temperature/0): 41.0 °C
»                Temperature #3 (/lpc/it879xe/temperature/2): 35.0 °C
»
»  Intel Core i9-9900K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 38.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 34.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 42.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 35.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 33.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 34.0 °C
»                CPU Core #7 (/intelcpu/0/temperature/6): 33.0 °C
»                CPU Core #8 (/intelcpu/0/temperature/7): 36.0 °C
»                CPU Package (/intelcpu/0/temperature/8): 42.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/9): 62.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/10): 66.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/11): 58.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/12): 65.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/13): 67.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/14): 66.0 °C
»                CPU Core #7 Distance to TjMax (/intelcpu/0/temperature/15): 67.0 °C
»                CPU Core #8 Distance to TjMax (/intelcpu/0/temperature/16): 64.0 °C
»                Core Max (/intelcpu/0/temperature/17): 42.0 °C
»                Core Average (/intelcpu/0/temperature/18): 35.6 °C
»
»  NVIDIA GeForce RTX 2080 Ti:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 27.0 °C
»
»  GIGABYTE GP-ASM2NE6200TTTD:
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 34.0 °C
»
-------------------------------
Press any key to continue...

MoshiMoshi0, 2020-08-18:
Configuration is explained in the docs: https://moshimoshi0.github.io/TTController/
Also there are no detected controllers. What fans do you have? Is official thermaltake software working for you?

powellplayz, 2020-08-18:
> Hey mate,
>
> I have all tt stuff RGB wise, so i have ring plus fans and ring plus fittings and the RGB resivours.
>
> The thermaltake software worked once for about 5 mins, never again. I have tried one fan and one controller or all of them i just get the not detecting controllor error.
>
> Ive been told to buy a external hub so i did that still not working.
>
> Thought I'd try your software out.
>
> [Image]

MoshiMoshi0, 2020-08-18:
Please download this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/build/artifacts
And post the debug report again.

MoshiMoshi0, 2020-08-28:
Closing because of no response.
Reply here and I will reopen the issue.

closed 2020-08-28 by MoshiMoshi0