MoshiMoshi0/TTController #70: Razer Synapse update broke RazerConnectEffect

raiven60, 2020-09-03:
**Describe the issue**
Razyer Synapse and its components updates a few days ago and since then RazerConnectEffect seems to be not working properly. No errors on command line.

**Debug report**
```
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8102
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 65, Rpm: 1158, Unknown: 255]
»                        Identifier: [9802, 8102, 1]
»                2:
»                        Data: [PortId: 2, Speed: 65, Rpm: 1161, Unknown: 255]
»                        Identifier: [9802, 8102, 2]
»                3:
»                        Data: [PortId: 3, Speed: 65, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 65, Rpm: 1195, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 65, Rpm: 1154, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 65, Rpm: 1172, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 65, Rpm: 1129, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6776F:
»        Temperature:
»                CPU Core (/lpc/nct6776f/temperature/0): 51,5 °C
»                Temperature #2 (/lpc/nct6776f/temperature/2): 0,5 °C
»                Temperature #3 (/lpc/nct6776f/temperature/3): 55,0 °C
»
»  Intel Core i7-4930K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 38,0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 40,0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 39,0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 40,0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 40,0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 40,0 °C
»                CPU Package (/intelcpu/0/temperature/6): 41,0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): 52,0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): 50,0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): 51,0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): 50,0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): 50,0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): 50,0 °C
»                Core Max (/intelcpu/0/temperature/13): 40,0 °C
»                Core Average (/intelcpu/0/temperature/14): 39,5 °C
»
»  NVIDIA GeForce RTX 2070 SUPER:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 56,0 °C
»
»  ST2000DM001-1ER164:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 43,0 °C
»
»  Samsung SSD 850 EVO 500GB:
»        Temperature:
»                Temperature (/ssd/1/temperature/0): 46,0 °C
»
»  Samsung SSD 860 EVO 1TB:
»        Temperature:
»                Temperature (/ssd/0/temperature/0): 46,0 °C
»
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Case",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
		[9802, 8101, 4],
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [60, 30],
              [65, 50],
              [70, 80],
              [75, 90],
              [82, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [                              
        {
          "Type": "RazerConnectEffect",
          "Config": {            
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
	{
      "Name": "CPU",
      "Ports": [
		[9802, 8102, 1],
		[9802, 8102, 2],
		[9802, 8102, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 80],
              [80, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/13"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [                              
        {
          "Type": "RazerConnectEffect",
          "Config": {            
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
		[9802, 8101, 4],
		[9802, 8102, 1],
		[9802, 8102, 2],
		[9802, 8102, 3]
      ],
      "Speed": 65,
      "EffectType": "Full",
      "EffectColors": [
        [128, 0, 0]
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8101, 1],
				[9802, 8101, 2],
				[9802, 8101, 3],
				[9802, 8101, 4]],
      "Config": {
        "Name": "Case Fan",
		"LedCountHandling": "Copy" 
      }
    },
	{
      "Ports": [[9802, 8102, 1],
				[9802, 8102, 2],
				[9802, 8102, 3]],
      "Config": {
        "Name": "CPU Fan",
		"LedCountHandling": "Copy"
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/13"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

MoshiMoshi0, 2020-09-03:
Similar to #64, what version of synapse are you running?
I tested "3.5.0806.080617" and the effect works fine, if anything its a problem with razer that I cant fix. They broke many games/apps since last few updates.

treybert3, 2020-09-07:
I experienced a similar issue today, with Riing Plus as well. TTController was able to communicate just fine to the fans, but would not update the colors from Razer. I uninstalled and reinstalled everything, with countless reboots but still no luck. Finally, I completely powered off my machine and waited for the residual power to drain. After that razer -> ttcontroller started working again

MoshiMoshi0, 2020-09-10:
Like I said this is an issue with razers latest updates that im not able to fix. A lot of applications were having issues including overwatch and thermaltakes software.
I tested version "3.5.0806.080617" and the effect works correctly. So I would try to update to latest synapse version, or if that does not help maybe try clean reinstalling synapse.

Closing for now, if you start getting errors in the logs/console, please create a new issue.

closed 2020-09-10 by MoshiMoshi0