MoshiMoshi0/TTController #71: Configuration Help

mhomorody, 2020-09-09:
**Describe your desired configuration**
I have 3 TT intake fans that are always speeding up and slowing down even when idle. I believe they're too sensitive to the CPU temp so I wanted to setup a better curve. The CPU CCD1 (Tdie) on my Ryzen 3600 is usually around 40-43c but will spike often to about 48.5c. I also just usually use a static LED color for each of the 3 fans. I'll switch the color when I feel like it. I tried the config below but the LEDs didn't light up. Not sure if I got the fan curve right either since I couldn't monitor what the RPMs were during different uses. Thanks!

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 1903 [18362.239.x86fre.19h1_release_svc_prod1.190628-1641]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 40, Rpm: 757, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 40, Rpm: 755, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 40, Rpm: 726, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6797D:
»        Temperature:
»                CPU (PECI) (/lpc/nct6797d/temperature/1): 35.0 °C
»                System (/lpc/nct6797d/temperature/2): 31.0 °C
»                VR MOSS (/lpc/nct6797d/temperature/3): 42.0 °C
»                PCH (/lpc/nct6797d/temperature/5): 40.0 °C
»
»  AMD Ryzen 5 3600:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 48.4 °C
»                CCD1 (Tdie) (/amdcpu/0/temperature/3): 46.8 °C
»
»  NVIDIA GeForce GTX 1070:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 47.0 °C
»
»  ST3000DM008-2DM166:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 31.0 °C
»
»  Samsung SSD 850 EVO 500GB:
»        Temperature:
»                Temperature (/ssd/0/temperature/0): 28.0 °C
»
»  WDC WDS100T2B0C-00PXH0:
»        Temperature:
»                Temperature (/nvme/3/temperature/0): 38.0 °C
»
-------------------------------
```

**Current configuration**
```
{
  "Profiles": [
    {
      "Name": "SideIntake",
      "Ports": [
        [9802, 8101, 1],
		[9802, 8101, 2],
		[9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 30],
              [50, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/3"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "StaticColorEffect",
          "Config": {
            "Full": [98, 28, 135],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
		[9802, 8101, 2],
		[9802, 8101, 3]
      ],
      "Speed": 35,
      "Colors": {
        "Full": [255, 238, 0]
      }
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
		[9802, 8101, 1],
		[9802, 8101, 2],
		[9802, 8101, 3]
		],
      "Config": {
        "Name": "SideIntake"        
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/amdcpu/0/temperature/3"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

MoshiMoshi0, 2020-09-09:
You didnt correctly configure the effect:
```json
      "Effects": [
        {
          "Type": "StaticColorEffect",
          "Config": {
            "Color": {
              "Full": [98, 28, 135]
            },
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
```

You can see fan rpm when you run the service in console mode, but the service needs to be stopped for the option to be enabled.

mhomorody, 2020-09-09:
Ah great thanks for the quick reply. I thought I might be missing that
"Color" But the Static Color Effect page didn't have an example. I'll try
this out soon.

MoshiMoshi0, 2020-09-15:
Im assuming everything is working, closing.

closed 2020-09-15 by MoshiMoshi0

mhomorody, 2020-09-15:
Yeah working great as far as I can tell.