MoshiMoshi0/TTController #72: Configuration Help

phraxtal, 2020-09-12:
**Describe your desired configuration**

I have two Pure Plus 12 fans connected to an official controller, they work on the TTRGB Plus software, which I despise.
All I want to do is setup a config file that let's me change a static fan speed to any percentage value by editing the config.json, I also want to change the effects at well by editing the config file at any time. Reading the docs and trying to setup the config file correctly has alluded me. Here's all the info I currently have:

**Hardware info**
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 2004 [19041.1.x86fre.vb_release.191206-1406]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 60, Rpm: 1044, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 60, Rpm: 1022, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  ITE IT8688E:
»        Temperature:
»                Temperature #1 (/lpc/it8688e/temperature/0): 42.0 °C
»                Temperature #2 (/lpc/it8688e/temperature/1): 50.0 °C
»                Temperature #3 (/lpc/it8688e/temperature/2): 43.0 °C
»                Temperature #4 (/lpc/it8688e/temperature/3): 53.0 °C
»                Temperature #5 (/lpc/it8688e/temperature/4): 46.0 °C
»
»  ITE IT8792E/IT8795E:
»        Temperature:
»                Temperature #1 (/lpc/it879xe/temperature/0): 46.0 °C
»                Temperature #3 (/lpc/it879xe/temperature/2): 38.0 °C
»
»  Intel Core i7-9700K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 49.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 49.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 49.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 46.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 48.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 47.0 °C
»                CPU Core #7 (/intelcpu/0/temperature/6): 47.0 °C
»                CPU Core #8 (/intelcpu/0/temperature/7): 51.0 °C
»                CPU Package (/intelcpu/0/temperature/8): 51.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/9): 51.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/10): 51.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/11): 51.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/12): 54.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/13): 52.0 °C
[... 88 lines cut ...]
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.DeserializeConvertable(JsonConverter converter, JsonReader reader, Type objectType, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   at TTController.Service.Managers.ConfigManager.LoadOrCreateConfig()
00:47:30.5820: Failed to load the config!

_________________________

What do I need to do to setup the config correctly to allow for easy static speed and effect changes?

MoshiMoshi0, 2020-09-12:
I think the error is pretty self explanatory and the docs have a few examples so you should be able to fix your config as you were pretty close.

```json
{
  "Profiles": [
    {
      "Name": "Zonic",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController",
          "Config": {
            "Speed": 77,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Saturation": 1,
            "Brightness": 1,
            "HueStep": 1,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ]
}
```

Note that each time you edit the `config.json` file you have to restart the service.

phraxtal, 2020-09-16:
thank you very much for this and your continued support :)

closed 2020-09-16 by phraxtal