MoshiMoshi0/TTController #73: Configuration Help

digitalhardcore1985, 2020-09-15:
**Describe your desired configuration**
I have nine TT Riing Trio 120 fans connected to two controllers. The first five fans are connected to the first the first controller with the remaining four connected to the second. I'd like to just have the inner ring of the fans set to a pale blue with both the outer rings switched off. I can only seem to get the top ring to illuminate even when trying to send a colour value to all 30 LEDs on each fan (please see my attempted config below). Thanks so much for this software and thanks in advance for any help, the official app is causing massive frame timing spikes in Steam VR every few seconds and has become unusable for me.

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 1909 [18362.1.wow64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8501
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 40, Rpm: 529, Unknown: 255]
»                        Identifier: [9802, 8501, 1]
»                2:
»                        Data: [PortId: 2, Speed: 40, Rpm: 523, Unknown: 255]
»                        Identifier: [9802, 8501, 2]
»                3:
»                        Data: [PortId: 3, Speed: 40, Rpm: 582, Unknown: 255]
»                        Identifier: [9802, 8501, 3]
»                4:
»                        Data: [PortId: 4, Speed: 40, Rpm: 599, Unknown: 255]
»                        Identifier: [9802, 8501, 4]
»                5:
»                        Data: [PortId: 5, Speed: 40, Rpm: 587, Unknown: 255]
»                        Identifier: [9802, 8501, 5]
»        Available effect types: PerLed
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8502
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 40, Rpm: 597, Unknown: 255]
»                        Identifier: [9802, 8502, 1]
»                2:
»                        Data: [PortId: 2, Speed: 40, Rpm: 579, Unknown: 255]
»                        Identifier: [9802, 8502, 2]
»                3:
»                        Data: [PortId: 3, Speed: 40, Rpm: 558, Unknown: 255]
»                        Identifier: [9802, 8502, 3]
»                4:
»                        Data: [PortId: 4, Speed: 40, Rpm: 554, Unknown: 255]
»                        Identifier: [9802, 8502, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8502, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 38.5 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 34.0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 14.0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 12.0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 47.0 °C
»
»  Intel Core i9-9900K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 42.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 40.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 40.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 41.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 39.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 40.0 °C
»                CPU Core #7 (/intelcpu/0/temperature/6): 38.0 °C
»                CPU Core #8 (/intelcpu/0/temperature/7): 49.0 °C
»                CPU Package (/intelcpu/0/temperature/8): 49.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/9): 58.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/10): 60.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/11): 60.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/12): 59.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/13): 61.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/14): 60.0 °C
»                CPU Core #7 Distance to TjMax (/intelcpu/0/temperature/15): 62.0 °C
»                CPU Core #8 Distance to TjMax (/intelcpu/0/temperature/16): 51.0 °C
»                Core Max (/intelcpu/0/temperature/17): 49.0 °C
»                Core Average (/intelcpu/0/temperature/18): 41.1 °C
»
»  NVIDIA GeForce GTX 1080 Ti:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 53.0 °C
»
»  SanDisk SDSSDH3 1T00:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 28.0 °C
»
»  Samsung SSD 970 EVO Plus 1TB:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 50.0 °C
»                Temperature 1 (/nvme/1/temperature/6): 50.0 °C
»                Temperature 2 (/nvme/1/temperature/7): 55.0 °C
»
-------------------------------
```

**Current configuration**
```{
  "Profiles": [{
  "Name": "Default",
  "Ports": [
    [9802, 8501, 1],
	[9802, 8501, 2],
	[9802, 8501, 3],
	[9802, 8501, 4],
	[9802, 8501, 5],
	[9802, 8502, 1],
	[9802, 8502, 2],
	[9802, 8502, 3],
	[9802, 8502, 4]
  ],
  "SpeedControllers": [
    {
      "Type": "PwmSpeedController", 
      "Config": {
        "CurvePoints": [
          [30, 30],
          [45, 50],
          [55, 60],
          [65, 75],
          [75, 100]
        ],
        "Sensors": ["/intelcpu/0/temperature/2"],
        "Trigger": {
          "Type": "AlwaysTrigger"
        }
      }
    }
  ],
  "Effects": [
    {
      "Type": "StaticColorEffect",
      "Config": {
        "Color": {
          "PerLed": [
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161],
				[5, 114, 161]
			]	
        },

        "Trigger": {
          "Type": "AlwaysTrigger"
        }
      }
    }
  ]
}],
  "ComputerStateProfiles": [],
  "PortConfigs": [],
  "SensorConfigs": []
}
```

MoshiMoshi0, 2020-09-15:
You have to set the [DeviceType](https://moshimoshi0.github.io/TTController/config/port#devicetype) for the ports, otherwise its using the default device with only 12 leds.
```json
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4]
      ],
      "Config": {
        "DeviceType": "RiingTrio",
      }
    }
  ],
```

Inner ring should be last 6 colors so just set everything else to `[0, 0, 0]`

digitalhardcore1985, 2020-09-16:
That did it, thanks so much!

closed 2020-09-16 by digitalhardcore1985