MoshiMoshi0/TTController #74: Unable to find controller

Sceptorrh, 2020-09-19:
**Describe the issue**
Unable to find my Pure Plus 12 controller, I am trying to configure the software. I however am unable to find my attached controller. It works fine using to original software and I don't have it running at the same time. I have tried restarting/replugging. But nothing seems to help.

**Debug report**
10:05:17.3469: Found plugin assembly for type IControllerDefinition: TTController.Plugin.DpsgController [0.10.0.0]
10:05:17.3469: Loading plugin assembly: TTController.Plugin.DpsgController [0.10.0.0]
10:05:17.3469: Found plugin assembly for type IControllerDefinition: TTController.Plugin.RiingController [0.10.0.0]
10:05:17.3469: Loading plugin assembly: TTController.Plugin.RiingController [0.10.0.0]
10:05:17.3469: Found plugin assembly for type IControllerDefinition: TTController.Plugin.RiingPlusController [0.10.0.0]
10:05:17.3469: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.10.0.0]
10:05:17.3609: Found plugin assembly for type IControllerDefinition: TTController.Plugin.RiingTrioController [0.10.0.0]
10:05:17.3609: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.10.0.0]
10:05:17.3609: Creating Device Manager...
10:05:17.3769: Searching for "Dpsg" controllers
10:05:17.3769: Searching for "Riing" controllers
10:05:17.3769: Searching for "Riing Plus" controllers
10:05:17.3769: Searching for "Riing Trio" controllers
10:05:17.3769: Disposing Device Manager...
10:05:17.3769: Disposed devices: 0

**Current configuration**
3x Pure Plus 12 fans with 1 controller

MoshiMoshi0, 2020-09-19:
Try latest develop build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/build/artifacts
And post output of `Debug -> Report` output.

MoshiMoshi0, 2020-09-23:
Did that build solve your issue?

LordPotaynut, 2020-09-24:
Hello, I am also having the same problem. TT RGB Plus software can find the controller, but TTController isn't seeing any. I attempted to install the latest developer build and it still can't find a controller.

(whenever I try to copy the text from the console it's massive and I can't figure out why, so I'm attaching a .txt)

[report.txt](https://github.com/MoshiMoshi0/TTController/files/5278691/report.txt)

MoshiMoshi0, 2020-09-24:
@LordPotaynut that report is from a build that is from a master branch, you must have run the wrong exe

LordPotaynut, 2020-09-24:
You were right, I accidentally reinstalled the master branch one. Running the one you linked to worked perfectly, the controller is detected. Thank you!

MoshiMoshi0, 2020-09-25:
@Sceptorrh Im assuming that build will work for you too, so I will close this issue.
If it still does not work comment here or create a new one.

closed 2020-09-25 by MoshiMoshi0