MoshiMoshi0/TTController #77: TTController Crashing, Requiring Manual Reinstall

LordPotaynut, 2020-09-29:
**Describe the issue**
Every time I boot up my computer, the TTController isn't working. I run the program to see what it is, and it crashes either when stopping, or when I run in-console. Every time this happens I need to delete all the files in the folder and unzip the build, and reinstall. However, there's one file that won't delete, `LibreHardwareMonitorLib.sys`. In order to delete it, I have to rename it to a .txt, reboot the computer, and then delete it. Otherwise it always says that it's running. I have the crash report here:
[image]
I've also noticed that the controller stops being detected upon reboot. I think this might be part of, if not all of the cause. However, when I reinstall it detects it just fine again.

**Debug report**
```
Info
-------------------------------
»
»        OS: Windows 10 Home 2004 [19041.1.x86fre.vb_release.191206-1406]
»        Build: 0.10.0-alpha.30+Branch.develop.Sha.4e5958c3d773d03d6565bdf4558213ca6cba58ab
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»
-------------------------------
Controllers
-------------------------------
»
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 29.0 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 29.0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 14.0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 9.0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 49.0 °C
»
»  Intel Core i7-8700:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 32.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 34.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 33.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 31.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 32.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 35.0 °C
»                CPU Package (/intelcpu/0/temperature/6): 35.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): 68.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): 66.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): 67.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): 69.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): 68.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): 65.0 °C
»                Core Max (/intelcpu/0/temperature/13): 35.0 °C
»                Core Average (/intelcpu/0/temperature/14): 32.8 °C
»
»  NVIDIA GeForce RTX 2080 SUPER:
»        Temperature:
»                GPU Core (/gpu-nvidia/0/temperature/0): 34.0 °C
»
»  Force MP510:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 36.0 °C
»
»  WDC WD20EZAZ-00GGJB0:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 24.0 °C
»
-------------------------------
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Front Intake",
      "Ports": [
        [9802, 8800, 1],
        [9802, 8800, 2]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 20],
              [45, 40],
              [55, 60],
              [65, 75],
              [70, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/14"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/14"],
            "ColorGradient": [
              [20, [255, 1, 1]],
              [40, [255, 1, 1]],
              [56, [1, 255, 48]],
              [72, [1, 255, 140]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Rear Exhaust",
      "Ports": [
        [9802, 8800, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 20],
              [55, 60],
              [75, 100]
            ],
            "Sensors": ["/gpu-nvidia/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/gpu-nvidia/0/temperature/0"],
            "ColorGradient": [
              [20, [255, 255, 1]],
              [40, [255, 255, 1]],
              [60, [128, 255, 1]],
              [75, [1, 255, 1]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfigs": [
  {
    "Ports": [
      [9802, 8800, 1],
      [9802, 8800, 2],
      [9802, 8800, 3]
    ],
    "Config": {
      "DeviceType": "RiingQuad",
      "IgnoreColorCache": true
  }
} 
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/14"],
      "Config": {
        "CriticalValue": 90
      }
    },
    {
      "Sensors": ["/gpu-nvidia/0/temperature/0"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ],

  "CpuSensorsEnabled": true,
  "GpuSensorsEnabled": true,
  "StorageSensorsEnabled": false,
  "MotherboardSensorsEnabled": true,
  "MemorySensorsEnabled": true,
  "NetworkSensorsEnabled": false,
  "ControllerSensorsEnabled": true,

  "SensorTimerInterval": 500,
  "DeviceSpeedTimerInterval":1000,
  "DeviceRgbTimerInterval": 32,
  "DebugTimerInterval": 2500
}
```

MoshiMoshi0, 2020-09-29:
Yes it seems the service is starting before the controller and there was a small bug (1 character :roll_eyes:).
This build should fix the crash: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/35474076/artifacts

`LibreHardwareMonitorLib.sys` will be stubborn to delete because it is a dynamic driver generated by LibreHardwareMonitor for reading sensors. After restart it will delete fine assuming the service did not start/load it again.

LordPotaynut, 2020-09-30:
When I first installed the software, it worked perfectly fine. But now it seems to be doing the same things, except this time loading the plugins and sensors takes much longer than it used to. No different configuration.

Debug Report:
`Info
-------------------------------
»
»        OS: Windows 10 Home 2004 [19041.1.x86fre.vb_release.191206-1406]
»        Build: 0.10.0-alpha.31+Branch.develop.Sha.b3e259b0e504b90dfc825dfe7b0016f923a10647
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»
-------------------------------
Controllers
-------------------------------
»
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 32.0 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 32.0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 14.0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 9.0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 49.0 °C
»
»  Intel Core i7-8700:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 33.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 33.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 33.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 35.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 33.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 35.0 °C
»                CPU Package (/intelcpu/0/temperature/6): 35.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): 67.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): 67.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): 67.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): 65.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): 67.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): 65.0 °C
»                Core Max (/intelcpu/0/temperature/13): 35.0 °C
»                Core Average (/intelcpu/0/temperature/14): 33.7 °C
»
»  NVIDIA GeForce RTX 2080 SUPER:
»        Temperature:
»                GPU Core (/gpu-nvidia/0/temperature/0): 37.0 °C
»
»  Force MP510:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 41.0 °C
»
»  WDC WD20EZAZ-00GGJB0:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 30.0 °C
»
-------------------------------`

MoshiMoshi0, 2020-09-30:
Can't tell you what is happening but when you made that debug report the controller was completely missing, like its unplugged (it should be present in HID section). So unless for some reason the controller froze, there might be problem with connection to the motherboard.

Did you have problems running thermaltake software?
Can you find steps to reproduce this issue easily? Like for example does it only happen after some time or only when you wake from sleep.

About long sensor scanning its because you have enabled other sensors that you are not using, they are taking time to scan.
Set:
```
  "MotherboardSensorsEnabled"
  "MemorySensorsEnabled"
  "ControllerSensorsEnabled"
```
to `false` instead of `true`.

MoshiMoshi0, 2020-10-07:
Do you still have this issue?

LordPotaynut, 2020-10-07:
Yes, sorry. I am still having the controller not being detected, but all I have to do is just restart my computer and hope that it detects. It's not glamorous but as long as the fans are quiet and the fan curve is proper.

MoshiMoshi0, 2020-10-07:
If the service does not detect any controllers, does restarting the service fix it? You can restart it from the console menu.
Also please post latest logs file (from the "Logs" folder) when the service fails to detect the controllers.

LordPotaynut, 2020-10-12:
Alright, here is "current.log".

[current.log](https://github.com/MoshiMoshi0/TTController/files/5362899/current.log)

MoshiMoshi0, 2020-10-12:
When you boot the pc and the service does not detect any controllers, what happens if you restart the service instead of restarting the pc?

LordPotaynut, 2020-10-12:
Restarting does nothing at all. Even though my controller is plugged in every single restart, it just doesn't detect it sometimes, so restarting doesn't help the program see the fans. IDK if it's a cable issue though, because when it works it works flawlessly. But it might just be Windows not loading in the USB device fast enough, I'm not sure.

MoshiMoshi0, 2020-10-12:
Im talking about restarting the service, not rebooting the PC. 
If the service does not detect the controller run the service exe and go to `Manage Service -> Restart`, dont restart the pc.

If that does not help then there must be something wrong with the cable or motherboard.

LordPotaynut, 2020-10-13:
Sorry, the way I worded it was wrong. I was restarting the program from within the Manage Service menu. I tried both using the Restart button and the Stop/Start buttons and they don't work. It's only when I fully restart my PC that it might detect the controller.

MoshiMoshi0, 2020-10-13:
Then sorry but I dont think I can do anything about this. Either the controller is bad or the connection. Maybe try a different usb port.

closed 2020-10-13 by MoshiMoshi0