MoshiMoshi0/TTController #81: Configuration Help

cwills75, 2020-11-03:
**Describe your desired configuration**
Hi, I'm having issues configuring this, any help would be appreciated.  I have a Floe DX 360 Triple Riing Duo cooler with 1 controller.  3 radiator fans connected to the controller, along with pump and a 140mm Riing Duo fan.  I'm trying to get the pump running at 100%, the 3 radiator fans running on PWM depending on max CPU core temp and the 140mm fan running at a static 40% speed.  Also, I don't 100% understand the docs in regards to the ignorecolorcache.  I set it to true, but I'm not sure if that is correct.  Ideally, I would like to set the color profile and have it persist at all times, such as when booting, when the service isn't running, etc.  I'm not sure if that is possible.  Right now with the stock software, whenever I exit it, it defaults back to the rainbow colors.  When trying to start the service, I get:

"Failed to start the service...
System.ServiceProcess.TimeoutException:  Time out has expired and the operation has not been completed.
  at System.ServiceProcess.ServiceController.WaitforStatus(ServiceControllerStatus desired Status, TimeSpan timeout)
  at TTController.Service.Program.,ManageServiceMenu>g__StartService|6_1()

**Hardware info**
```
Info
-----------------------------
OS: Windows 10 Enterprise 2004 [19041.1.x86fre.vb_release.191206-1406]
Build: 0.11.0-beta+Branch.master.Sha.2b6096e1aeefc2bc0b2e3e715c364bcd849d9ddc
-----------------------------
Applications
-----------------------------
HID
-----------------------------
[0x264a, 0x2135]: \\?\hid#vid_264a$pid_2135#6&25841900%0%0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
-----------------------------
Controllers
-----------------------------
Name: Riing Trio
VendorID: 9802
ProductId: 8501
Version 3.1.1
Ports:
1.
Data: [PortID: 1, Speed: 100, Rpm: 0, Unknown: 255]
Identifier: [9802, 8501, 1]
2.
Data: [PortID: 2, Speed: 100, Rpm: 1661, Unknown: 255]
Identifier: [9802, 8501, 2]
3.
Data: [PortID: 3, Speed: 100, Rpm: 1637, Unknown: 255]
Identifier: [9802, 8501, 3]
4.
Data: [PortID: 4, Speed: 100, Rpm: 1630, Unknown: 255]
Identifier: [9802, 8501, 4]
5.
Data: [PortID: 5, Speed: 58, Rpm: 928, Unknown: 255]
Identifier: [9802, 8501, 5]
Available effect types: PerLed
-----------------------------
Sensors
-----------------------------
ITE IT8686E:
Temperature:
Temperature #1 (/lpc/it8686e/temperature/0): 34.0 C
Temperature #2 (/lpc/it8686e/temperature/1): 44.0 C
Temperature #3 (/lpc/it8686e/temperature/2): 19.0 C
Temperature #4 (/lpc/it8686e/temperature/3): 31.0 C
Temperature #5 (/lpc/it8686e/temperature/4): 36.0 C
Temperature #6 (/lpc/it8686e/temperature/5): 44.0 C

Intel Core i9-9900K:
Temperature:
CPU Core #1 (/intelcpu/0/temperature/0): 27.0 C
CPU Core #2 (/intelcpu/0/temperature/1): 27.0 C
CPU Core #3 (/intelcpu/0/temperature/2): 26.0 C
CPU Core #4 (/intelcpu/0/temperature/3): 27.0 C
CPU Core #5 (/intelcpu/0/temperature/4): 26.0 C
CPU Core #6 (/intelcpu/0/temperature/5): 28.0 C
CPU Core #7 (/intelcpu/0/temperature/6): 26.0 C
CPU Core #8 (/intelcpu/0/temperature/7): 29.0 C
CPU Package (/intelcpu/0/temperature/8): 29.0 C
CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/9): 63.0 C
CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/10): 63.0 C
CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/11): 64.0 C
CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/12): 63.0 C
CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/13): 64.0 C
CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/14): 62.0 C
CPU Core #7 Distance to TjMax (/intelcpu/0/temperature/15): 64.0 C
CPU Core #8 Distance to TjMax (/intelcpu/0/temperature/16): 61.0 C
Core Max (/intelcpu/0/temperature/17): 29.0 C
Core Average (/intelcpu/0/temperature/18): 27.0 C

Nvidia GeForce RTX 3080:
Temperature:
GPU Core (/gpu-nvidia/0/temperature/0): 38.0 C

Samsung SSD 840 EVO 250GB:
Temperature:
Temperature (/ssd/2/temperature/0): 24.0 C

WDC WD30EFRX-68EUZN0:
Temperature:
Temperature (/hdd/3/temperature/0): 30.0 C

WDC WD40EZRZ-00WN9B0:
Temperature:
Temperature (/hdd/1/temperature/0): 31.0 C

WDC WD30EFRX-68EUZN0:
Temperature:
Temperature (/hdd/5/temperature/0): 28.0 C

Samsung SSD 960 EVO 250GB:
Temperature:
Temperature (/nvme/0/temperature/0): 33.0 C
Temperature 1 (/nvme/0/temperature/6) 33.0 C
Temperature 2 (/nvme/0/temperature/7) 53.0 C

WDC WD30EFRX-68N32N0:
Temperature:
Temperature (/hdd/4/temperature/0): 28.0 C

-----------------------------
Plugins
-----------------------------
Detected plugins:
TTController.Plugin.DpsgController.dll
TTController.Plugin.Level20DeskController.dll
TTController.Plugin.RiingController.dll
TTController.Plugin.RiingPlusController.dll
TTController.Plugin.RiingQuadController.dll
TTController.Plugin.RiingTrioController.dll
TTController.Plugin.AuroraEffect.dll
TTController.Plugin.BlinkEffect.dll
TTController.Plugin.ColorSequenceEffect.dll
TTController.Plugin.CopyColorEffect.dll
TTController.Plugin.FlowEffect.dll
TTController.Plugin.IpcEffect.dll
TTController.Plugin.PingPongEffect.dll
TTController.Plugin.PulseEffect.dll
TTController.Plugin.RawEffect.dll
TTController.Plugin.RazerConnectEffect.dll
TTController.Plugin.RippleEffect.dll
TTController.Plugin.SensorEffect.dll
TTController.Plugin.SoundEffect.dll
TTController.Plugin.SpectrumEffect.dll
TTController.Plugin.StaticColorEffect.dll
TTController.Plugin.WaveEffect.dll
TTController.Plugin.ZoneEffect.dll
TTController.Plugin.LerpLedColorModifier.dll
TTController.Plugin.ReverseLedColorModifier.dll
TTController.Plugin.RotateLedColorModifier.dll
TTController.Plugin.TrimLedColorModifier.dll
TTController.Plugin.CopySpeedController.dll
TTController.Plugin.IpcSpeedController.dll
TTController.Plugin.PwmSpeedController.dll
TTController.Plugin.StaticSpeedController.dll
TTController.Plugin.AlwaysTrigger.dll
TTController.Plugin.IpcTrigger.dll
TTController.Plugin.LogicTrigger.dll
TTController.Plugin.OneTimeTrigger.dll
TTController.Plugin.ProcessTrigger.dll
TTController.Plugin.PulseTrigger.dll
TTController.Plugin.ScheduleTrigger.dll
TTController.Plugin.SensorTrigger.dll
TTController.Plugin.WebSocketIpcServer.dll
```

**Current configuration**
{
  "Profiles": [
	{
		"Name": Pump",
		"Ports": [
			[9802, 8501, 1]
		],
		"SpeedControllers": [
			{
			"Type": "StaticSpeedController",
				"Config": {
				"Speed": 100,
				"Trigger": {
					"Type": "AlwaysTrigger"
					}
				}
			}
		],
		"Effects": [
			{
			"Type": "SpectrumEffect",
				"Config": {
					"Saturation": 1,
					"Brightness": 1,
					"HueStep": 1,
					"Trigger": {
						"Type": AlwaysTrigger"
						}
				}
			}
		]
	}
		{
		"Name": Radiator",
		"Ports": [
			[9802, 8501, 2],
			[9802, 8501, 3],
			[9802, 8501, 4]
		],
		"SpeedControllers": [
			{
			"Type": "PwmSpeedController",
				"Config": {
					"CurvePoints": [
						[30, 30],
						[45, 50],
						[55, 60],
						[65, 75],
						[75, 100]
					],
					"Sensors": ["/intelcpu/0/temperature/17"],
					"Trigger": {
						"Type": "AlwaysTrigger"
						}
				}
			}
		],
		"Effects": [
			{
			"Type": "SpectrumEffect",
				"Config": {
					"Saturation": 1,
					"Brightness": 1,
					"HueStep": 1,
					"Trigger": {
						"Type": AlwaysTrigger"
						}
				}
			}
		]
	}
		{
		"Name": 140mm",
		"Ports": [
			[9802, 8501, 5]
		],
		"SpeedControllers": [
			{
			"Type": "StaticSpeedController",
				"Config": {
				"Speed": 40,
				"Trigger": {
					"Type": "AlwaysTrigger"
					}
				}
			}
		],
		"Effects": [
			{
			"Type": "SpectrumEffect",
				"Config": {
					"Saturation": 1,
					"Brightness": 1,
					"HueStep": 1,
					"Trigger": {
						"Type": AlwaysTrigger"
						}
				}
			}
		]
	}
  ],
  "ComputerStateProfiles": [],
	"PortConfigs": [
	{
		"Ports": [
			[9802, 8501, 1]
		],
		"Config": {
			"DeviceType": "FloeRiing",
			"IgnoreColorCache": true
			}
		{
		"Ports": [
			[9802, 8501, 2],
			[9802, 8501, 3],
			[9802, 8501, 4],
			[9802, 8501, 5]
		],
		"Config": {
			"DeviceType": "RiingDuo",
			"IgnoreColorCache": true
			}
		}
	]
	}
  "SensorConfigs": []
}
```

MoshiMoshi0, 2020-11-03:
You pretty much configured it how you want just had minor errors. Its best to test in console mode, it will show you the log in real time so you can see what is happening and more detailed errors. The service was probably crashing while the console was waiting for it to start, so it timed out.

`IgnoreColorCache` is only needed if your controller goes to the default rainbow when it does not receive any colors for some time. I think only Riing Quad controllers do this.

>  Ideally, I would like to set the color profile and have it persist at all times, such as when booting, when the service isn't running, etc. I'm not sure if that is possible. Right now with the stock software, whenever I exit it, it defaults back to the rainbow colors.

Yea you have to have the service running for the fan curve and other effects to work. It runs as system account so it will start in the background before you login, you dont have to have any windows open or apps in tray.

This config should work:
```json
{
  "Profiles": [
    {
      "Name": "Pump",
      "Ports": [
        [9802, 8501, 1]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController",
          "Config": {
            "Speed": 100,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Saturation": 1,
            "Brightness": 1,
            "HueStep": 1,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Radiator",
      "Ports": [
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8501, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/17"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Saturation": 1,
            "Brightness": 1,
            "HueStep": 1,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "140mm",
      "Ports": [
        [9802, 8501, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController",
          "Config": {
            "Speed": 40,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Saturation": 1,
            "Brightness": 1,
            "HueStep": 1,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8501, 1]
      ],
      "Config": {
        "DeviceType": "FloeRiing"
      }
    },
    {
      "Ports": [
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8501, 4],
        [9802, 8501, 5]
      ],
      "Config": {
        "DeviceType": "RiingDuo"
      }
    }
  ]
}
```

MoshiMoshi0, 2020-11-03:
If you want to keep the boot effect all the time I guess you could remove all effects from the above config.
Also I dont think you can control the speed of the pump, iirc it will run at 100% on its own.

cwills75, 2020-11-04:
That seems to have done it, thank you very much.  I sent you a tip on Paypal, you should get it today or tomorrow.

closed 2020-11-04 by cwills75