MoshiMoshi0/TTController #83: Configuration Help - Ryzen 5800x and Floe Riing 360

Soothx, 2020-11-07:
**Describe your desired configuration**
I am having problems with my Curve Points, as whenever I start the service, all fans immediately spins to 100% and doesn't slow at all. I am unsure if it's a sensor problem, since for some examples I've seen the Sensors: "amdcpu/" or something like it, where I cannot find one like it with the Ryzen 5800x. I will adjust the curve points later. 

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 2009 [19041.1.amd64fre.vb_release.191206-1406]
»        Build: 0.11.0-beta+Branch.master.Sha.2b6096e1aeefc2bc0b2e3e715c364bcd849d9ddc
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»  [0x264a, 0x1fb4]: \\?\hid#vid_264a&pid_1fb4#9&c25ecd8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8116
»        Version: 2.0.2
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 35, Rpm: 636, Unknown: 255]
»                        Identifier: [9802, 8116, 1]
»                2:
»                        Data: [PortId: 2, Speed: 35, Rpm: 624, Unknown: 255]
»                        Identifier: [9802, 8116, 2]
»                3:
»                        Data: [PortId: 3, Speed: 35, Rpm: 619, Unknown: 255]
»                        Identifier: [9802, 8116, 3]
»                4:
»                        Data: [PortId: 4, Speed: 35, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8116, 4]
»                5:
»                        Data: [PortId: 5, Speed: 35, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8116, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 40,0 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 35,0 °C
»                Temperature #3 (/lpc/nct6798d/temperature/3): 78,0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 35,0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 28,0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 82,0 °C
»
»  NVIDIA GeForce RTX 3080:
»        Temperature:
»                GPU Core (/gpu-nvidia/0/temperature/0): 52,0 °C
»
»  Samsung SSD 970 EVO 250GB:
»        Temperature:
»                Temperature (/nvme/5/temperature/0): 49,0 °C
»                Temperature 1 (/nvme/5/temperature/6): 49,0 °C
»                Temperature 2 (/nvme/5/temperature/7): 56,0 °C
»
»  Samsung SSD 840 EVO 120GB:
»        Temperature:
»                Temperature (/ssd/4/temperature/0): 40,0 °C
»
»  Samsung SSD 850 EVO 250GB:
»        Temperature:
»                Temperature (/ssd/2/temperature/0): 31,0 °C
»
»  WDC WD10EARS-00Y5B1:
»        Temperature:
»                Temperature (/hdd/3/temperature/0): 45,0 °C
»
»  Samsung SSD 840 EVO 120GB:
»        Temperature:
»                Temperature (/ssd/1/temperature/0): 37,0 °C
»
»  ST2000DM001-1ER164:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 45,0 °C
»
-------------------------------
Press any key to continue...

```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8116, 1],
        [9802, 8116, 2],
        [9802, 8116, 3],
        [9802, 8116, 4],
        [9802, 8116, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [40, 15],
              [50, 25],
              [60, 35],
              [70, 45],
              [80, 60],
              [90, 80]
            ],
            "Sensors": ["/lpc/nct6798d/temperature/2"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "SensorMixFunction": "Average",
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "StaticColorEffect",
          "Config": {
            "Color": {
              "Full": [255, 149, 45]
            },
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8116, 1],
        [9802, 8116, 2],
        [9802, 8116, 3],
        [9802, 8116, 4],
        [9802, 8116, 5]
      ],
      "Speed": 35,
      "Color": {
        "Full": [0, 0, 0]
      }
    }
  ],
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 5000
}
```

I just wanted to provide some additional Sensor data, in case it might be useful since it's a new CPU.

```Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Control:
»                Fan Control #1 (/lpc/nct6798d/control/0): 43,5 %
»                Fan Control #2 (/lpc/nct6798d/control/1): 72,2 %
»                Fan Control #3 (/lpc/nct6798d/control/2): 43,5 %
»                Fan Control #4 (/lpc/nct6798d/control/3): 100,0 %
»                Fan Control #5 (/lpc/nct6798d/control/4): 100,0 %
»                Fan Control #6 (/lpc/nct6798d/control/5): 100,0 %
»                Fan Control #7 (/lpc/nct6798d/control/6): 100,0 %
»        Voltage:
»                Vcore (/lpc/nct6798d/voltage/0): 1,01 V
»                Voltage #2 (/lpc/nct6798d/voltage/1): 0,99 V
»                AVCC (/lpc/nct6798d/voltage/2): 3,39 V
»                +3.3V (/lpc/nct6798d/voltage/3): 3,30 V
»                Voltage #5 (/lpc/nct6798d/voltage/4): 1,00 V
»                Voltage #6 (/lpc/nct6798d/voltage/5): 0,84 V
»                Voltage #7 (/lpc/nct6798d/voltage/6): 0,32 V
»                3VSB (/lpc/nct6798d/voltage/7): 3,39 V
»                VBat (/lpc/nct6798d/voltage/8): 3,36 V
»                VTT (/lpc/nct6798d/voltage/9): 0,90 V
»                Voltage #11 (/lpc/nct6798d/voltage/10): 0,42 V
»                Voltage #12 (/lpc/nct6798d/voltage/11): 0,48 V
»                Voltage #13 (/lpc/nct6798d/voltage/12): 1,03 V
»                Voltage #14 (/lpc/nct6798d/voltage/13): 0,39 V
»                Voltage #15 (/lpc/nct6798d/voltage/14): 0,36 V
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 40,0 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 35,0 °C
»                Temperature #3 (/lpc/nct6798d/temperature/3): 78,0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 35,0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 28,0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 82,0 °C
»        Fan:
»                Fan #1 (/lpc/nct6798d/fan/0): 867 RPM
»                Fan #2 (/lpc/nct6798d/fan/1): 1598 RPM
»
»  AMD Ryzen 7 5800X:
»        Load:
»                CPU Core #1 (/amdcpu/0/load/1): 0,0 %
»                CPU Core #2 (/amdcpu/0/load/2): 0,0 %
»                CPU Core #3 (/amdcpu/0/load/3): 0,0 %
»                CPU Core #4 (/amdcpu/0/load/4): 0,0 %
»                CPU Core #5 (/amdcpu/0/load/5): 0,0 %
»                CPU Core #6 (/amdcpu/0/load/6): 0,0 %
»                CPU Core #7 (/amdcpu/0/load/7): 0,0 %
»                CPU Core #8 (/amdcpu/0/load/8): 16,7 %
»                CPU Core #9 (/amdcpu/0/load/9): 0,0 %
»                CPU Core #10 (/amdcpu/0/load/10): 0,0 %
»                CPU Core #11 (/amdcpu/0/load/11): 0,0 %
»                CPU Core #12 (/amdcpu/0/load/12): 0,0 %
»                CPU Core #13 (/amdcpu/0/load/13): 0,0 %
»                CPU Core #14 (/amdcpu/0/load/14): 14,3 %
»                CPU Core #15 (/amdcpu/0/load/15): 0,0 %
»                CPU Core #16 (/amdcpu/0/load/16): 42,9 %
»                CPU Total (/amdcpu/0/load/0): 4,6 %
»
»  Generic Memory:
»        Data:
»                Memory Used (/ram/data/0): 4 GB
»                Memory Available (/ram/data/1): 28 GB
»                Virtual Memory Used (/ram/data/2): 9 GB
»                Virtual Memory Available (/ram/data/3): 28 GB
»        Load:
»                Memory (/ram/load/0): 13,3 %
»                Virtual Memory (/ram/load/1): 24,0 %
»```

MoshiMoshi0, 2020-11-07:
For Zen 3 temperatures you will have to wait for LibreHardwareMonitor/LibreHardwareMonitor#310 to be fixed.
Those Nuvoton temperatures will probably be some sensors on the motherboard, maybe there is one near cpu that you could use temporarily.

Not sure why it would ramp up to 100% tho, can you stop the service, run console mode for some time and post the output? Console mode will print sensor temperature and fan speed every 5 seconds.

Soothx, 2020-11-07:
Makes sense in regards to the temp sensors then. Doesn't seem like console mode is printing any sensor temperatures.

Console mode:
```02:56:31.3540: ================================================================
02:56:31.3780: Initializing service, version "0.11.0-beta+Branch.master.Sha.2b6096e1aeefc2bc0b2e3e715c364bcd849d9ddc"
02:56:31.4035: Loading plugin assembly: TTController.Plugin.DpsgController [0.11.0.0]
02:56:31.4175: Loading plugin assembly: TTController.Plugin.Level20DeskController [0.11.0.0]
02:56:31.4305: Loading plugin assembly: TTController.Plugin.RiingController [0.11.0.0]
02:56:31.4440: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.11.0.0]
02:56:31.4579: Loading plugin assembly: TTController.Plugin.RiingQuadController [0.11.0.0]
02:56:31.4713: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.11.0.0]
02:56:31.4840: Loading plugin assembly: TTController.Plugin.AuroraEffect [0.11.0.0]
02:56:31.4840: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.11.0.0]
02:56:31.5084: Loading plugin assembly: TTController.Plugin.ColorSequenceEffect [0.11.0.0]
02:56:31.5213: Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.11.0.0]
02:56:31.5328: Loading plugin assembly: TTController.Plugin.FlowEffect [0.11.0.0]
02:56:31.5328: Loading plugin assembly: TTController.Plugin.IpcEffect [0.11.0.0]
02:56:31.5568: Loading plugin assembly: TTController.Plugin.PingPongEffect [0.11.0.0]
02:56:31.5693: Loading plugin assembly: TTController.Plugin.PulseEffect [0.11.0.0]
02:56:31.5808: Loading plugin assembly: TTController.Plugin.RawEffect [0.11.0.0]
02:56:31.5953: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.11.0.0]
02:56:31.5953: Loading plugin assembly: TTController.Plugin.RippleEffect [0.11.0.0]
02:56:31.6196: Loading plugin assembly: TTController.Plugin.SensorEffect [0.11.0.0]
02:56:31.6637: Loading plugin assembly: TTController.Plugin.SoundEffect [0.11.0.0]
02:56:31.6762: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.11.0.0]
02:56:31.6892: Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.11.0.0]
02:56:31.6892: Loading plugin assembly: TTController.Plugin.WaveEffect [0.11.0.0]
02:56:31.7126: Loading plugin assembly: TTController.Plugin.ZoneEffect [0.11.0.0]
02:56:31.7256: Loading plugin assembly: TTController.Plugin.LerpLedColorModifier [0.11.0.0]
02:56:31.7391: Loading plugin assembly: TTController.Plugin.ReverseLedColorModifier [0.11.0.0]
02:56:31.7521: Loading plugin assembly: TTController.Plugin.RotateLedColorModifier [0.11.0.0]
02:56:31.7646: Loading plugin assembly: TTController.Plugin.TrimLedColorModifier [0.11.0.0]
02:56:31.7646: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.11.0.0]
02:56:31.7915: Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.11.0.0]
02:56:31.8050: Loading plugin assembly: TTController.Plugin.IpcSpeedController [0.11.0.0]
02:56:31.8200: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.11.0.0]
02:56:31.8334: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.11.0.0]
02:56:31.8455: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.11.0.0]
02:56:31.8455: Loading plugin assembly: TTController.Plugin.IpcTrigger [0.11.0.0]
02:56:31.8705: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.11.0.0]
02:56:31.8821: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.11.0.0]
02:56:31.8947: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.11.0.0]
02:56:31.9067: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.11.0.0]
02:56:31.9067: Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.11.0.0]
02:56:31.9328: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.11.0.0]
02:56:31.9478: Loading plugin assembly: TTController.Plugin.WebSocketIpcServer [0.11.0.0]
02:56:31.9543: Creating Config Manager...
02:56:32.2336: Loading config...
02:56:32.3662: Creating DataCache...
02:56:32.3662: Creating Plugin Store...
02:56:32.3767: Loaded plugin "StaticColorEffect"
02:56:32.3767: Loaded plugin "AlwaysTrigger"
02:56:32.3767: Loaded plugin "PwmSpeedController"
02:56:32.3767: Loaded plugin "AlwaysTrigger"
02:56:32.3767: Loaded 4 plugins
02:56:32.3767: Creating Sensor Manager...
02:56:32.3767: Initializing Libre Hardware Monitor...
02:56:32.5106: Detected 30 sensors
02:56:32.5106: Creating Device Manager...
02:56:32.6762: Searching for controller changes
02:56:32.6762: Searching for "Dpsg" controllers
02:56:32.8184: Searching for "Level 20 Desk" controllers
02:56:32.8184: Searching for "Riing" controllers
02:56:32.8184: Searching for "Riing Plus" controllers
02:56:32.8410: Initialized "Riing Plus" controller [9802, 8116], version: "2.0.2"
02:56:32.8440: Searching for "Riing Quad" controllers
02:56:32.8440: Searching for "Riing Trio" controllers
02:56:32.8440: Processing profile "Default"
02:56:32.8440: Assigned plugin "StaticColorEffect" to "Default" profile
02:56:32.8440: Assigned plugin "PwmSpeedController" to "Default" profile
[... 59 lines cut ...]
02:57:27.8811: Port [9802, 8116, 5] data: [PortId: 5, Speed: 100, Rpm: 0, Unknown: 255]
02:57:32.8882: Port [9802, 8116, 1] data: [PortId: 1, Speed: 100, Rpm: 1568, Unknown: 255]
02:57:32.8882: Port [9802, 8116, 2] data: [PortId: 2, Speed: 100, Rpm: 1570, Unknown: 255]
02:57:32.8882: Port [9802, 8116, 3] data: [PortId: 3, Speed: 100, Rpm: 1559, Unknown: 255]
[... cut ...]
02:57:33.5758: Disposing Device Manager...
02:57:33.5788: Disposed controllers: 1
02:57:33.5788: Disposing Config Manager...
02:57:33.5788: Disposing Plugin Store...
02:57:33.5788: Disposing plugin "StaticColorEffect"
02:57:33.5788: Disposing plugin "AlwaysTrigger"
02:57:33.5788: Disposing plugin "PwmSpeedController"
02:57:33.5788: Disposing plugin "AlwaysTrigger"
02:57:33.5959: Finalizing done!
02:57:33.5959: ================================================================
Press any key to return to the menu...
```

MoshiMoshi0, 2020-11-07:
Interesting, it seems like it cant find the sensor when you run it in console/service mode, but it is clearly there on the debug reports. If it fails to read temperature it will fail safe to 100% fan speed. 
Might be a bug with LibreHardwareMonitor, I'll investigate.

For more info can you stop the service/console, edit the `TTController.Service.exe.Config` file, and change:
`      <logger name="*" minLevel="Info" writeTo="file" enabled="true" />`
to:
`      <logger name="*" minLevel="Trace" writeTo="file" enabled="true" />`

Then start console mode again, run for 10s, stop it by pressing any key, and post the latest log file from the `Logs` folder. After that, revert the change in that config file so your logs are not huge. 
As a workaround, for now you should be able to use `/gpu-nvidia/0/temperature/0` just fine, or just use a static fan speed.

Soothx, 2020-11-07:
[log.20201107.18.log](https://github.com/MoshiMoshi0/TTController/files/5504178/log.20201107.18.log)

Is this done as you requested?

MoshiMoshi0, 2020-11-07:
Yes, thanks.

So I forgot that by default motherboard sensors are disabled to improve startup time, while in the sensors debug report everything is enabled.

If you still want to use that sensor add `"MotherboardSensorsEnabled": true,` to your config:
```json
  ...
  "DeviceRgbTimerInterval": 5000,
  "MotherboardSensorsEnabled": true
}
```

That will fix the 100% fan issue with Nuvoton sensors.
Like I said for Zen 3 temperatures we will have to wait, I'll push a new build when its added to LibreHardwareMonitor.

Soothx, 2020-11-07:
I'll just keep it on the GPU sensor until the update comes around. Thanks for the great support and help! If you do need testing, I'll be happy to help.

MoshiMoshi0, 2020-11-14:
Please try this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/36313433/artifacts

Soothx, 2020-11-19:
Does not appear to have any new sensors, only shows the following Sensors for the CPU:

`AMD Ryzen 7 5800X:
»        Load:
»                CPU Core #1 (/amdcpu/0/load/1): 25,0 %
»                CPU Core #2 (/amdcpu/0/load/2): 12,5 %
»                CPU Core #3 (/amdcpu/0/load/3): 0,0 %
»                CPU Core #4 (/amdcpu/0/load/4): 0,0 %
»                CPU Core #5 (/amdcpu/0/load/5): 0,0 %
»                CPU Core #6 (/amdcpu/0/load/6): 0,0 %
»                CPU Core #7 (/amdcpu/0/load/7): 0,0 %
»                CPU Core #8 (/amdcpu/0/load/8): 0,0 %
»                CPU Core #9 (/amdcpu/0/load/9): 0,0 %
»                CPU Core #10 (/amdcpu/0/load/10): 0,0 %
»                CPU Core #11 (/amdcpu/0/load/11): 0,0 %
»                CPU Core #12 (/amdcpu/0/load/12): 0,0 %
»                CPU Core #13 (/amdcpu/0/load/13): 12,5 %
»                CPU Core #14 (/amdcpu/0/load/14): 0,0 %
»                CPU Core #15 (/amdcpu/0/load/15): 25,0 %
»                CPU Core #16 (/amdcpu/0/load/16): 0,0 %
»                CPU Total (/amdcpu/0/load/0): 4,7 %`

MoshiMoshi0, 2020-11-19:
Ah, there is an issue with the build process of LibreHardwareMonitor, the changes does not get pushed so we are stuck on old version. I'll make an issue about that, hopefully it gets fixed soon, then I'll rebuild again.

MoshiMoshi0, 2020-11-20:
Ok this build should work: 
[TTController_0.11.0.1-beta.9f1e13bdf2ee8be75343d4c8276187c3a50617e2.zip](https://github.com/MoshiMoshi0/TTController/files/5573705/TTController_0.11.0.1-beta.9f1e13bdf2ee8be75343d4c8276187c3a50617e2.zip)

MoshiMoshi0, 2020-11-20:
Ok this build should work:
[TTController_0.11.0.0-beta.42768ba987b22e18ff8cdd114f754e5f7ce16d72.zip](https://github.com/MoshiMoshi0/TTController/files/5573716/TTController_0.11.0.0-beta.42768ba987b22e18ff8cdd114f754e5f7ce16d72.zip)

MoshiMoshi0, 2020-11-28:
I'm assuming it works, closing.

closed 2020-11-28 by MoshiMoshi0