MoshiMoshi0/TTController #84: Riing Quad color bleed and value mismatch

monners, 2020-11-10:
**Describe the issue**
Riing Quad fans/controller
1. When setting `PerLed` values in config some LEDs appear to take on colors that haven't been set.
This doesn't seem to happen if all LEDs are set to the same color but if, for example, I set the innermost zone to black, and the second innermost zone to red, one LED on all three fans appears yellow/orange.

2. Color byte codes don't appear to be mapped to [RED, GREEN, BLUE], and instead respond to [BLUE, RED, GREEN].

Config below results in:
Outer zone: white but with one LED purple
Outer-mid zone: white
Inner-mid zone: red but with one LED yellow
Inner zone: dark/off

**Debug report**
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 2004 [19041.1.x86fre.vb_release.191206-1406]
»        Build: 0.11.0-beta+Branch.master.Sha.2b6096e1aeefc2bc0b2e3e715c364bcd849d9ddc
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»  [0x264a, 0x2260]: \\?\hid#vid_264a&pid_2260#8&272c7f0f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Quad
»        VendorId: 9802
»        ProductId: 8800
»        Version: 3.0.3
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 40, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8800, 1]
»                2:
»                        Data: [PortId: 2, Speed: 40, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8800, 2]
»                3:
»                        Data: [PortId: 3, Speed: 40, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8800, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8800, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8800, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 36.5 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 33.0 °C
»                Temperature #3 (/lpc/nct6798d/temperature/3): 14.0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 14.0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 25.0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 63.0 °C
»
»  Intel Core i9-10940X:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 38.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 38.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 41.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 35.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 38.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 37.0 °C
»                CPU Core #7 (/intelcpu/0/temperature/6): 40.0 °C
»                CPU Core #8 (/intelcpu/0/temperature/7): 40.0 °C
»                CPU Core #9 (/intelcpu/0/temperature/8): 37.0 °C
»                CPU Core #10 (/intelcpu/0/temperature/9): 38.0 °C
»                CPU Core #11 (/intelcpu/0/temperature/10): 37.0 °C
»                CPU Core #12 (/intelcpu/0/temperature/11): 39.0 °C
»                CPU Core #13 (/intelcpu/0/temperature/12): 40.0 °C
»                CPU Core #14 (/intelcpu/0/temperature/13): 39.0 °C
»                CPU Package (/intelcpu/0/temperature/14): 40.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/15): 72.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/16): 72.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/17): 69.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/18): 75.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/19): 72.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/20): 73.0 °C
»                CPU Core #7 Distance to TjMax (/intelcpu/0/temperature/21): 70.0 °C
»                CPU Core #8 Distance to TjMax (/intelcpu/0/temperature/22): 70.0 °C
»                CPU Core #9 Distance to TjMax (/intelcpu/0/temperature/23): 73.0 °C
»                CPU Core #10 Distance to TjMax (/intelcpu/0/temperature/24): 72.0 °C
»                CPU Core #11 Distance to TjMax (/intelcpu/0/temperature/25): 73.0 °C
»                CPU Core #12 Distance to TjMax (/intelcpu/0/temperature/26): 71.0 °C
»                CPU Core #13 Distance to TjMax (/intelcpu/0/temperature/27): 70.0 °C
»                CPU Core #14 Distance to TjMax (/intelcpu/0/temperature/28): 71.0 °C
»                Core Max (/intelcpu/0/temperature/29): 41.0 °C
»                Core Average (/intelcpu/0/temperature/30): 38.4 °C
»
»  NVIDIA GeForce RTX 2080 Ti:
»        Temperature:
»                GPU Core (/gpu-nvidia/0/temperature/0): 43.0 °C
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Riing Quad Controller",
      "Ports": [
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 3],
        [9802, 8800, 4]
      ],
      "Effects": [
        {
          "Type": "StaticColorEffect",
          "Config": {
            "Color": {
              "PerLed": [
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],

                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],
                [255, 255, 255],

                [0, 255, 0],
                [0, 255, 0],
                [0, 255, 0],
                [0, 255, 0],
                [0, 255, 0],
                [0, 255, 0],
                [0, 255, 0],
                [0, 255, 0],
                [0, 255, 0],

                [0, 0, 0],
                [0, 0, 0],
                [0, 0, 0],
                [0, 0, 0],
                [0, 0, 0],
                [0, 0, 0],
                [0, 0, 0],
                [0, 0, 0],
                [0, 0, 0]
              ]
            },
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfigs": [{
    "Ports": [
      [9802, 8800, 1],
      [9802, 8800, 2],
      [9802, 8800, 3],
      [9802, 8800, 4]
    ],
    "Config": {
      "DeviceType": "RiingQuad",
      "IgnoreColorCache": true
    }
  }],
  "SensorConfigs": []
}
```

monners, 2020-11-10:
Love the project by the way.  It's such a shame TT don't give these beautiful fans the support they deserve.

MoshiMoshi0, 2020-11-10:
Thanks for reporting!

Seems like the colors were offset by one byte by accident.
Please try this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/36230141/artifacts

MoshiMoshi0, 2020-11-10:
Also you can use the new [ZoneEffect](https://moshimoshi0.github.io/TTController/plugins/effects/zoneeffect) for easier configuration if you just want each zone with a specific color:
```json
  ...
  "Effects": [
    {
      "Type": "ZoneEffect",
      "Config": {
        "Effects": [
          {
            "Type": "StaticColorEffect",
            "Config": { "Color": { "Full": [255, 255, 255] } }
          },
          {
            "Type": "StaticColorEffect",
            "Config": { "Color": { "Full": [255, 255, 255] } }
          },
          {
            "Type": "StaticColorEffect",
            "Config": { "Color": { "Full": [0, 255, 0] } }
          },
          {
            "Type": "StaticColorEffect",
            "Config": { "Color": { "Full": [0, 0, 0] } }
          }
        ]
      }
    }
  ]
  ...
```

Each `StaticColorEffect` will be mapped to consecutive riing quad zone.

monners, 2020-11-10:
Absolute legend.  Worked perfectly.  Donation incoming.

MoshiMoshi0, 2020-11-11:
Thanks!

I'll close this issue and post a new release soon with the fix, it will be exactly the same as the build from the link above so you dont have to redownload.

closed 2020-11-11 by MoshiMoshi0