MoshiMoshi0/TTController #85: ColorModifiers won't instantiate

MrBradford, 2020-11-13:
**Describe the issue**
Any use of the new ColorModifiers causes a crash on startup. The error suggests that it can't find the inherited class for any of the modifier types - it's trying to instantiate the ModifierBase class. I've tested this with all modifier types with various config parameters (I've included one config.json below).

**Debug report**
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.wow64fre.19h1_release.190318-1202]
»        Build: 0.11.0-beta+Branch.master.Sha.9f1e13bdf2ee8be75343d4c8276187c3a50617e2
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»  [0x264a, 0x1fa5]: \\?\hid#vid_264a&pid_1fa5#8&70563ff&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»  [0x264a, 0x1fa6]: \\?\hid#vid_264a&pid_1fa6#9&2db26cf4&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»  [0x264a, 0x1fa7]: \\?\hid#vid_264a&pid_1fa7#9&d1809eb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Version: 2.0.3
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 32, Rpm: 568, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 32, Rpm: 578, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 32, Rpm: 570, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 32, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 32, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8102
»        Version: 2.0.3
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 32, Rpm: 573, Unknown: 255]
»                        Identifier: [9802, 8102, 1]
»                2:
»                        Data: [PortId: 2, Speed: 32, Rpm: 580, Unknown: 255]
»                        Identifier: [9802, 8102, 2]
»                3:
»                        Data: [PortId: 3, Speed: 32, Rpm: 554, Unknown: 255]
»                        Identifier: [9802, 8102, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Plus
[... 229 lines cut ...]
				"LedReverse": [ false ],
				"LedCountHandling": "Lerp"
			}
		},
		{
			"Ports": [
				[ 9802, 8101, 5 ]
			],
			"Config": {
				"Name": "RAM",
				"LedRotation": [],
				"LedReverse": [ false ],
				"LedCountHandling": "Copy"
			}
		}
	],
	"SensorConfigs": [
		{
			"Sensors": [ "/intelcpu/0/temperature/8" ],
			"Config": {
				"CriticalValue": 90
			}
		}
	]
}

MrBradford, 2020-11-13:
Oops - that's my json without a colormodifier just to test whether an empty array would work. An actual array might be:
"ColorModifiers": [
{
		"Type": "LerpLedColorModifier",
		"LerpType": "Nearest"
	}
]

But really it doesn't matter which modifier you try to load.

MrBradford, 2020-11-13:
Also, I tried removing the old style params from the other port configs just in case that was causing issues. From what I can tell it just ignores those now anyway.

MrBradford, 2020-11-13:
Here's the callstack portion of the log:

2020-11-13 11:37:48.6034|INFO|TTController.Service.Managers.ConfigManager|Loading config...
2020-11-13 11:37:48.7567|FATAL|TTController.Service.Managers.ConfigManager|Newtonsoft.Json.JsonSerializationException: Could not create an instance of type TTController.Common.Plugin.ILedColorModifierBase. Type is an interface or abstract class and cannot be instantiated. Path 'ColorModifiers[0].Type', line 5, position 13.
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateNewObject(JsonReader reader, JsonObjectContract objectContract, JsonProperty containerMember, JsonProperty containerProperty, String id, Boolean& createdFromNonDefaultCreator)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at TTController.Service.Config.Converters.AbstractNamedValueTupleConverter`2.ReadJson(JsonReader reader, Type objectType, ValueTuple`2 existingValue, Boolean hasExistingValue, JsonSerializer serializer)
   at Newtonsoft.Json.JsonConverter`1.ReadJson(JsonReader reader, Type objectType, Object existingValue, JsonSerializer serializer)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.DeserializeConvertable(JsonConverter converter, JsonReader reader, Type objectType, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   at TTController.Service.Managers.ConfigManager.LoadOrCreateConfig()
2020-11-13 11:37:48.7606|FATAL|TTController.Service.Managers.ConfigManager|Failed to load the config!

MrBradford, 2020-11-13:
Generated with this config: 

				"ColorModifiers": [
					{
						"Type": "TrimLedColorModifier",
						"Config": {
							"WrapRemainder": true
						}
					}
				]

MoshiMoshi0, 2020-11-13:
Thanks for detailed info, I have no idea how I tested those modifiers. 
I had to accidentally delete some code before pushing the release, sorry about that.

This build should work: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/36301960/artifacts

MrBradford, 2020-11-13:
No worries - I was poking in the code and it seems like maybe the ILedColorModifierBase is incomplete in the release branch? Iat doesn't inherit from iPlugin. I'm just guessing though - I'm not familiar with how the plugin stuff works.

Can I clone the source for the zip you just sent me? It will instantiate the colorModifiers now (yay!) but there are some bugs with the way it handles the lights. I'd like to help fix them if I can.

MrBradford, 2020-11-13:
Never mind - I see the updates now!

MoshiMoshi0, 2020-11-13:
> but there are some bugs with the way it handles the lights
What are the bugs you found?

MoshiMoshi0, 2020-11-19:
I'll close this since the original issue is fixed, if you find any bugs in the modifiers please make a new one.

closed 2020-11-19 by MoshiMoshi0