MoshiMoshi0/TTController #87: Funding Required

gib310890, 2020-11-14:
**Describe the feature you'd like**
A clear and concise description of what you want to happen.

Hi, how much would it cost the developers to build a drop down list (in the menu) to configure the TT controller ?
This would simplify every newbie's hurdle of using this software. 

It could be just basic setup to make the script work and for more advanced configs they can learn to code it and add to the already working script. 

I'd be interested to pay the developer for this development/feature.

MoshiMoshi0, 2020-11-14:
I think i tried to design the GUI 3 or 4 times throughout the whole development time but never got anything that I was satisfied with. I would like to make the GUI generate the forms/controls dynamically based on class properties, so each plugin would not have to reference a bunch of gui stuff and have a custom wpf control. Basically the plugins and the service should be decoupled from the GUI and have no references to it, I dont want the service loading wpf dlls if it has no window. I also tried to generate a json schema for the whole config class and then use a json schema to web form generator, but there were some issues too.

Maybe the GUI should just be in another repository and have a similar project setup to the service, where each plugin would have a separate project with its own custom control.

So its not really about lack of funding, but rather about finding the correct design.
If anyone has any ideas or wants to do a PR I would appreciate it.

gib310890, 2020-11-14:
I was thinking in the lines of a simplified approach. The Drop down list of settings would just generate a working config file. No GUI. Just the detection of ports, effects from the report option would setup the config file as a working Baseline.

redaxmedia, 2020-12-19:
@gib310890 There was an [GUI ](https://github.com/MoshiMoshi0/TTController/files/3398830/RiingPlusController_v1.7.zip)2 years ago from what I found on this [thread ](https://community.thermaltake.com/index.php?/topic/75867-alternative-to-tt-rgb-software/)at the thermaltake forums.

@MoshiMoshi0 This is a bit confusing to me, where and why is it gone? :-)

MoshiMoshi0, 2020-12-20:
> This is a bit confusing to me, where and why is it gone? :-)

The gui version I wrote for myself and it only supported one type of fans, this is a rewrite from scratch. I planned to do a GUI for this project too but like I said I have not found a satisfying solution yet.

redaxmedia, 2020-12-20:
Well, I could create a web frontend in Angular that could be ported to an cross-platform Desktop App using Electron.

The only thing I need is a REST or websocket API...

samaelestevez, 2022-09-11:
Have you thought of "merging" (for lack of better word) with the project "Fan Control"?

"Fan Control" is here in github. That piece of software is nice.