MoshiMoshi0/TTController #88: Configuration Help

skankeh, 2020-11-26:
First, thank you very much for your efforts on this project, please let me know if I can donate somehow. I honestly don't understand why TT can't make some decent software for their great RGB fans.

I'm having difficulty understanding how to get started, if you can set me up with a basic config as a starting point I will be very grateful!

**Desired Configuration**
I have 9 Riing+ fans connected to 2 controllers (seem to be different despite the product I purchased being the same listing). Controllers are connected using split USB header cable. Controllers are set with dip switches to controller 1 and 2. I want all fans running at 20% normally (idle) but fan curve linked to CPU temp. For lighting, a single colour is fine as a start (I'd love Aurora as it looks great using the TTRGBPlus software, but I don't know if it will work in this case as it seems the 2nd controller only shows per-led mode here).

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 2009 [19041.1.amd64fre.vb_release.191206-1406]
»        Build: 0.11.0-beta+Branch.master.Sha.9f1e13bdf2ee8be75343d4c8276187c3a50617e2
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»  [0x264a, 0x1fa5]: \\?\hid#vid_264a&pid_1fa5#7&297efe38&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»  [0x264a, 0x2261]: \\?\hid#vid_264a&pid_2261#7&1c58fa5a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Version: 2.0.3
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 60, Rpm: 1072, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 60, Rpm: 1071, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 60, Rpm: 1064, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 1018, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 1039, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Quad
»        VendorId: 9802
»        ProductId: 8801
»        Version: 3.0.3
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 60, Rpm: 1063, Unknown: 255]
»                        Identifier: [9802, 8801, 1]
»                2:
»                        Data: [PortId: 2, Speed: 60, Rpm: 1059, Unknown: 255]
»                        Identifier: [9802, 8801, 2]
»                3:
»                        Data: [PortId: 3, Speed: 60, Rpm: 1078, Unknown: 255]
»                        Identifier: [9802, 8801, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 1061, Unknown: 255]
»                        Identifier: [9802, 8801, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8801, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  ITE IT8628E:
»        Temperature:
»                Temperature #1 (/lpc/it8628e/temperature/0): 26.0 °C
»                Temperature #2 (/lpc/it8628e/temperature/1): 34.0 °C
»                Temperature #3 (/lpc/it8628e/temperature/2): 35.0 °C
»
»  Intel Core i7-6700K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 31.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 29.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 30.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 35.0 °C
»                CPU Package (/intelcpu/0/temperature/4): 35.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/5): 69.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/6): 71.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/7): 70.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/8): 65.0 °C
»                Core Max (/intelcpu/0/temperature/9): 35.0 °C
»                Core Average (/intelcpu/0/temperature/10): 31.3 °C
»
»  AMD Radeon RX 5700 XT:
»        Temperature:
»                GPU Core (/gpu-amd/0/temperature/0): 39.0 °C
»                GPU Hot Spot (/gpu-amd/0/temperature/7): 39.0 °C
»                GPU VR VDDC (/gpu-amd/0/temperature/2): 31.0 °C
»                GPU Memory (/gpu-amd/0/temperature/1): 40.0 °C
»
»  :
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 0.0 °C
»
»  Crucial_CT480M500SSD1:
»        Temperature:
»                Temperature (/ssd/3/temperature/0): 30.0 °C
»
»  ST2000DX002-2DV164:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 31.0 °C
»
»  GIGABYTE GP-ASACNE2512GTTDR:
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 30.0 °C
»
-------------------------------
```

**Current configuration**
```{
  "Profiles": [],
  "ComputerStateProfiles": [],
  "PortConfigs": [],
  "SensorConfigs": []
}
```

MoshiMoshi0, 2020-11-26:
If you need more help, check the docs, they explain a little bit what each parameter does.

This should do it:
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8801, 1],
        [9802, 8801, 2],
        [9802, 8801, 3],
        [9802, 8801, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "MinimumChange": 4,
            "MaximumChange": 8
          }
        }
      ],
      "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Step": 0.003,
            "Length": 64,
            "Mirror": false,
            "Brightness": 1,
            "Saturation": 1,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ]
}
```

skankeh, 2020-11-26:
Thank you for your quick response, this worked great!

Just one thing, though I am not sure this is possible. when using TTRGBplus software, setting Aurora makes the colour-change progress from one fan to the next, so the gradient changes in series, making it look like the "aurora" is moving down the fans. Is this effect possible with TTController?

MoshiMoshi0, 2020-11-26:
Yes, there is `"ColorGenerationMethod"` property, I noticed its not in the docs for some reason.

```json
        {
          "Type": "AuroraEffect",
          "Config": {
            "ColorGenerationMethod": "SpanPorts",
            "Step": 0.003,
            "Length": 64,
            "Mirror": false,
            "Brightness": 1,
            "Saturation": 1,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

You might want to also try setting `"Mirror"` to `true`, I think thats how the thermaltake aurora works.

skankeh, 2020-11-26:
Super, that worked too.

Last issue, it seems the fan control is not working, they are stuck at 60 fan speed (1100RPM) I tested this by attempting to configure them all at 0 or 100 and neither seemed to affect fan speed.

MoshiMoshi0, 2020-11-26:
Oh whoops sorry, I forgot to add a trigger to the speed controller:
```json
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

skankeh, 2020-11-27:
Brilliant, thanks Moshi, I've played around and got some nice effects working thanks to your assistance!

closed 2020-11-27 by skankeh