MoshiMoshi0/TTController #89: Configuration Help

BillisC, 2020-12-04:
**Describe your desired configuration**
Hi there, your work looks incredible especially because you do it for free. But I have a couple of questions for the sound effect:
-Is it possible to implement something similar to thermaltake's gaming mode?
-In the included sound effect mode everything works just great but some leds look very sensitive sometimes(Not talking about ScalingFactor). Also is it possible to make it look like just a static color that changes gradient without the extra stuff?
-Is it possible to add a small delay to a specific fan to make aurora look a bit different between them?
-How can I control each one of the leds seperately?

**Hardware info**
```
Info
-------------------------------
?
?        OS: Windows 10 Enterprise 2009 [19041.1.amd64fre.vb_release.191206-1406]
?        Build: 0.11.0-beta+Branch.master.Sha.42768ba987b22e18ff8cdd114f754e5f7ce16d72
?
-------------------------------
Applications
-------------------------------
?
?
-------------------------------
HID
-------------------------------
?
?  [0x264a, 0x2260]: \\?\hid#vid_264a&pid_2260#8&56eb47c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
?
-------------------------------
Controllers
-------------------------------
?
?  Name: Riing Quad
?        VendorId: 9802
?        ProductId: 8800
?        Version: 3.0.3
?        Ports:
?                1:
?                        Data: [PortId: 1, Speed: 36, Rpm: 683, Unknown: 255]
?                        Identifier: [9802, 8800, 1]
?                2:
?                        Data: [PortId: 2, Speed: 36, Rpm: 668, Unknown: 255]
?                        Identifier: [9802, 8800, 2]
?                3:
?                        Data: [PortId: 3, Speed: 36, Rpm: 671, Unknown: 255]
?                        Identifier: [9802, 8800, 3]
?                4:
?                        Data: [PortId: 4, Speed: 30, Rpm: 0, Unknown: 255]
?                        Identifier: [9802, 8800, 4]
?                5:
?                        Data: [PortId: 5, Speed: 30, Rpm: 0, Unknown: 255]
?                        Identifier: [9802, 8800, 5]
?        Available effect types: PerLed
?
-------------------------------
Sensors
-------------------------------
?
?  Nuvoton NCT6779D:
?        Temperature:
?                CPU (/lpc/nct6779d/temperature/1): 33,0 °C
?                Motherboard (/lpc/nct6779d/temperature/2): 32,0 °C
?                Auxiliary (/lpc/nct6779d/temperature/3): 7,0 °C
?                VRM (/lpc/nct6779d/temperature/4): 29,0 °C
?                AUXTIN2 (/lpc/nct6779d/temperature/5): 22,0 °C
?
?  AMD Ryzen 7 2700:
?        Temperature:
?                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 32,8 °C
?
?  NVIDIA GeForce GTX 1660 SUPER:
?        Temperature:
?                GPU Core (/gpu-nvidia/0/temperature/0): 33,0 °C
?
?  Samsung SSD 970 EVO Plus 500GB:
?        Temperature:
?                Temperature (/nvme/0/temperature/0): 41,0 °C
?                Temperature 1 (/nvme/0/temperature/6): 41,0 °C
?                Temperature 2 (/nvme/0/temperature/7): 51,0 °C
?
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "FrontRGB1",
	  "Ports": [
		[9802, 8800, 1]
	  ],
	  "SpeedControllers": [
	    {
		  "Type": "DpsgSpeedController",
		  "Config": {
		    "FanMode": "Silent",
			"Trigger": {
			  "Type": "AlwaysTrigger"
			}
		  }
		}
	  ],
	  "Effects": [
        {
	      "Type": "SoundEffect",
          "Config": {
            "UseAverage": true,
			"MinimumFrequency": 4000,
			"MaximumFrequency": 16000,

			"ScalingStrategy": "Linear",
			"ScalingFactor": 4.0,
			"ColorGradient": [
			  [0.0, [0, 0, 0]],
			  [0.75, [255, 255, 255]],
			  [1.0, [255, 0, 0]]
			],
			
            "ColorGenerationMethod": "PerPort",
            "Trigger": {
			  "Type": "AlwaysTrigger"
			}
          }
        }
      ]
	},
	{
      "Name": "FrontRGB2",
	  "Ports": [
		[9802, 8800, 2]
	  ],
	  "SpeedControllers": [
	    {
		  "Type": "DpsgSpeedController",
		  "Config": {
		    "FanMode": "Silent",
			"Trigger": {
			  "Type": "AlwaysTrigger"
			}
		  }
		}
	  ],
	  "Effects": [
        {
	      "Type": "SoundEffect",
          "Config": {
            "UseAverage": false,
			"MinimumFrequency": 500,
			"MaximumFrequency": 4000,

			"ScalingStrategy": "Linear",
			"ScalingFactor": 2.0,
			"ColorGradient": [
			  [0.0, [0, 0, 0]],
			  [0.75, [255, 255, 255]],
			  [1.0, [255, 0, 0]]
			],
			
            "ColorGenerationMethod": "PerPort",
            "Trigger": {
			  "Type": "AlwaysTrigger"
			}
          }
        }
      ]
	},
	{
      "Name": "FrontRGB3",
	  "Ports": [
		[9802, 8800, 3]
	  ],
	  "SpeedControllers": [
	    {
		  "Type": "DpsgSpeedController",
		  "Config": {
		    "FanMode": "Silent",
			"Trigger": {
			  "Type": "AlwaysTrigger"
			}
		  }
		}
	  ],
	  "Effects": [
        {
	      "Type": "SoundEffect",
          "Config": {
            "UseAverage": true,
			"MinimumFrequency": 40,
			"MaximumFrequency": 60,
			"ScalingStrategy": "Sqrt",
			"ScalingFactor": 1.5,
			"ColorGradient": [
			  [0.0, [0, 0, 0]],
			  [0.25, [0, 0, 0]],
			  [1.0, [255, 0, 0]]
			],
			
            "ColorGenerationMethod": "PerPort",
            "Trigger": {
			  "Type": "AlwaysTrigger"
			}
          }
        }
      ]
	}
  ],
  "PortConfigs": [
    {
	  "Ports": [[9802, 8800, 1]],
	  "Config": {
	    "Name": "mpureplus1",
		"DeviceType": "PurePlus",
		"IgnoreColorCache": true,
		"ColorModifiers": [
		  {
		    "Type": "LerpLedColorModifier",
			"Config": {
				"LerpType": "Nearest"
			}
		  }
		]
	  }
	},
	{
	  "Ports": [[9802, 8800, 2]],
	  "Config": {
	    "Name": "mpureplus2",
		"DeviceType": "PurePlus",
		"IgnoreColorCache": true,
		"ColorModifiers": [
		  {
		    "Type": "LerpLedColorModifier"
		  }
		]
	  }
	},
	{
	  "Ports": [[9802, 8800, 3]],
	  "Config": {
	    "Name": "mpureplus3",
		"DeviceType": "PurePlus",
		"IgnoreColorCache": true,
		"ColorModifiers": [
		  {
		    "Type": "LerpLedColorModifier"
		  }
		]
	  }
	}
  ]
}``` (The pasted json looks a bit messed for some reason but it's fully working)

MoshiMoshi0, 2020-12-04:
> Is it possible to implement something similar to thermaltake's gaming mode?

Not sure which part do you want from the gaming effect, but anything can be added.
I thought about adding a "bar" sound effect, which would display average "loudness" in a bar instead of mapping each led to a frequency.

> In the included sound effect mode everything works just great but some leds look very sensitive sometimes

Yes that effect requires some trial and error. But I'd rather leave it very configurable so the user can do what he wants instead of adding multiple preset effects.
I implemented that effect a long time ago kinda as a test so the math there might not be best suited for this usecase, I just matched the FFT example from CSCore project.

> Also is it possible to make it look like just a static color that changes gradient without the extra stuff?

Not sure what you mean.

> Is it possible to add a small delay to a specific fan to make aurora look a bit different between them?

Not currently, you could add rotate modifier to the ports to make them slightly different.
But neat idea, I'll see if its feasible to add. I think easiest would be to add a `"DelayEffect"` that wraps another effect and just outputs the wrapped effect colors with a delay.

> How can I control each one of the leds seperately?

Again not sure what you mean, you can set the colors by hand for each led with [StaticColorEffect](https://moshimoshi0.github.io/TTController/plugins/effects/staticcoloreffect) with `"PerLed"` color provider.

Also few notes for your config above:
* `"DpsgSpeedController"` can be only used with thermaltake PSU's, it does not work with anything else. Preferably you should use the [PwmSpeedController](https://moshimoshi0.github.io/TTController/plugins/speed-controllers/pwmspeedcontroller)
* You dont need to add `"LerpLedColorModifier"` since almost all effects will generate the correct amount of colors to match the led count. It is mainly used for effects like `"RazerConnectEffect"` which generate static amount of colors, in that case 4.
* The config looks messed up because the project uses 2 spaces for json indentation, but you added tabs which are commonly 4 spaces. It does not matter at all.

BillisC, 2020-12-04:
Thanks for the fixes and for the quick reply ;) About the gaming mode yeah I meant that bar that you said. But it would be a good idea to add frequencies too, some people like me may prefer to see it only with sub-bass. One more question, what's the difference between decibels, linear and sqrt? I just choose the one that looks better but what do they actually do?

MoshiMoshi0, 2020-12-04:
It determines how the raw FFT values are scaled for the spectrum:
https://github.com/MoshiMoshi0/TTController/blob/4755b33fd21d74d1b7b797105aed88b51c6a60fe/Plugins/Effects/TTController.Plugin.SoundEffect/SpectrumBase.cs#L84-L92

Like I said that plugin probably needs another look to check the math, I dont fully remember/understand it.

I'll make this issue a feature request, but not sure when this will be implemented. You can also make your own effects pretty easy if you want.

BillisC, 2020-12-04:
That's all for now, I appreciate your help 👍

MoshiMoshi0, 2020-12-14:
Pushed a build with the support for delaying ports: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/36831613/artifacts

Usage:
```json
        "ColorModifiers": [
          {
            "Type": "DelayPortModifier",
            "Config": {
              "FrameCount": 63
            }
          }
        ]
```

The `"FrameCount"` determines the delay, with default config each frame is 32ms (30hz/fps), so delaying by 63 frames makes it about 2s.

Also I changed the names of all modifiers, so if you use any, change `LedColor` to `Port` in the name: `LerpLedColorModifier` -> `LerpPortModifier`

MoshiMoshi0, 2020-12-19:
Pushed a build with refactored sound effect: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/36920583/artifacts
There are now two separate sound effects, `"LedSpectrumSoundEffect"` is the renamed `"SoundEffect"`, and a new `"BarSpectrumSoundEffect"`.

They share the same settings that control the fft math. With `"BarSpectrumSoundEffect"` you want to target a small frequency range, 50-250 in the example, the effect then averages the frequency values to calculate the bar length. `"FrequencyPointCount"` determines how many fft points will be generated in the specified frequency range.
This is what I think thermaltake different sound modes do, they target low/mid/high frequencies for different types of sound.

Example:
```json
        {
          "Type": "BarSpectrumSoundEffect",
          "Config": {
            "Mirror": true,
            "FrequencyPointCount": 4,
            "BackgroundColor": {
              "Full": [16, 16, 16]
            },
            "BarColor": {
              "Gradient": [
                [0, [0, 255, 0]],
                [1, [0, 0, 255]]
              ]
            },

            "UseAverage": false,
            "MinimumFrequency": 50,
            "MaximumFrequency": 250,
            "ScalingStrategy": "Linear",
            "ScalingFactor": 9,
            "IsXLogScale": false,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

closed 2020-12-20 by MoshiMoshi0