MoshiMoshi0/TTController #90: Effect Mixer

BillisC, 2020-12-05:
**Describe the feature you'd like**
One other cool idea would be to let the user combine music with spectrum. Music would work normally as intended while the color changes at the same time. For now with razer synapse it's easy to achieve something like this but it would be cool if no additional programs were needed.

MoshiMoshi0, 2020-12-05:
I dont want to add too many separate effects so I think a better way to do this would be to support blending multiple effects with some predefined blend functions.

MoshiMoshi0, 2020-12-06:
Try this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/36684182/artifacts
Example usage:
```json
        {
          "Type": "BlendEffect",
          "Config": {
            "BlendMode": "Multiply",
            "Effects": [
              {
                "Type": "WaveEffect",
                "Config": {
                  "Color": {
                    "Gradient": [
                      [0, [0, 0, 0]],
                      [1, [255, 255, 255]]
                    ]
                  }
                }
              },
              {
                "Type": "AuroraEffect"
              }
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

Note that for the effects passed to the `"BlendEffect"`, the `"Trigger"` and `"ColorGenerationMethod"` properties are not used and are not needed.

The `"BlendMode"` values are based on [photoshop](https://helpx.adobe.com/photoshop/using/blending-modes.html) but `"Multiply"` is what will be used 99% of the time.
You should be able to just replace the `"WaveEffect"` from above with `"SoundEffect"` to get what you want. Best to use black->white gradient only for the sound effect for the blending to work predictably. But you could use colors too to get some unusual results.

BillisC, 2020-12-06:
Very well I'll try it in a bit

BillisC, 2020-12-06:
Wow! This works really well with the sound mode indeed. It now looks really awesome nice job

MoshiMoshi0, 2020-12-06:
Nice! Thanks for the feature request.

closed 2020-12-06 by MoshiMoshi0