MoshiMoshi0/TTController #91: Configuration Help

bigchungus2222, 2020-12-06:
**Describe your desired configuration**
hi, really hoping someone can help.

I just want the stadard TT RGB temperature configuration change.

I.e. the fans change colour based on temp (I have 6 fans)
same with speed as well.

**Hardware info**
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 2009 [19041.1.amd64fre.vb_release.191206-1406]
»        Build: 0.11.0-beta+Branch.master.Sha.42768ba987b22e18ff8cdd114f754e5f7ce16d72
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»  [0x264a, 0x1fa7]: \\?\hid#vid_264a&pid_1fa7#8&341c1bff&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»  [0x264a, 0x1fa6]: \\?\hid#vid_264a&pid_1fa6#8&9ef146d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8103
»        Version: 2.0.2
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 30, Rpm: 433, Unknown: 255]
»                        Identifier: [9802, 8103, 1]
»                2:
»                        Data: [PortId: 2, Speed: 30, Rpm: 456, Unknown: 255]
»                        Identifier: [9802, 8103, 2]
»                3:
»                        Data: [PortId: 3, Speed: 30, Rpm: 416, Unknown: 255]
»                        Identifier: [9802, 8103, 3]
»                4:
»                        Data: [PortId: 4, Speed: 30, Rpm: 363, Unknown: 255]
»                        Identifier: [9802, 8103, 4]
»                5:
»                        Data: [PortId: 5, Speed: 30, Rpm: 413, Unknown: 255]
»                        Identifier: [9802, 8103, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8102
»        Version: 1.0.0
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 100, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 1]
»                2:
»                        Data: [PortId: 2, Speed: 100, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 2]
»                3:
»                        Data: [PortId: 3, Speed: 100, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 3]
»                4:
»                        Data: [PortId: 4, Speed: 30, Rpm: 539, Unknown: 255]
»                        Identifier: [9802, 8102, 4]
»                5:
»                        Data: [PortId: 5, Speed: 100, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 47.5 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 42.0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 15.0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 4.0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 43.0 °C
»
»  Intel Core i9-9900K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 50.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 53.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 49.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 50.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 47.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 50.0 °C
»                CPU Core #7 (/intelcpu/0/temperature/6): 50.0 °C
»                CPU Core #8 (/intelcpu/0/temperature/7): 52.0 °C
»                CPU Package (/intelcpu/0/temperature/8): 53.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/9): 50.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/10): 47.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/11): 51.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/12): 50.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/13): 53.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/14): 50.0 °C
»                CPU Core #7 Distance to TjMax (/intelcpu/0/temperature/15): 50.0 °C
»                CPU Core #8 Distance to TjMax (/intelcpu/0/temperature/16): 48.0 °C
»                Core Max (/intelcpu/0/temperature/17): 53.0 °C
»                Core Average (/intelcpu/0/temperature/18): 50.1 °C
»
»  NVIDIA GeForce GTX 1080 Ti:
»        Temperature:
»                GPU Core (/gpu-nvidia/0/temperature/0): 40.0 °C
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8101, 1]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/8"],
            "ColorGradient": [
              [40, [16, 16, 128]],
              [60, [16, 16, 16]],
              [86, [128, 16, 16]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8103, 1]
      ],
      "Speed": 35,
      "Colors": {
        "Full": [255, 0, 0]
      }
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8103, 1]],
      "Config": {
        "Name": "Top Fan",
        "ColorModifiers": [
          {
            "Type": "RotateLedColorModifier",
            "Config": {
              "Rotation": 11
            }
          },
          {
            "Type": "ReverseLedColorModifier",
            "Config": {
              "Reverse": true
            }
          }
        ]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/8"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

MoshiMoshi0, 2020-12-06:
The config you posted pretty much does what you want, you just needed to add correct port identifiers.

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8103, 1],
        [9802, 8103, 2],
        [9802, 8103, 3],
        [9802, 8103, 4],
        [9802, 8103, 5],
        [9802, 8102, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/8"],
            "ColorGradient": [
              [40, [16, 16, 128]],
              [60, [16, 16, 16]],
              [86, [128, 16, 16]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ]
}
```

You have to change `"CurvePoints"` to tweak the fan curve and `"ColorGradient"` to tweak the color based on temperature.

bigchungus2222, 2020-12-06:
so would i change "SenorEffect" to something else if I want it to do something else basically,

does this run at boot or is it something I have to manually click?

(sorry if this isnt exactly a configuration question

[image]

cool seems to be working!

MoshiMoshi0, 2020-12-06:
> so would i change "SenorEffect" to something else if I want it to do something else basically,

Yes but you also need to change the `"Config"` part as each effect has different settings.
Its all (hopefully) explained on the [docs](https://moshimoshi0.github.io/TTController/) page. Each plugin/effect has its own section with explained config variables.

> does this run at boot or is it something I have to manually click?

If you install it then yes it runs at boot, before you even login to your account.
But it can also run in the console mode, its used for quick testing.

> cool seems to be working!

Hmm, that does seem to match the color from the config I posted above. Did you change them?

bigchungus2222, 2020-12-06:
Yea I used what you posted

Tbh the original i had was just the 1 on the example page

I would have made more an effort however I spent 20 hours diagnosing why I was getting bsods to check tt rgb last :( appreciate the help my man I may recommend on this thread tomorrow however I think this should do it!

MoshiMoshi0, 2020-12-06:
Just to be sure, run in console mode (stop the service if its running, select console mode from menu) and check that the sensor is getting correct temperature and the fans are responding correctly. You will see the data printed every 5s.
The config I sent was set to output red at 86+ degrees.

bigchungus2222, 2020-12-08:
00:19:29.8450: Sensor "/intelcpu/0/temperature/8" value: 32.23256
00:19:34.8473: Port [9802, 8103, 1] data: [PortId: 1, Speed: 33, Rpm: 609, Unknown: 255]
00:19:34.8473: Port [9802, 8103, 2] data: [PortId: 2, Speed: 33, Rpm: 613, Unknown: 255]
00:19:34.8473: Port [9802, 8103, 3] data: [PortId: 3, Speed: 32, Rpm: 620, Unknown: 255]
00:19:34.8473: Port [9802, 8103, 4] data: [PortId: 4, Speed: 32, Rpm: 590, Unknown: 255]
00:19:34.8473: Port [9802, 8103, 5] data: [PortId: 5, Speed: 32, Rpm: 506, Unknown: 255]
00:19:34.8473: Port [9802, 8102, 4] data: [PortId: 4, Speed: 32, Rpm: 611, Unknown: 255]

MoshiMoshi0, 2020-12-08:
Ok, so that seems correct, only the color was incorrect.
I think you found a bug with the gradient calculation. Your temperature is lower than the first gradient stop (40) and it outputs the last stop color (86).

Try changing the gradient to this:
```json
            "ColorGradient": [
              [20, [16, 16, 128]],
              [40, [16, 16, 16]],
              [60, [128, 16, 16]]
            ],
```

It will go:
* 20C -> blue
* 40C -> white
* 60C -> red

bigchungus2222, 2020-12-08:
ok cool cheers man!

once the service is running do I need to keep any of these files?

if so I'll just move em to a secure location and save this thread

MoshiMoshi0, 2020-12-08:
Yes, dont move or delete any files or the service will fail to start. It uses the same executable as the one you use to install/uninstall etc.
If you want to move it, first uninstall it, move to new location, and install.

closed 2020-12-08 by MoshiMoshi0