MoshiMoshi0/TTController #98: Failed to start the service

FSergey78, 2020-12-14:
**Describe the issue**
Main Menu > Manage Service
================================
[1] Start
[2] Stop
[3] Restart
[4] Uninstall
[5] Install
[0] Back

[1, 4, 0]: 1
Starting the service...
Failed to start the service...
System.ServiceProcess.TimeoutException: Истекло время ожидания, но операция не была выполнена.
   в System.ServiceProcess.ServiceController.WaitForStatus(ServiceControllerStatus desiredStatus, TimeSpan timeout)
   в TTController.Service.Program.<ManageServiceMenu>g__StartService|6_1()
Press any key to return to the menu...

**Debug report**
```
Info
-------------------------------
>
>        OS: Windows 10 Enterprise 2009 [19041.1.amd64fre.vb_release.191206-1406]
>        Build: 0.11.0-beta+Branch.master.Sha.42768ba987b22e18ff8cdd114f754e5f7ce16d72
>
-------------------------------
Applications
-------------------------------
>
>
-------------------------------
HID
-------------------------------
>
>  [0x264a, 0x2135]: \\?\hid#vid_264a&pid_2135#8&178f2f2f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
>  [0x264a, 0x1f43]: \\?\hid#vid_264a&pid_1f43#8&108679a0&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
>  [0x264a, 0x1f42]: \\?\hid#vid_264a&pid_1f42#9&ca33009&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
>
-------------------------------
Controllers
-------------------------------
>
>  Name: Riing
>        VendorId: 9802
>        ProductId: 8003
>        Version: 1.0.0
>        Ports:
>                1:
>                        Data: [PortId: 1, Speed: 30, Rpm: 313, Unknown: 3]
>                        Identifier: [9802, 8003, 1]
>                2:
>                        Data: [PortId: 2, Speed: 30, Rpm: 313, Unknown: 3]
>                        Identifier: [9802, 8003, 2]
>                3:
>                        Data: [PortId: 3, Speed: 30, Rpm: 364, Unknown: 3]
>                        Identifier: [9802, 8003, 3]
>                4:
>                        Data:
>                        Identifier: [9802, 8003, 4]
>                5:
>                        Data:
>                        Identifier: [9802, 8003, 5]
>        Available effect types: Flow, Full, PerLed
>  Name: Riing
>        VendorId: 9802
>        ProductId: 8002
>        Version: 1.0.0
>        Ports:
>                1:
>                        Data: [PortId: 1, Speed: 30, Rpm: 0, Unknown: 3]
>                        Identifier: [9802, 8002, 1]
>                2:
>                        Data: [PortId: 2, Speed: 30, Rpm: 0, Unknown: 3]
>                        Identifier: [9802, 8002, 2]
>                3:
>                        Data: [PortId: 3, Speed: 30, Rpm: 475, Unknown: 3]
>                        Identifier: [9802, 8002, 3]
>                4:
>                        Data:
>                        Identifier: [9802, 8002, 4]
>                5:
>                        Data:
>                        Identifier: [9802, 8002, 5]
>        Available effect types: Flow, Full, PerLed
>  Name: Riing Trio
>        VendorId: 9802
>        ProductId: 8501
>        Version: 3.0.1
>        Ports:
>                1:
>                        Data: [PortId: 1, Speed: 72, Rpm: 1251, Unknown: 255]
>                        Identifier: [9802, 8501, 1]
>                2:
>                        Data: [PortId: 2, Speed: 72, Rpm: 1233, Unknown: 255]
>                        Identifier: [9802, 8501, 2]
>                3:
>                        Data: [PortId: 3, Speed: 72, Rpm: 1208, Unknown: 255]
>                        Identifier: [9802, 8501, 3]
>                4:
>                        Data: [PortId: 4, Speed: 72, Rpm: 1228, Unknown: 255]
>                        Identifier: [9802, 8501, 4]
>                5:
>                        Data: [PortId: 5, Speed: 58, Rpm: 951, Unknown: 255]
>                        Identifier: [9802, 8501, 5]
>        Available effect types: PerLed
>
-------------------------------
Sensors
-------------------------------
>
>  Nuvoton NCT6798D:
>        Temperature:
>                Temperature #1 (/lpc/nct6798d/temperature/1): 37,5 °C
>                Temperature #2 (/lpc/nct6798d/temperature/2): 39,0 °C
>                Temperature #3 (/lpc/nct6798d/temperature/3): 29,0 °C
>                Temperature #4 (/lpc/nct6798d/temperature/4): 12,0 °C
>                Temperature #5 (/lpc/nct6798d/temperature/5): 17,0 °C
>                Temperature #6 (/lpc/nct6798d/temperature/6): 52,0 °C
>
>  Intel Core i9-9900K:
>        Temperature:
>                CPU Core #1 (/intelcpu/0/temperature/0): 37,0 °C
>                CPU Core #2 (/intelcpu/0/temperature/1): 44,0 °C
>                CPU Core #3 (/intelcpu/0/temperature/2): 39,0 °C
>                CPU Core #4 (/intelcpu/0/temperature/3): 39,0 °C
>                CPU Core #5 (/intelcpu/0/temperature/4): 42,0 °C
>                CPU Core #6 (/intelcpu/0/temperature/5): 37,0 °C
>                CPU Core #7 (/intelcpu/0/temperature/6): 41,0 °C
>                CPU Core #8 (/intelcpu/0/temperature/7): 42,0 °C
>                CPU Package (/intelcpu/0/temperature/8): 45,0 °C
>                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/9): 63,0 °C
>                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/10): 56,0 °C
>                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/11): 61,0 °C
>                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/12): 61,0 °C
>                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/13): 58,0 °C
>                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/14): 63,0 °C
>                CPU Core #7 Distance to TjMax (/intelcpu/0/temperature/15): 59,0 °C
>                CPU Core #8 Distance to TjMax (/intelcpu/0/temperature/16): 58,0 °C
>                Core Max (/intelcpu/0/temperature/17): 44,0 °C
>                Core Average (/intelcpu/0/temperature/18): 40,1 °C
>
>  NVIDIA GeForce RTX 2080 Ti:
>        Temperature:
>                GPU Core (/gpu-nvidia/0/temperature/0): 33,0 °C
>
>  Samsung SSD 840 PRO Series:
>        Temperature:
>                Temperature (/ssd/3/temperature/0): 38,0 °C
>
>  ST4000VN000-1H4168:
>        Temperature:
>                Temperature (/hdd/2/temperature/0): 43,0 °C
>
>  Samsung SSD 850 EVO 2TB:
>        Temperature:
>                Temperature (/ssd/1/temperature/0): 46,0 °C
>
>  Samsung SSD 960 EVO 250GB:
>        Temperature:
>                Temperature (/nvme/0/temperature/0): 47,0 °C
>                Temperature 1 (/nvme/0/temperature/6): 47,0 °C
>                Temperature 2 (/nvme/0/temperature/7): 54,0 °C
>
-------------------------------
Press any key to continue...

```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 80],
              [80, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 0, 139],
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Boot",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "Speed": 35
    },
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "Speed": 35
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

MoshiMoshi0, 2020-12-14:
Run the service in console mode and post the output.

FSergey78, 2020-12-14:
thanks for the quick reply, here is the data from the console

20:40:24.5240: ================================================================
20:40:24.5460: Initializing service, version "0.11.0-beta+Branch.master.Sha.42768ba987b22e18ff8cdd114f754e5f7ce16d72"
20:40:24.5589: Loading plugin assembly: TTController.Plugin.DpsgController [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.Level20DeskController [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.RiingController [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.RiingQuadController [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.AuroraEffect [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.ColorSequenceEffect [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.FlowEffect [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.IpcEffect [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.PingPongEffect [0.11.0.0]
20:40:24.5589: Loading plugin assembly: TTController.Plugin.PulseEffect [0.11.0.0]
20:40:24.5739: Loading plugin assembly: TTController.Plugin.RawEffect [0.11.0.0]
20:40:24.5739: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.11.0.0]
20:40:24.5739: Loading plugin assembly: TTController.Plugin.RippleEffect [0.11.0.0]
20:40:24.5739: Loading plugin assembly: TTController.Plugin.SensorEffect [0.11.0.0]
20:40:24.5739: Loading plugin assembly: TTController.Plugin.SoundEffect [0.11.0.0]
20:40:24.5739: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.11.0.0]
20:40:24.5739: Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.11.0.0]
20:40:24.5739: Loading plugin assembly: TTController.Plugin.WaveEffect [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.ZoneEffect [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.LerpLedColorModifier [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.ReverseLedColorModifier [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.RotateLedColorModifier [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.TrimLedColorModifier [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.IpcSpeedController [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.IpcTrigger [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.11.0.0]
20:40:24.5898: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.11.0.0]
20:40:24.6048: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.11.0.0]
20:40:24.6048: Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.11.0.0]
20:40:24.6048: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.11.0.0]
20:40:24.6048: Loading plugin assembly: TTController.Plugin.WebSocketIpcServer [0.11.0.0]
20:40:24.6048: Creating Config Manager...
20:40:24.6716: Loading config...
20:40:24.8372: Newtonsoft.Json.JsonReaderException: Invalid plugin name "FullColorEffect"
   в TTController.Service.Config.Converters.AbstractPluginConverter`2.ReadJson(JsonReader reader, Type objectType, TPlugin existingValue, Boolean hasExistingValue, JsonSerializer serializer)
   в Newtonsoft.Json.JsonConverter`1.ReadJson(JsonReader reader, Type objectType, Object existingValue, JsonSerializer serializer)
   в Newtonsoft.Json.Serialization.JsonSerializerInternalReader.DeserializeConvertable(JsonConverter converter, JsonReader reader, Type objectType, Object existingValue)
   в Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   в Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   в Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   в Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
[... cut ...]
   в Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   в Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   в Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   в Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   в TTController.Service.Managers.ConfigManager.LoadOrCreateConfig()
20:40:24.8462: Failed to load the config!

MoshiMoshi0, 2020-12-14:
> 20:40:24.8372: Newtonsoft.Json.JsonReaderException: Invalid plugin name "FullColorEffect"

It says what is wrong, there is no `"FullColorEffect"`, you must have updated from old version.
You want to use [StaticColorEffect](https://moshimoshi0.github.io/TTController/plugins/effects/staticcoloreffect).

FSergey78, 2020-12-14:
I don't know English, it's hard to understand through a translator
I did not update from the old version, I downloaded the TTController_0.11.0.2-beta version here
can you specifically prescribe what is needed to illuminate the fans with white light?

MoshiMoshi0, 2020-12-14:
Oh, you should have said that you just grabbed random config, I thought it was yours.

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8003, 1],
        [9802, 8003, 2],
        [9802, 8003, 3],
        [9802, 8002, 1],
        [9802, 8002, 2],
        [9802, 8002, 3],
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8501, 4],
        [9802, 8501, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 80],
              [80, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "StaticColorEffect",
          "Config": {
            "Color": {
              "Full": [255, 255, 255]
            },
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/8"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8003, 1],
        [9802, 8003, 2],
        [9802, 8003, 3],
        [9802, 8002, 1],
        [9802, 8002, 2],
        [9802, 8002, 3]
      ],
      "Config": {
        "DeviceType": "Riing"
      }
    },
    {
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8501, 4],
        [9802, 8501, 5]
      ],
      "Config": {
        "DeviceType": "RiingTrio"
      }
    }
  ]
}
```

You have 4x Riing fan and 5x Riing Trio fan, but I dont know what is connected to `[9802, 8002, 1]` and `[9802, 8002, 2]`, it has no rpm so its a pump or light strip?

MoshiMoshi0, 2020-12-14:
Also, this is not really an issue, im moving this to [discussions](https://github.com/MoshiMoshi0/TTController/discussions)

closed 2020-12-14 by MoshiMoshi0