MoshiMoshi0/TTController #117: faulty color mapping in Razer Studio

mihi83, 2021-01-09:
Chroma Connect to Razer Synapse basically works, im able to see the TT Controller Plugin in chroma studio.
But the mapping to my LEDs of the fans (RiingQuad as well as Riing Trio) is wrong, since only about a quarter of the LEDs are lit up in the outer rim of each fan.

{
  "Profiles": [
	{
		"Name": "Default",
		"Ports": [
			[9802, 8800, 1],
			[9802, 8800, 2],
			[9802, 8800, 3],
			[9802, 8800, 4],
			[9802, 8800, 5],
			[9802, 8801, 1],
			[9802, 8801, 2],
			[9802, 8801, 3],
			[9802, 8801, 4],
			[9802, 8801, 5]
		],

		"SpeedControllers": [
		
			{
			  "Type": "PwmSpeedController", 
			  "Config": {
				"CurvePoints": [
				  [20, 0],
				  [30, 20],
				  [40, 25],
				  [50, 30],
				  [60, 45],
				  [70, 65],
				  [80, 100]
				],
				"Sensors": ["/amdcpu/0/temperature/2"],
				"Trigger": {
				  "Type": "AlwaysTrigger"
				}
			  }
        }
		
		],
		"Effects": [
				
			{
			  "Type": "RazerConnectEffect",
			  "Config": {
				"Layer": "Custom",

				"ColorGenerationMethod": "PerPort",
				"Trigger": {
				
					"Type": "AlwaysTrigger"
				
				}
			  }
			},
			
			
		
		]
}
  ],
  "ComputerStateProfiles": [
  
	{
	"StateType":  "Boot",
	"Ports": [
	
		[9802, 8800, 1],
		[9802, 8800, 2],
		[9802, 8800, 3],
		[9802, 8800, 4],
		[9802, 8800, 5],
		[9802, 8801, 1],
		[9802, 8801, 2],
		[9802, 8801, 3],
		[9802, 8801, 4],
		[9802, 8801, 5]
	
	],
	"Speed": 50,
	"EffectType": "PerLed",
	"Color": {
		"Full": [255, 0, 0]
		}
	}
  
  ],
  "PortConfigs": [
  
	{
	  "Ports": [
	  
		[9802, 8800, 1],
		[9802, 8800, 2],
		[9802, 8800, 3]
	  
	  ],
	  "Config": {
		"Name": "Bottom Fans",
		"DeviceType": "RiingQuad",
		"IgnoreColorCache": true,
		"IgnoreSpeedCache": true
	  }
	},
	
	{
	  "Ports": [
	  
		[9802, 8800, 4],
		[9802, 8800, 5],
		[9802, 8801, 1]
	  
	  ],
	  "Config": {
		"Name": "Side Fans",
		"DeviceType": "RiingQuad",
		"IgnoreColorCache": true,
		"IgnoreSpeedCache": true
	  }
	},
	
	{
	  "Ports": [
	  
		[9802, 8801, 2],
		[9802, 8801, 3],
		[9802, 8801, 4]
	  
	  ],
	  "Config": {
		"Name": "Top Fans",
		"DeviceType": "RiingTrio",
		"IgnoreColorCache": true,
		"IgnoreSpeedCache": true
	  }
	},
	
	{
	  "Ports": [
	  
		[9802, 8801, 5]
		
	  
	  ],
	  "Config": {
		"Name": "Back Fan",
		"DeviceType": "RiingTrio",
		"IgnoreColorCache": true,
		"IgnoreSpeedCache": true
	  }
	}
	
  
  ],
  "SensorConfigs": [
  
	{
	  "Sensors": ["/amdcpu/0/temperature/2"],
	  "Config": {
		"Offset": 0,
		"CriticalValue": 85
	  }
	}
  
  ]
}

MoshiMoshi0, 2021-01-09:
That is how many colors razer provides, you need to stretch them to match the fan with [ColorModifiers](https://moshimoshi0.github.io/TTController/config/port#colormodifiers) + [LerpLedColorModifier](https://moshimoshi0.github.io/TTController/plugins/modifiers/lerpledcolormodifier)

mihi83, 2021-01-09:
Thanks, I will take a look into it and get back to you

mihi83, 2021-01-12:
Allright, I basically got it to work thanks.
But there is another question popping up, regarding the Chroma connect:
As far as I understand this the software basically stretches the colors received by Chroma to match the fans led count. When I configure a wave pattern this results into like 4 colors on my Quad fans (1 color per led-zone).
What Im trying to achieve is a wave like pattern (almost like the standard pattern of the TT fans), so are there any settings to mess around with to achieve this?

MoshiMoshi0, 2021-01-12:
Not sure I understand what you want to do. For the gradient to be somewhat working with zones you need to include zone led counts, for riing quad the gradient should be divided in 33%, 33%, 16%, 16% ratio, but it will not look right anyways.

The default thermaltake effect will be comparable to `AuroraEffect`. 
If you want each zone to be the same you would have to use `ZoneEffect` with 4x `AuroraEffect` with each `Length` matching the zone led count.

For now you cant make `RazerConnectEffect` output one color to each zone, it should be possible when more effect modifiers are added.

mihi83, 2021-01-12:
Ok, maybe I mixed some thing up...
Short version: what Im basically try to achieve is the AuroraEffect on my RiingQuad fans with the RazerConnectEffect (since I want to be able to maintain all my RGB effects via Razer Synapse).

MoshiMoshi0, 2021-01-12:
Thats not really possible since razer only outputs 4 colors. Best you can do is just to stretch those 4 colors to 54 leds.
You could maybe use `ZoneEffect` with 4x `RazerConnectEffect`, but it will still look pretty bad.

mihi83, 2021-01-12:
Too bad... I based my build around this functionality
But just theoretically: It should be kinda possible, if Razer sends 4 colors to the fan (for example red, green, blue and yellow) and TTController duplicates them (like for the Riing Quad 4 Times, or for the Trio 3 times) and stretches them across all the LEDs.

So basically what the fans would be display is "red, green, blue, yellow, red, green, blue, yellow, red, green, blue, yellow, red, green, blue, yellow".

MoshiMoshi0, 2021-01-12:
> So basically what the fans would be display is "red, green, blue, yellow, red, green, blue, yellow, red, green, blue, yellow, red, green, blue, yellow".

If you mean that want to copy those 4 colors until they fill the fan then there was a modifier like this but I removed it. I will be adding a bunch of new modifiers so i'll add it back.

> Is it possible to activate different Effects based on different programs running?

Add another effect with a [ProcessTrigger](https://moshimoshi0.github.io/TTController/plugins/triggers/processtrigger) and place it as first in the list of effects.

mihi83, 2021-01-12:
Allright, nice...

I also figured out the ProcessTrigger, took me a while.
As a matter of fact one need to use the name of the .exe file, not the process naem.

closed 2021-01-15 by MoshiMoshi0