MoshiMoshi0/TTController #139: Not all leds of Riing Trio 20 are working

MoshiMoshi0, 2021-02-19:


MoshiMoshi0, 2021-02-19:
Fixed build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/37862784/artifacts

jtarshis, 2021-02-19:
Just tried this build. I assume I just copy my old config.json over right? Currently only half of the LEDs on the hub are lit up.

MoshiMoshi0, 2021-02-19:
I just realized that you have Riing Quad controllers not Riing Trio controllers, so the fix was unnecessary for you and you could make it work on the version you were using. But it wouldn't be possible if you had Riing Trio controllers.

Anyways, with your old config you need to set some of the ports to RiingTrio20 device, I have no way of knowing which ones are Riing Trio 20 fan but just looking at the fan rpm i think those two. Replace old PortConfigs section with this:

```json
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 3],
        [9802, 8801, 1],
        [9802, 8801, 2],
        [9802, 8801, 3]
      ],
      "Config": {
        "DeviceType": "RiingTrio",
        "IgnoreColorCache": true
      }
    },
    {
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4]
      ],
      "Config": {
        "DeviceType": "RiingTrio20",
        "IgnoreColorCache": true
      }
    }
  ],
```

jtarshis, 2021-02-19:
OK so I copied the new portconfig into 0.11.0.2-beta and 0.11.0.0-fix. Now none of the LEDs on the hub are lit up and it looks like the LEDs on the back outer ring aren't lit up either.

MoshiMoshi0, 2021-02-19:
Post full `config.json` contents

jtarshis, 2021-02-19:
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4],
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 3],
        [9802, 8801, 1],
        [9802, 8801, 2],
        [9802, 8801, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 80],
              [80, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/3"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
            "Type": "AuroraEffect",
          "Config": {
            "Step": 0.005,
            "Length": 30,
            "Mirror": false,
            "Brightness": 1,
            "Saturation": 1,
            "Trigger": {
              "Type": "AlwaysTrigger"
             
                }
            }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 3],
        [9802, 8801, 1],
        [9802, 8801, 2],
        [9802, 8801, 3]
      ],
      "Config": {
        "DeviceType": "RiingTrio",
        "IgnoreColorCache": true
      }
    },
    {
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4]
      ],
      "Config": {
        "DeviceType": "RiingTrio20",
        "IgnoreColorCache": true
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/amdcpu/0/temperature/3"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}

MoshiMoshi0, 2021-02-19:
Run the fixed version in console mode, and post the console output.

jtarshis, 2021-02-19:
16:36:15.3171: ================================================================
16:36:15.3351: Initializing service, version "0.11.0-fix-riing-trio.1+36.Branch.fix-riing-trio.Sha.4a0c3cab488ba66bb45a31e46de4c41617f46453"
16:36:15.6094: Loading plugin assembly: TTController.Plugin.DpsgController [0.11.0.0]
16:36:15.6154: Loading plugin assembly: TTController.Plugin.Level20DeskController [0.11.0.0]
16:36:15.6154: Loading plugin assembly: TTController.Plugin.RiingController [0.11.0.0]
16:36:15.6154: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.11.0.0]
16:36:15.6154: Loading plugin assembly: TTController.Plugin.RiingQuadController [0.11.0.0]
16:36:15.6154: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.11.0.0]
16:36:15.6334: Loading plugin assembly: TTController.Plugin.AuroraEffect [0.11.0.0]
16:36:15.6334: Loading plugin assembly: TTController.Plugin.BlendEffect [0.11.0.0]
16:36:15.6334: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.11.0.0]
16:36:15.6334: Loading plugin assembly: TTController.Plugin.ColorSequenceEffect [0.11.0.0]
16:36:15.6459: Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.11.0.0]
16:36:15.6459: Loading plugin assembly: TTController.Plugin.FlowEffect [0.11.0.0]
16:36:15.6459: Loading plugin assembly: TTController.Plugin.IpcEffect [0.11.0.0]
16:36:15.6459: Loading plugin assembly: TTController.Plugin.PingPongEffect [0.11.0.0]
16:36:15.6459: Loading plugin assembly: TTController.Plugin.PulseEffect [0.11.0.0]
16:36:15.6621: Loading plugin assembly: TTController.Plugin.RawEffect [0.11.0.0]
16:36:15.6621: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.11.0.0]
16:36:15.6621: Loading plugin assembly: TTController.Plugin.RippleEffect [0.11.0.0]
16:36:15.6621: Loading plugin assembly: TTController.Plugin.SensorEffect [0.11.0.0]
16:36:15.6831: Loading plugin assembly: TTController.Plugin.SoundEffect [0.11.0.0]
16:36:15.6831: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.11.0.0]
16:36:15.6831: Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.11.0.0]
16:36:15.6943: Loading plugin assembly: TTController.Plugin.WaveEffect [0.11.0.0]
16:36:15.6943: Loading plugin assembly: TTController.Plugin.ZoneEffect [0.11.0.0]
16:36:15.6943: Loading plugin assembly: TTController.Plugin.DelayPortModifier [0.11.0.0]
16:36:15.6943: Loading plugin assembly: TTController.Plugin.LerpPortModifier [0.11.0.0]
16:36:15.6943: Loading plugin assembly: TTController.Plugin.ReversePortModifier [0.11.0.0]
16:36:15.7103: Loading plugin assembly: TTController.Plugin.RotatePortModifier [0.11.0.0]
16:36:15.7103: Loading plugin assembly: TTController.Plugin.TrimPortModifier [0.11.0.0]
16:36:15.7103: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.11.0.0]
16:36:15.7103: Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.11.0.0]
16:36:15.7236: Loading plugin assembly: TTController.Plugin.IpcSpeedController [0.11.0.0]
16:36:15.7236: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.11.0.0]
16:36:15.7236: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.11.0.0]
16:36:15.7236: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.11.0.0]
16:36:15.7236: Loading plugin assembly: TTController.Plugin.IpcTrigger [0.11.0.0]
16:36:15.7417: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.11.0.0]
16:36:15.7417: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.11.0.0]
16:36:15.7417: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.11.0.0]
16:36:15.7417: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.11.0.0]
16:36:15.7563: Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.11.0.0]
16:36:15.7563: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.11.0.0]
16:36:15.7563: Loading plugin assembly: TTController.Plugin.WebSocketIpcServer [0.11.0.0]
16:36:15.7563: Creating Config Manager...
16:36:15.8162: Loading config...
16:36:15.9633: Creating DataCache...
16:36:15.9633: Creating Plugin Store...
16:36:15.9743: Loaded plugin "AuroraEffect"
16:36:15.9743: Loaded plugin "AlwaysTrigger"
16:36:15.9743: Loaded plugin "PwmSpeedController"
16:36:15.9743: Loaded plugin "AlwaysTrigger"
16:36:15.9743: Loaded 4 plugins
16:36:15.9743: Creating Sensor Manager...
16:36:15.9743: Initializing Libre Hardware Monitor...
16:36:16.3145: Detected 68 sensors
16:36:16.3175: Creating Device Manager...
16:36:16.3485: Searching for controller changes
16:36:16.3485: Searching for "Dpsg" controllers
16:36:16.5592: Searching for "Level 20 Desk" controllers
16:36:16.5592: Searching for "Riing" controllers
16:36:16.5592: Searching for "Riing Plus" controllers
16:36:16.5592: Searching for "Riing Quad" controllers
16:36:16.5852: Initialized "Riing Quad" controller [9802, 8800], version: "3.0.3"
16:36:16.5993: Initialized "Riing Quad" controller [9802, 8801], version: "3.0.3"
16:36:16.6193: Initialized "Riing Quad" controller [9802, 8802], version: "3.0.3"
16:36:16.6193: Searching for "Riing Trio" controllers
16:36:16.6193: Enabling sensor: /amdcpu/0/temperature/3
16:36:16.6193: Processing profile "Default"
16:36:16.6313: Assigned plugin "AuroraEffect" to "Default" profile
16:36:16.6313: Assigned plugin "PwmSpeedController" to "Default" profile
16:36:16.6549: Applying computer state profile: Boot
16:36:16.6549: Creating Timer Manager...
16:36:16.6616: Starting 4 timers...
[... cut ...]
16:36:23.0056: Port [9802, 8800, 2] data: [PortId: 2, Speed: 50, Rpm: 890, Unknown: 255]
16:36:23.0056: Port [9802, 8800, 3] data: [PortId: 3, Speed: 50, Rpm: 866, Unknown: 255]
16:36:23.0056: Port [9802, 8801, 1] data: [PortId: 1, Speed: 50, Rpm: 938, Unknown: 255]
16:36:23.0056: Port [9802, 8801, 2] data: [PortId: 2, Speed: 50, Rpm: 907, Unknown: 255]
16:36:23.0056: Port [9802, 8801, 3] data: [PortId: 3, Speed: 50, Rpm: 849, Unknown: 255]
16:36:23.0056: Sensor "/amdcpu/0/temperature/3" value: 42.2281

MoshiMoshi0, 2021-02-20:
Everything looks correct. I dont think I understand what is wrong.
What do you mean by
> none of the LEDs on the hub are lit up and it looks like the LEDs on the back outer ring aren't lit up either.

Are there no colors on any fan? Are there no colors only on trio 20 fan? Are there some colors on the trio 20 fan?

jtarshis, 2021-02-20:
I uploaded some pics, here is the link: [https://1drv.ms/u/s!Apv6DVIE61GihIYH49R_QWsJsps5_g?e=NeRdml]

You'll notice that the Trio12 fans I've referenced have different colors of lighting on the front and rear parts of the outer ring. On the Trio20 fans, it appears that only the front ring is being lit, the rear ring is showing the light from the front ring. And you can see the hub in the Trio20 front image isn't lit either.

MoshiMoshi0, 2021-02-20:
Oh I see... I thought all of the leds were off.
Are you sure you edited the config in the fixed version directory?

Also set the profile ports to:
```json
      "Name": "Default",
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4]
      ],
```

Only the front trio 20 fans should have updating colors.

jtarshis, 2021-02-20:
Yep, the config file I posted was from the fixed version's config file.

MoshiMoshi0, 2021-02-20:
Did you made the change i posted above

jtarshis, 2021-02-20:
Oh ok I think didn't understand what you meant, which was to do this.

 {
      "Name": "Default",
      "Ports": [
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 3],
        [9802, 8801, 1],
        [9802, 8801, 2],
        [9802, 8801, 3]
      ],
        "Name": "Default",
        "Ports": [
            [9802, 8802, 1],
            [9802, 8802, 4]
      ],

I did make the change and restarted it. The lighting is now on both front and rear rings of the Trio 20s, but hub is still half lit. I added a new photo to the link above that shows the effect.

MoshiMoshi0, 2021-02-20:
No, you were supposed to change the list of ports from:
```
        [9802, 8802, 1],
        [9802, 8802, 4],
        [9802, 8800, 1],
        [9802, 8800, 2],
        [9802, 8800, 3],
        [9802, 8801, 1],
        [9802, 8801, 2],
        [9802, 8801, 3]
```
to
```
        [9802, 8802, 1],
        [9802, 8802, 4]
```

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4]
      ],
...
```

jtarshis, 2021-02-20:
OK I updated the file but the result is still the same. Should I have also removed the other hubs from PortConfigs?

MoshiMoshi0, 2021-02-20:
> the result is still the same. 

The same result meaning?

> Should I have also removed the other hubs from PortConfigs?

No.

Also which fan colors are moving, only 2 front riing trio 20?

jtarshis, 2021-02-20:
What I meant by the result being the same was that both front and rear rings of the Trio 20s are working again but only half of the hub LEDs are lit.

MoshiMoshi0, 2021-02-20:
Please respond to all questions I ask not only the first one.
Are the only fans with moving colors the front trio 20?

jtarshis, 2021-02-20:
So I have six Trio 12s and two Trio 20s. The Trio 12s are working as intended with the same color scheme and movement. The Trio 20s also have the same color scheme and movement, however the fan hub on them only has half of the LEDs lit.

MoshiMoshi0, 2021-02-20:
Use this config and post what happens with riing trio 20 fans:

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4]
      ],
      "Effects": [
        {
          "Type": "ZoneEffect",
          "Config": {
            "Effects": [
              {
                "Type": "StaticColorEffect",
                "Config": {
                  "Color": {
                    "Full": [255, 0, 0]
                  }
                }
              },
              {
                "Type": "StaticColorEffect",
                "Config": {
                  "Color": {
                    "Full": [0, 255, 0]
                  }
                }
              },
              {
                "Type": "StaticColorEffect",
                "Config": {
                  "Color": {
                    "Full": [0, 0, 255]
                  }
                }
              }
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4]
      ],
      "Config": {
        "DeviceType": "RiingTrio20",
        "IgnoreColorCache": true
      }
    }
  ]
}
```

jtarshis, 2021-02-20:
I tried that config and both front and rear outer rings are lit up, the fan hub is still only using half of the LEDs.

MoshiMoshi0, 2021-02-20:
Is one outer ring full red, second outer ring full green and inner half blue?

jtarshis, 2021-02-20:
OK I didn't realize my computer was running both the 0.11.0.0-fx and 0.11.0.2 beta. I removed the beta for now. After changing the config the outer ring (both front and rear) is sold red and the hub is solid green with all LEDs lit up.

jtarshis, 2021-02-20:
So I also tried just removing the entire effects section as a whole, and I actually really like just the default effect. All the fans are illuminated properly and I believe (correct me if I'm wrong) I can just use the controller for PWM control.

I'm more than happy to keep testing things for you as I imagine there are others that are having the same issue with their Trio 20s.

MoshiMoshi0, 2021-02-20:
Stop the service, edit `TTController.Service.exe.Config` file and change line:
`<logger name="*" minLevel="Info" writeTo="file" enabled="true" />`
to:
`<logger name="*" minLevel="Trace" writeTo="file" enabled="true" />`

Then start the service in console mode wait 15s then exit and post latest log file in "Logs" folder.

jtarshis, 2021-02-20:
2021-02-20 11:46:08.1837|INFO|TTController.Service.TTService|================================================================
2021-02-20 11:46:08.2027|INFO|TTController.Service.TTService|Initializing service, version "0.11.0-fix-riing-trio.1+36.Branch.fix-riing-trio.Sha.4a0c3cab488ba66bb45a31e46de4c41617f46453"
2021-02-20 11:46:08.4253|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgController [0.11.0.0]
2021-02-20 11:46:08.4253|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.Level20DeskController [0.11.0.0]
2021-02-20 11:46:08.4253|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingController [0.11.0.0]
2021-02-20 11:46:08.4253|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingPlusController [0.11.0.0]
2021-02-20 11:46:08.4253|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingQuadController [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingTrioController [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AuroraEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlendEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ColorSequenceEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.IpcEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PingPongEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RippleEffect [0.11.0.0]
2021-02-20 11:46:08.4323|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorEffect [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SoundEffect [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.WaveEffect [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ZoneEffect [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DelayPortModifier [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.LerpPortModifier [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ReversePortModifier [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RotatePortModifier [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.TrimPortModifier [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopySpeedController [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.11.0.0]
2021-02-20 11:46:08.4524|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.IpcSpeedController [0.11.0.0]
2021-02-20 11:46:08.4644|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.11.0.0]
2021-02-20 11:46:08.4644|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.11.0.0]
2021-02-20 11:46:08.4644|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.11.0.0]
2021-02-20 11:46:08.4644|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.IpcTrigger [0.11.0.0]
[... cut ...]
2021-02-20 11:55:50.8308|INFO|TTController.Service.Utils.PluginStore|Disposing plugin "StaticColorEffect"
2021-02-20 11:55:50.8308|INFO|TTController.Service.Utils.PluginStore|Disposing plugin "StaticColorEffect"
2021-02-20 11:55:50.8378|INFO|TTController.Service.Utils.PluginStore|Disposing plugin "StaticColorEffect"
2021-02-20 11:55:50.8378|INFO|TTController.Service.TTService|Finalizing done!
2021-02-20 11:55:50.8378|INFO|TTController.Service.TTService|================================================================

MoshiMoshi0, 2021-02-20:
This is not the latest log, if you made that change in `TTController.Service.exe.Config` it will be very big. Attach the whole file instead of pasting it.

jtarshis, 2021-02-20:
[log.20210220.5.log](https://github.com/MoshiMoshi0/TTController/files/6016113/log.20210220.5.log)

MoshiMoshi0, 2021-02-20:
can you set the config to this and post pictures of how the riing trio 20 fans look?

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4]
      ],
      "Effects": [
        {
          "Type": "ZoneEffect",
          "Config": {
            "Effects": [
              {
                "Type": "StaticColorEffect",
                "Config": {
                  "Color": {
                    "Full": [255, 0, 0]
                  }
                }
              },
              {
                "Type": "StaticColorEffect",
                "Config": {
                  "Color": {
                    "Full": [255, 255, 255]
                  }
                }
              },
              {
                "Type": "StaticColorEffect",
                "Config": {
                  "Color": {
                    "Full": [0, 0, 255]
                  }
                }
              }
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8802, 1],
        [9802, 8802, 4]
      ],
      "Config": {
        "DeviceType": "RiingTrio20",
        "IgnoreColorCache": true
      }
    }
  ]
}
```

jtarshis, 2021-02-20:
OK uploaded new photos. [https://1drv.ms/u/s!Apv6DVIE61GihIYH49R_QWsJsps5_g?e=1VK6Rb](url)

MoshiMoshi0, 2021-02-20:
I dont understand what is happening to be honest.
If you still want to help I would need to see the data the official software sends.

Here is a step by step guide:

1. Install [Wireshark](https://www.wireshark.org/download.html) with usbpcap (you will be asked if you want to install usbpcap in the installer)
1. Start Wireshark
1. Stop the TTController and official software (if running)
1. Double click on USBPCap1 capture on the main screen in Wireshark
1. Start the official software
1. Set the one of riing trio 20 fans to: 
   1. first outer riing = full red
   2. second outer ring = full white
   3. inner ring = full blue
1. After 10s stop the capture in wireshark using the red square button
1. Save and upload the dump using `File -> Save As`

Thanks.

MoshiMoshi0, 2021-03-05:
Are you able to provide the above data?
Thanks.