MoshiMoshi0/TTController #146: Level 20 Mouse Pad help

JesseHolwell, 2021-03-14:
Hey, first off - this software looks fantastic, amazing job.

I have a ThermalTake Level 20 RGB Extended Gaming Mouse Pad that I am trying to integrate into my home automation system so that I can set the lights with a batch file, and the IPC server trigger looks like it's going to get me there.

However the device is not supported, so I have attempted to add support to your codebase but I am hitting a wall and not sure how to proceed. I am assuming I have not configured the device correctly and am hoping you can give me some guidance.

First time using wireshark, but I can see the packet that is setting the setup data, just unsure which values are important here.

Crash log:
```
12:11:19.5146: Creating Device Manager...
12:11:19.6047: Searching for controller changes
12:11:19.6097: Searching for "Dpsg" controllers
12:11:19.7138: Searching for "Level 20 Desk" controllers
12:11:19.7138: Searching for "Level 20 Mouse Pad" controllers
Service failed with unhandled exception:
System.ArgumentOutOfRangeException: Non-negative number required.
Parameter name: length
   at System.Array.Copy(Array sourceArray, Int32 sourceIndex, Array destinationArray, Int32 destinationIndex, Int32 length, Boolean reliable)
   at System.Array.Copy(Array sourceArray, Int32 sourceIndex, Array destinationArray, Int32 destinationIndex, Int32 length)
   at TTController.Service.Hardware.HidDeviceProxy.WriteBytes(Byte[] bytes) in P:\Source\Username\TTController\Source\TTController.Service\Hardware\HidDeviceProxy.cs:line 40
   at TTController.Service.Hardware.HidDeviceProxy.WriteReadBytes(Byte[] bytes) in P:\Source\Username\TTController\Source\TTController.Service\Hardware\HidDeviceProxy.cs:line 76
   at TTController.Plugin.Level20MousePadController.Level20MousePadControllerProxy.Init()
   at TTController.Service.Managers.DeviceManager.SearchForControllers() in P:\Source\Username\TTController\Source\TTController.Service\Managers\DeviceManager.cs:line 80
   at TTController.Service.Managers.DeviceManager..ctor() in P:\Source\Username\TTController\Source\TTController.Service\Managers\DeviceManager.cs:line 36
   at TTController.Service.TTService.Initialize() in P:\Source\Username\TTController\Source\TTController.Service\TTService.cs:line 64
   at TTController.Service.Program.<>c.<MainMenu>b__5_1() in P:\Source\Username\TTController\Source\TTController.Service\Program.cs:line 49
```

Here is my Debug Report:
I only have the one TT product, and if I remove the mousepad then all four 0x9007 lines disappear.
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 2004 [19041.1.x86fre.vb_release.191206-1406]
»        Build: 0.11.0-beta+24.Branch.master.Sha.724a739238b6d72d9f539b5a72a91917db668f1a
»
-------------------------------
Applications
-------------------------------
»
»
-------------------------------
HID
-------------------------------
»
»  [0x264a, 0x9007]: \\?\hid#vid_264a&pid_9007&mi_01&col03#9&2122ec9b&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
»  [0x264a, 0x9007]: \\?\hid#vid_264a&pid_9007&mi_01&col01#9&2122ec9b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
»  [0x264a, 0x9007]: \\?\hid#vid_264a&pid_9007&mi_01&col02#9&2122ec9b&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
»  [0x264a, 0x9007]: \\?\hid#vid_264a&pid_9007&mi_00#9&345308d9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
»
-------------------------------
Controllers
-------------------------------
»
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 47.0 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 33.0 °C
»                Temperature #3 (/lpc/nct6798d/temperature/3): 82.5 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 52.0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 76.0 °C
»
»  AMD Ryzen 7 3700X:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 58.8 °C
»                CCD1 (Tdie) (/amdcpu/0/temperature/3): 57.8 °C
»
»  NVIDIA GeForce GTX 970:
»        Temperature:
»                GPU Core (/gpu-nvidia/0/temperature/0): 48.0 °C
»
»  ST3000DM001-1ER166:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 37.0 °C
»
»  Crucial_CT256MX100SSD1:
»        Temperature:
»                Temperature (/ssd/0/temperature/0): 40.0 °C
»
-------------------------------
```

Here is my wireshark dump.
I went through plugging in the mouse pad, starting iTAKE engine, and setting all three LEDs to red
[level20mousepad.zip](https://github.com/MoshiMoshi0/TTController/files/6135886/level20mousepad.zip)

Thanks!

MoshiMoshi0, 2021-03-14:
That wireshark dump looks really quiet, can you post another one where you change the color more? Maybe set each led to a seperate color or maybe run some effects.
It also looks like the mouse pad might operate using reports and not just sending raw data. It that case it might be difficult to figure out the full report layout.

Also for some reason its detected 4 times? Do you have 4 mouse pads?

JesseHolwell, 2021-03-16:
Here is another wireshark dump where I go through the same as above and then switch profiles and go through different effects and colors

I'm unsure about the 4 showing up, only have the one mousepad.

Really appreciate your assistance

[level20mousepad2.zip](https://github.com/MoshiMoshi0/TTController/files/6146581/level20mousepad2.zip)

MoshiMoshi0, 2021-03-16:
Yeah I dont think I will be able to implement this without actual mouse pad to test any time soon.

First you would have to add `WriteReport`/`ReadReport` to [IHidDeviceProxy](https://github.com/MoshiMoshi0/TTController/blob/master/Source/TTController.Common/IHidDeviceProxy.cs) and implement them in [HidDeviceProxy](https://github.com/MoshiMoshi0/TTController/blob/master/Source/TTController.Service/Hardware/HidDeviceProxy.cs) using [HidSharp](https://www.zer7.com/software/hidsharp)
Then use those methods in `Init` and `SetRgb` methods in custom `Level20MousePadControllerProxy` class.
The set rgb data seems to be `09 00 12 ?? 00 00 00 00 00 00 4646b9 4646b9 4646b9 4646b9 00` but not sure why only 4 colors, from pictures the mouse pad seems to have more.  Not sure which reports will be the initialization ones.

JesseHolwell, 2021-03-16:
Right so it uses a different technology than the other TT devices? I am regretting buying this mousepad more and more as time goes on 😅

Thanks for the write up about how to implement reports that is very helpful, I'll see what I can do.

Regarding the zones, the mousepad has 3 LEDs: 1 in the logo and 2 that go around the edges. It uses an optic fibre cable to carry the light all the way around and just blends the two colors together.