MoshiMoshi0/TTController #217: ability to save fanprofile

Derik-for-real24, 2025-06-21:
HI, im concerned that TT RGB software does not have the capability to save fanspeed profiles, is it possible to have a feature that saves the fanspeed profile on a hardware level without completely relying on software by using TTController just ones.

Can such feature be offered to save fan profile ones and never ever be relying on software solution ?

MoshiMoshi0, 2025-06-22:
As far as I am aware there is no way to save any speed or RGB profiles directly to Thermaltake controllers.

closed 2025-06-22 by MoshiMoshi0

Derik-for-real24, 2025-06-22:
> As far as I am aware there is no way to save any speed or RGB profiles directly to Thermaltake controllers.

Thanks for response, but someone from this [https://community.thermaltake.com/index.php?/topic/75855-rgb-software-not-saving-settings/](url) post suggest, that with TT rgb plus 1.1.6 it actually has the ability to save fan profile, but the problem is that the particular software version for TT is not available anymore. So this should means that the device have a build in memory controller right ?

If yes, can you please look at the post and see if you can add such feature to TTcontroller tool.